openthread/openthread #7188: why Commissioner does it always turn off?

kenvals, 2021-11-19:
Due to the fact that the border router's web interface is still very crude, I have to enter devices using the command line.
i enter the following commands:
`sudo ot-ctl`
`commissioner start`
Commissioner: petitioning
Done
> Commissioner: active
`commissioner joiner add * **KEYDEVICE**`

then I press the button on the device and see the requests from the device:
`> ~ Discovery Request from 77826d590aa58043: version=2,joiner=1
Commissioner: Joiner start 77826d590aa58043
Commissioner: Joiner connect 77826d590aa58043
Commissioner: Joiner finalize 77826d590aa58043
Commissioner: Joiner end 77826d590aa58043`

But after a while everything dies for me, tell me why does it work like that?
`> Commissioner: disabled`

RPI 3b+, latest RCP, device nrf52840

jwhui, 2021-11-20:
The commissioner should not stop on its own unless it loses connectivity with the Leader.

Can you provide more information about your issue? Provide as much information as you can as requested by the GitHub Issue template:

---

**Describe the bug** A clear and concise description of what the bug is.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2021-12-22:
Closing stale issue.

closed 2021-12-22 by jwhui