openthread/openthread #7206: Spinel: How to use functions like spinel_prop_key_to_cstr() ?

OlivierGre, 2021-11-25:
In Spinel.c, there are: 
spinel_prop_key_to_cstr(spinel_prop_key_t prop_key)
spinel_capability_to_cstr(spinel_capability_t capability)
(and some others)

How are they intended to be used? They seem not used.

Thank you

abtink, 2021-11-29:
These are intended for logging (debugging) and often used by the host side driver (which also uses the `spinel` lib).
(`spinel` is the interface between NCP/RCP and a host).

As an example, these functions are used by [`wpantund`](https://github.com/openthread/wpantund) to log all sent and received spinel messages exchanged between host and NCP (e.g. from [SpinelNCPInstance::log_spinel_frame()](https://github.com/openthread/wpantund/blob/a8f3f761f6753b567d1e5ad22cbe6b0ceb6f2649/src/ncp-spinel/SpinelNCPInstance.cpp#L7000)).

OlivierGre, 2021-11-30:
Thanks.
in the meantime, I have added my own traces in NcpBase class and I have used them.

closed 2021-12-09 by OlivierGre