openthread/openthread #7208: [csl] Inconsistency when using otPlatRadioGetNow timestamp as a time reference for CSL Timer

marius-preda, 2021-11-25:
CSL scheduling for TX/RX will not work correctly if `otPlatRadioGetNow` returns a timestamp from the radio symbol timer, as the name suggests it should. This is because this timestamp is then used in the CSL code as an absolute timestamp for a CSL event to trigger using the CSL timer. This CSL timer uses otPlatTimerMicro implementation witch is a separate timer with a different timestamp.

For example:
```
    GetCslWindowEdges(timeAhead, timeAfter);

    switch (mCslState)
    {
    case kCslSample:
        mCslState = kCslSleep;

        mCslTimer.FireAt(mCslSampleTime - timeAhead);
        if (mState == kStateCslSample)
        {
#if !OPENTHREAD_CONFIG_MAC_CSL_DEBUG_ENABLE
            IgnoreError(Get<Radio>().Sleep()); // Don't actually sleep for debugging
#endif
            otLogDebgMac("CSL sleep %u", mCslTimer.GetNow().GetValue());
        }
        break;
```

Here `mCslSampleTime` is a timestamp from the radio but it is used as a timestamp for `mCslTimer.FireAt` that uses `TimerMicro` type. 
This is will not work correctly and timer events will not happen when they are supposed to.

I have checked Nordic's implementation and it seems to me that there is a workaround for this issue. I don't think this is the right solution as the CSL functionality should work ok for all platforms and I think `otPlatRadioGetNow` should actually be the radio timestamp.  What I mean is that `otPlatRadioGetNow` returns a timestamp from the Micro timer and the SFD detect timestamp is adjusted to the Micro timer timestamp:

```
    uint32_t offset =  (int32_t)otPlatAlarmMicroGetNow() - (int32_t)nrf_802154_first_symbol_timestamp_get(time, p_data[0]);
    receivedFrame->mInfo.mRxInfo.mTimestamp = nrf5AlarmGetCurrentTime() - offset;
```

My proposal would be to change the way we use `mCslTimer` from `FireAt` to `Start` and set a delay value calculated as the difference between the current radio timestamp and the calculated one. Also make `mCslSampleTime` `uint32_t` instead of `TimeMicro`. For example:

```
   GetCslWindowEdges(timeAhead, timeAfter);

    switch (mCslState)
    {
    case kCslSample:
        mCslState = kCslSleep;

        uint32_t delay = mCslSampleTime - timeAhead - otPlatRadioGetNow(&GetInstance()) ;
        mCslTimer.Start(delay);
```
#7054 
@George-Stefan

jwhui, 2021-11-25:
@marius-preda , thanks for raising this issue.

Yes, this is related to what we have been discussing in #7054.

I believe the intent of `otPlatRadioGetNow()` was to be called by code on the host to ensure it is using RCP's time.

I think we should replace use of `otPlatRadioGetNow()` in `sub_mac.cpp` with `otPlatAlarmMicroGetNow()`.

@Irving-cl , thoughts?

marius-preda, 2021-11-25:
@jwhui I think `otPlatRadioGetNow() ` is very useful and should not be replaced by `otPlatAlarmMicroGetNow` because the SFD detect timestamp and CSL sample time set in radio come from the radio timestamp. Also for platforms that use the Transmit/Receive At functionality the start time provided to the radio driver should be based on the `otPlatAlarmMicroGetNow` timestamp.

jwhui, 2021-11-25:
`otPlatRadioGetNow()` was introduced to maintain time offsets between the host and RCP in #5089.

nRF52840 uses the same timer for both `otPlatAlarmMicroGetNow()` and `otPlatRadioGetNow()` both simply return `nrf5AlarmGetCurrentTime()`.

Does your platform require using a different timer for `otPlatAlarmMicroGetNow()` and `otPlatRadioGetNow()`?

George-Stefan, 2021-11-25:
```C
        /**
         * Structure representing radio frame receive information.
         */
        struct
        {
            /**
             * The timestamp when the frame was received in microseconds.
             *
             * The value SHALL be the time when the SFD was received when TIME_SYNC or CSL is enabled.
             * Otherwise, the time when the MAC frame was fully received is also acceptable.
             *
             */
            uint64_t mTimestamp;

/**
 * Get the current estimated time (64bits width) of the radio chip.
 *
 * @param[in]   aInstance    A pointer to an OpenThread instance.
 *
 * @returns The current time in microseconds. UINT64_MAX when platform does not support or radio time is not ready.
 *
 */
uint64_t otPlatRadioGetNow(otInstance *aInstance);
```

From radio.h, I get that the radio chip has its own time base from which the timestamp is sampled from.

The time base of otPlatAlarmMicroGetNow() is not necessary the same as for the radio chip.
```C
/**
 * Get the current time.
 *
 * @returns  The current time in microseconds.
 *
 */
uint32_t otPlatAlarmMicroGetNow(void);
```

That's why we should only use time deltas when programing CSL related events using TimerMicro

George-Stefan, 2021-11-25:
#6981

Irving-cl, 2021-12-02:
Sorry for the late reply.

I think replacing `Timer::FireAt` by `Timer::Start` should be an easy solution and is not error-prone. (As I recall, I chose `FireAt` when implementing it because that way the code is simpler. Not 100% sure anyway).

Regarding replacing `otPlatRadioGetNow` with `otPlatAlarmMicroGetNow`, I think we inevitably have some interactions between the time obtained from the two sources. For example:
```
// In SubMac::HandleReceiveDone, we have
...
        if ((mCslPeriod > 0) && aFrame->mInfo.mRxInfo.mAckedWithSecEnhAck)
        {
            mCslLastSync = TimeMicro(static_cast<uint32_t>(aFrame->mInfo.mRxInfo.mTimestamp));
        }
...
// And this timestamp here comes from the radio

// Then in `SubMac::GetCslWindowEdges`, we have
...
    uint64_t curTime    = otPlatRadioGetNow(&GetInstance());
    uint64_t elapsed;
    uint32_t semiWindow;

    if (mCslLastSync.GetValue() > curTime)
    {
        elapsed = UINT64_MAX - mCslLastSync.GetValue() + curTime;
    }
    else
    {
        elapsed = curTime - mCslLastSync.GetValue();
    }
...

// Here `mCslLastSync` will interact with a 'timeNow'. If we use `otPlatAlarmMicroGetNow` here then it will cause some issues.
```

George-Stefan, 2021-12-02:
I can do a PR with a proposal to cover #7054 as well

Irving-cl, 2021-12-02:
Sorry again because I think I made a mistake in my last comment...

I just figured out that the only (original) intent for introducing `otPlatRadioGetNow` is to enabled the 'host' side (posix platform) to get the time from 'rcp' side (vendor platform) under RCP mode. So the 'host' side do have two time sources (one from the posix platform, one from the RCP). But for 'RCP' side, it should only have one. (Or say, our code just need one time source).

So I think the key issue here is, whether we want to have more than one time sources on RCP side. (I thought there has been two and they may be different).

From my perspective, our code only needs one source. Even in the example I mentioned above: let's say we have a platform which has two different time sources, and the frame's timestamp come from one source (Call it `GetNowA`). Then we always use the API `GetNowA` everywhere in our code. We also use it for the timer, alarmApi.

And even if we want to support RCP to get two different time sources, we may probably add another new API in OT. Because `otPlatRadioGetNow` is for the host to get time from the RCP. And now if RCP has two time sources, we don't know which one should be passed to the host.

As a summary, if we do need another time source on RCP, we may add another new API. But for our current code, we'd better always stick to one time source for our code (the one which set frame's timestamp).

Thoughts? @jwhui @George-Stefan @marius-preda

George-Stefan, 2021-12-02:
@Irving-cl,
I'm looking at https://openthread.io/platforms#platform_designs.
For SoC, NCP modes. it's obvious that the CSL scheduling doesn't cross the Spinel interface, it's local to the radio processor.
otPlatRadioGetNow() is not used on host side.

For RCP mode, could you please clarify where does the CSL Tx/Rx scheduling happen?
It looks like the CSL Rx scheduling is just on radio processor.
But the CSL Tx scheduling is on host side.

Irving-cl, 2021-12-03:
@George-Stefan 

> For SoC, NCP modes. it's obvious that the CSL scheduling doesn't cross the Spinel interface, it's local to the radio processor.
otPlatRadioGetNow() is not used on host side.

Yes, that' correct. Under NCP mode, everything just happens in the NCP part. (The host side should be just a set of interface to send spinel frames and get response).

---
For RCP, your understanding is basically correct. But note that we only support CSL_TRANSMITTER on RCP mode and CSL_RECEIVER cannot be used on RCP mode (can only be used with SoC or NCP). The reason is: 1. it's technically difficult to implement, should be feasible but may need to refactor and update many things. 2. RECEIVER is for low power devices, real products under RCP mode must have a main SoC and those things are usually have enough power supply (for example, routers)

jwhui, 2021-12-03:
> For RCP mode, could you please clarify where does the CSL Tx/Rx scheduling happen? It looks like the CSL Rx scheduling is just on radio processor. But the CSL Tx scheduling is on host side.

`otPlatRadioGetNow()` is use on host side here: https://github.com/openthread/openthread/blob/1267bdbf6897813ecd7c197df5d45cfe36027d21/src/core/thread/csl_tx_scheduler.cpp#L154

And results in a spinel transaction to retrieve the time from RCP here:
https://github.com/openthread/openthread/blob/1267bdbf6897813ecd7c197df5d45cfe36027d21/src/posix/platform/radio.cpp#L585-L589

marius-preda, 2021-12-03:
Hello, sorry for the late reply, I was on vacation for a few days. It looks like I and @George-Stefan have a different understanding of what `otPlatRadioGetNow()` does.

Most if not all platforms have a symbol timer running in radio/phy for precise timestamps like SFD timestamp and TX/RX hardware scheduling. This timer is different than the platform timer used in alarm.c for both micro and mili timers. We interpreted `otPlatRadioGetNow()` as being the radio/phy timestamp as this made a lot of sense for TX/RX AT operations and for precise receive timestamp used for CSL calculations. Apparently, this is not the case and `otPlatRadioGetNow()` should return the same time as `otPlatAlarmMicroGetNow`.

In this case, we always need to adjust the timestamps between the OT stack and the radio.c layer in both directions. For example, TX/RX AT functionality in our case(and probably others) uses the radio/phy timestamps to schedule the operations in hardware and if the absolute timestamp of the operation calculated and received from OT stack is using the `otPlatAlarmMicroGetNow` timestamp it will need to be adjusted to work correctly. This doesn't seem efficient. 

It seems that the best solution for our platforms and probably for the other new ones that want to implement CSL is to use our proposal. Start using the radio/phy timestamp for all CSL operations. Not sure what to do with `otPlatRadioGetNow()` in this case, if we should use a different function or reuse `otPlatRadioGetNow()` for this.

jwhui, 2021-12-04:
For the two platforms that have demonstrated CSL so far, the micro timer and radio timer are exactly the same.

- On nRF5 platforms, that is `nrf5AlarmGetCurrentTime()`.
- On EFR32 platforms, that is `RAIL_GetTime()`.

OpenThread only uses the microsecond timer in the sub-MAC component. Is it possible for your platform to have OpenThread's micro timer based on the symbol timer?

marius-preda, 2022-01-05:
Closing this issue as it clear to me now the use `otPlatRadioGetNow()`. My confusion was caused by a different interpretation of what this function should do.

closed 2022-01-05 by marius-preda