openthread/openthread #7211: How to use otPingSenderPing

purvisb, 2021-11-29:
I'm trying to use [otPingSenderPing](https://openthread.io/reference/group/api-ping-sender#otpingsenderping):

```
...
    else if(strcmp(command, "ping") == 0){
        otPingSenderConfig config;
        memset(&config, 0, sizeof(config));
        otIp6Address aDestination;

        otIp6AddressFromString(argument, &aDestination);

        config.mDestination         = aDestination;
        memcpy(&config.mSource, otThreadGetRloc(aContext), 16);
        config.mStatisticsCallback  = HandlePingStatistics;
        config.mCallbackContext     = aContext;
 
        error = otPingSenderPing(aContext, &config);

        otCliOutputFormat("%s\r\n",otThreadErrorToString(error));
    }

}

void HandlePingStatistics(const otPingSenderStatistics *aStatistics, void *aContext){
    otCliOutputFormat("%u packets transmitted, %u packets received.", aStatistics->mSentCount, aStatistics->mReceivedCount);
}
...
```

[Here is the rest of the code](https://github.com/purvisb/openthread/blob/8808b87b24fba94e7edcd3e1cb21dda3d1f285a9/examples/apps/cli/main.c#L395-L416)

It always says 1 packet transmitted and 0 packets received. I checked wireshark and I don't see any icmpv6 transmissions. Am I using `otPingSenderPing` wrong?

jwhui, 2021-11-29:
@purvisb , have you tried using the OpenThread CLI [ping](https://github.com/openthread/openthread/blob/main/src/cli/README.md#ping--i-source-ipaddr-size-count-interval-hoplimit-timeout) command?

purvisb, 2021-11-29:
@jwhui I wont be able to use the CLI for my use case. Maybe I can call CLI commands from main.c?

jwhui, 2021-12-13:
@purvisb , were you able to resolve this issue?

purvisb, 2021-12-13:
@jwhui no. I know how to output commands to the CLI. How do I take inputs from the CLI as well? If I ping something I want to receive the stats perhaps as a string.

perhaps I need an ping listener and just send the pings with the CLI?