openthread/openthread #7212: Incorrect source short address: "Frame rx failed, error:UnknownNeighbor"

tlifschitz, 2021-11-29:
**Describe the bug** A clear and concise description of what the bug is.
A host unix platform with RCP can't process incoming frames from a SED because the source address is not in the neighbor table of the host.  
After a factory reset, which makes all childs to reattach, the problem disappears and two more childs appear on the neighbor table. Probably those two which had an incorrect address.
**So is it possible that a SED get a pair of stale router ID and child ID, keeping an invalid RLOC?**

Note: I see that before the reset, the child table displays high child ID's (91, 92, 93, 94). This means many reattachs from the SEDs? Could this be related to the problem?

**To Reproduce** Information to reproduce the behavior, including:
Could not reproduce this in the lab. It happens to some of our deployed devices on the field.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Before factory reset:**
`
Nov 29 14:30:21 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:151, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:22 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:61, seqnum:152, type:Data, src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:22 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:153, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:22 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:154, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:22 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:155, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:22 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:156, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:22 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:157, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:23 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:158, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:23 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:159, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:23 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:160, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:23 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:161, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:23 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:162, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:24 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:163, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:24 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:164, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:24 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:165, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:24 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:166, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:24 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:167, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:24 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:168, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:25 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:169, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:25 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:170, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:25 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:171, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:25 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:172, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
Nov 29 14:30:25 GW0033 /usr/sbin/ot-daemon[460]: [INFO]-MAC-----: Frame rx failed, error:UnknownNeighbor, len:22, seqnum:173, type:Cmd(DataReq), src:0x9c59, dst:0x3000, sec:yes, ackreq:yes
`

`root@GW0033:~# ot-ctl router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
| 12 | 0x3000 |       63 |         0 |     0 |      0 |   0 | ea2412107eab65e4 |
`

`
root@GW0033:~# ot-ctl child table
| ID  | RLOC16 | Timeout    | Age        | LQ In | C_VN |R|D|N| Extended MAC     |
+-----+--------+------------+------------+-------+------+-+-+-+------------------+
|  91 | 0x305b |      20932 |          2 |     3 |   40 |0|0|1| d6d735dffd4a9395 |
|  92 | 0x305c |      20316 |          7 |     3 |   40 |0|0|1| 6a9d21a0269ff859 |
|  93 | 0x305d |      19418 |          8 |     3 |   40 |0|0|1| d2b7a96427c75cd7 |
|  94 | 0x305e |      19884 |          7 |     3 |   40 |0|0|1| 2e738f3e9eb1f243 |
`

** After factory reset:**
`
root@GW0033:~# ot-ctl router table
| ID | RLOC16 | Next Hop | Path Cost | LQ In | LQ Out | Age | Extended MAC     |
+----+--------+----------+-----------+-------+--------+-----+------------------+
| 43 | 0xac00 |       63 |         0 |     0 |      0 |   0 | 7e1845da3ffacee6 |
`

`
root@GW0033:~# ot-ctl child table
| ID  | RLOC16 | Timeout    | Age        | LQ In | C_VN |R|D|N| Extended MAC     |
+-----+--------+------------+------------+-------+------+-+-+-+------------------+
|   3 | 0xac03 |      19122 |          7 |     3 |   35 |0|0|1| 36c48dcbc9980ffb |
|   2 | 0xac02 |      19418 |          8 |     3 |   35 |0|0|1| d2b7a96427c75cd7 |
|   1 | 0xac01 |      20932 |          9 |     3 |   34 |0|0|1| d6d735dffd4a9395 |
|   5 | 0xac05 |      21492 |          4 |     3 |   35 |0|0|1| ce527ffc21643a87 |
|   4 | 0xac04 |      19884 |          0 |     3 |   35 |0|0|1| 2e738f3e9eb1f243 |
|   6 | 0xac06 |      20316 |          2 |     3 |   35 |0|0|1| 6a9d21a0269ff859 |
`

konradderda, 2021-11-29:
Hello @tlifschitz,
The problem you have described can be caused by a scenario where SED node is unable to communicate with its parent and if the timeout passes the child is dropped from the parent's perspective. 

Thread 1.2 introduces the "Enhanced Frame Pending" feature where SED sending a data poll is informed whether or not the parent has any data for it by the 'frame pending' bit in FCS part of ACK. The destination address is checked by the OpenThread's MAC layer, but at this moment the ACK is already transmitted (the radio platform will send ACK if it's requested by the originator).

This causes the problem because the child is happy that it has received an ACK with FP bit set to 0 so it can go to sleep while the parent drops every request in an "I don't know you" manner. Apparently, both parties comply with the current version of the protocol specification.

I believe this can be handled by the upper layer, but the method depends on your application design. You can add some short 'keep-alive' message that will be transmitted to the child from time to time, e.g. every Nth poll period. Also, the child can try to communicate with the parent from time to time. Or maybe the fact that child is dropped can be deduced by looking at your application design.

abtink, 2021-11-29:
You can check and enable ["child supervision"](https://openthread.io/guides/build/features/child-supervision) which was developed to address this situation.

For best performance, I would recommend enabling "child supervision" feature on both parent and child side, but it is also sufficient to enable it on sleepy child only.

tlifschitz, 2021-11-29:
@konradderda @abtink thank you both for your insight. We are currently detecting this situation by timeouts in out application messages. But we wanted to understand why this could happen, if it was a known issue and how to solve it.

In fact we suspected that this could be solved by enabling Child Supervision on both ends, so we did it. But we could not make a deploy yet with those changes, therefore we could not corroborate if the problem was solved. So thank you @abtink for confirming that.

tlifschitz, 2021-11-29:
Sorry @abtink why isn't child supervision enabled by default? Is there a reason for that? This is a quite inconvenient issue to find after having a couple of hundred deployed devices.

jwhui, 2021-11-29:
> Sorry @abtink why isn't child supervision enabled by default? Is there a reason for that?

Child Supervision is not a mechanism defined by the Thread Specification and is specific to the OpenThread implementation.

closed 2021-11-29 by tlifschitz