openthread/openthread #7215: Matter OT devices do not work with Apple BR implementations

sagar-apple, 2021-12-02:
Matter Devices using OT are unable to join and advertise networks hosted by Apple Border Routers. We need help identifying where the issue lies.

The Matter accessory log shows that it's unable to join the network and start the SRP client. "Operation refused for security reasons"
```
[00:00:30.914,337] <inf> net_openthread: [INFO]-MAC-----:     dst:[fdc7:3536:a992:0:a3f1:9d79:45b1:de0b]:49154
[00:00:30.914,520] <inf> net_openthread: [INFO]-SRP-----: [client] Received response
[00:00:30.914,642] <inf> net_openthread: [INFO]-SRP-----: [client] Server rejected Security code:5
[00:00:30.914,764] <inf> net_openthread: [INFO]-SRP-----: [client] Retry interval 1800 ms
[00:00:30.914,855] <inf> net_openthread: [INFO]-SRP-----: [client] State Updating -> ToRetry
[00:00:30.914,978] <err> chip: [DL]OnSrpClientNotification: Operation refused for security reasons
```

**Expected behavior** Matter OT devices should be able to join Apple Thread networks. 

**Console/log output** pcap from the BR and logs from the accessory with OT debug logs are attached.

[NRF5_SRP_NoClearSrpHost.txt](https://github.com/openthread/openthread/files/7644163/NRF5_SRP_NoClearSrpHost.txt)

[BR_utun0_no_clear.txt](https://github.com/openthread/openthread/files/7644171/BR_utun0_no_clear.txt) <- rename this to `.pcap`, I can't seem to attach it as a pcap file. 

@abtink

abtink, 2021-12-02:
@sagar-apple, we discussed this on matter slack already. I add a quick summary of findings below.

In short it seems to be an issue with how matter code is using the OT APIs and/or how the BR server is handling the message.

- The matter layer code seems to set host address as zero `::` and then request a host remove SRP update to be sent to server (usng `Srp::Clinet` OT APIs).
- This the message we see in the pcap file.
- The message itself (its format, included RRs, etc) seems to be all fine (no issue in OT SRP client implementation). 
- We see in next message from pcap that Apple SRP server implementation is reject (with response "refused (5)").
- It is likely that the server code does not like the `::` address of host.
- I know OpenThread SRP server code would check and reject a `::` (unspecified) address (so reasable for Apple SRP server implemetation to do the same).

My suggestion is to investigate the matter code base to see why/how the host address is being set as zero.

jwhui, 2021-12-03:
The Apple Border Router was not accepting a remove host update that did not include a AAAA record. The Apple Border Router implementation will be updated to support this.

While debugging, an bug in the SRP client was discovered around the lease time value on retransmissions. This has been addressed with:

- #7216

closed 2021-12-03 by jwhui