openthread/openthread #7224: [nrf52] spi-slave, the transaction length passed to `otPlatSpiSlaveTransactionCompleteCallback` seems incorrect

Irving-cl, 2021-12-09:
When we are using an NCP/RCP that uses SPI (as slave), after the a transaction completed, `otPlatSpiSlaveTransactionCompleteCallback` would be called (from here: https://github.com/openthread/ot-nrf528xx/blob/main/src/src/transport/spi-slave.c#L81):
```
        if (sCompleteCallback(sContext, aEvent->tx_buffer, aEvent->tx_buffer_size, aEvent->rx_buffer,
                              aEvent->rx_buffer_size, aEvent->rx_amount))
        {
            // Further processing is required.
            sFurtherProcessingFlag = true;

            otSysEventSignalPending();
        }
```
So the `aTransactionLength` in the complete callback is set as `aEvent->rx_amount`.

By the OT API definition (https://github.com/openthread/openthread/blob/main/include/openthread/platform/spi-slave.h#L142):
```
/**
 * ......
 * 
 * Note that even if `aInputBufLen` or `aOutputBufLen` (or both) are exhausted before the SPI master finishes a
 * transaction, the ongoing size of the transaction must still be kept track of to be passed to the transaction
 * complete callback. For example, if `aInputBufLen` is equal to 10 and `aOutputBufLen` equal to 20 and the SPI master
 * clocks out 30 bytes, the value 30 is passed to the transaction complete callback.
 *
 * ......
 * 
 * @param[in] aOutputBuf              Data to be written to MISO pin
 * @param[in] aOutputBufLen           Size of the output buffer, in bytes
 * @param[in] aInputBuf               Data to be read from MOSI pin
 * @param[in] aInputBufLen            Size of the input buffer, in bytes
 * @param[in] aRequestTransactionFlag Set to true if host interrupt should be set
 *
 * @retval OT_ERROR_NONE           Transaction was successfully prepared.
 * @retval OT_ERROR_BUSY           A transaction is currently in progress.
 * @retval OT_ERROR_INVALID_STATE  otPlatSpiSlaveEnable() hasn't been called.
 *
 */
otError otPlatSpiSlavePrepareTransaction(uint8_t *aOutputBuf,
                                         uint16_t aOutputBufLen,
                                         uint8_t *aInputBuf,
                                         uint16_t aInputBufLen,
                                         bool     aRequestTransactionFlag);

```
So the `aTransactionLength` should be set to the actual transaction length (determined by the SPI master). As I know, `aEvent->rx_amount` does not equal to the transaction amount. And it seems that both `tx_amount` and `rx_amount` will not exceed their buffer size in callback. However this API has not been updated for some time. Is it still the case now? @abtink 

---
Currently we don't have fatal issues with this because both sides will re-schedule the SPI frame if they think they haven't sent a complete frame. But it causes some waste and doesn't meet the specification of the API.

For example, when a host starts, it first reset the NCP and then receive the first SPI frame from the slave. The exchange would go like this (logs captured on host):
```
12-31 16:26:43.767  5160  5160 I otbr-agent: [INFO]-UTILS---: Running 0.2.0                                                                                                                                                                                                            
12-31 16:26:43.767  5160  5160 I otbr-agent: [INFO]-UTILS---: Thread version: 1.2.0                                                                                                                                                                                                    
12-31 16:26:43.767  5160  5160 I otbr-agent: [INFO]-UTILS---: Thread interface: wpan0                                                                                                                                                                                                  
12-31 16:26:43.767  5160  5160 I otbr-agent: [INFO]-UTILS---: Backbone interface:                                                                                                                                                                                                      
12-31 16:26:43.767  5160  5160 I otbr-agent: [INFO]-UTILS---: Radio URL: spinel+spi:///dev/spidev0.0?no-reset=1&gpio-int-device=/dev/gpiochi                                                                                                                                           
p0&gpio-int-line=32&gpio-reset-device=/dev/gpiochip0&gpio-reset-line=39&spi-mode=0&spi-speed=1000000&spi-cs-delay=0&spi-reset-delay=5                                                                                                                                                  
12-31 16:26:43.781  5160  5160 I otbr-agent: [NOTE]-PLAT----: Triggered hardware reset                                                                                                                                                                                                 
12-31 16:26:43.787  5160  5160 W otbr-agent: [WARN]-PLAT----: RadioSpinel::Init WaitResponse                                                                                                                                                                                           
12-31 16:26:43.789  5160  5160 W otbr-agent: [WARN]============================[SPI-TX len=053]=============================                                                                                                                                                           
12-31 16:26:43.789  5160  5160 W otbr-agent: [WARN]| 82 20 00 00 00 00 00 00 | 00 00 00 00 00 00 00 00 | . ..............                                                                                                                                                              
12-31 16:26:43.789  5160  5160 W otbr-agent: [WARN]| 00 00 00 00 00 00 00 00 | 00 00 00 00 00 00 00 00 | ................                                                                                                                                                              
12-31 16:26:43.789  5160  5160 W otbr-agent: [WARN]| 00 00 00 00 00 00 00 00 | 00 00 00 00 00 00 00 00 | ................                                                                                                                                                              
12-31 16:26:43.789  5160  5160 W otbr-agent: [WARN]| 00 00 00 00 00 .. .. .. | .. .. .. .. .. .. .. .. | ................                                                                                                                                                              
12-31 16:26:43.789  5160  5160 W otbr-agent: [WARN]------------------------------------------------------------------------                                                                                                                                                            
12-31 16:26:43.789  5160  5160 W otbr-agent: [WARN]============================[SPI-RX len=053]=============================                                                                                                                                                           
12-31 16:26:43.789  5160  5160 W otbr-agent: [WARN]| 82 00 00 04 00 80 06 00 | 71 FF FF FF FF FF FF FF | ........q.......                                                                                                                                                              
12-31 16:26:43.789  5160  5160 W otbr-agent: [WARN]| FF FF FF FF FF FF FF FF | FF FF FF FF FF FF FF FF | ................                                                                                                                                                              
12-31 16:26:43.793  5160  5160 W otbr-agent: [WARN]| FF FF FF FF FF FF FF FF | FF FF FF FF FF FF FF FF | ................                                                                                                                                                              
12-31 16:26:43.793  5160  5160 W otbr-agent: [WARN]| FF FF FF FF FF .. .. .. | .. .. .. .. .. .. .. .. | ................                                                                                                                                                              
12-31 16:26:43.794  5160  5160 W otbr-agent: [WARN]------------------------------------------------------------------------                                                                                                     [... cut ...] tus' frame again (and this time the receive buffer changes to a larger one).

Note that in the two transations, the actual transaction lengths are both **53**. So:
* Shall we still adhere to the spec of the API and pass the transaction length (specified by the master) to the callback?
* It looks like we cannot know this number from the event of nordic driver. Is there a way for us to get that? Or at least we pass max(event->tx_amount, event->rx_amount) to the callback.

Thoughts? @abtink @LuDuda

abtink, 2021-12-13:
@Irving-cl , yes the SPI-slave platform implementation should follow the documentation in the header regarding this:
https://github.com/openthread/openthread/blob/main/include/openthread/platform/spi-slave.h#L142

```c++
/**
...
* Note that even if `aInputBufLen` or `aOutputBufLen` (or both) are exhausted before the SPI master finishes a
 * transaction, the ongoing size of the transaction must still be kept track of to be passed to the transaction
 * complete callback. For example, if `aInputBufLen` is equal to 10 and `aOutputBufLen` equal to 20 and the SPI master
 * clocks out 30 bytes, the value 30 is passed to the transaction complete callback.
```

Irving-cl, 2021-12-14:
Hi @MarekPorwisz,

Could you take a look on this?

MarekPorwisz, 2021-12-14:
@Irving-cl  I did some investigation and it seems our SPI does not provide info about transaction length. I created a PR https://github.com/openthread/ot-nrf528xx/pull/215 that takes the maximum value of the rx and tx data length. 
This is not a perfect solution but should fix your issue.
I do not have possibility to test it so I would be grateful if you could try if the problem is gone for you after this fix.

Irving-cl, 2021-12-15:
@MarekPorwisz 

Thanks! Sure, I'll check. Actually I tried locally to have use MAX of tx_amount and rx_amount. It can work.

closed 2021-12-15 by jwhui