openthread/openthread #7225: How to ping a Thread device from an Android phone?

OlivierGre, 2021-12-09:
I have a Thread device (v1.1) connected to a thread network and I do have created an OMR prefix (thanks to Jonathan's advices in another ticket). I can see with "ipaddr" that the device does have an IP address starting with the OMR prefix.

I now want to ping this device from Android as indicated on https://openthread.io/codelabs/openthread-border-router#6

Unfortunately there is a restriction:
_"Unfortunately, the Android version of the Network Analyzer App doesn't support mDNS queries for the ping utility and we cannot ping the hostname ot-host.local directly (we can ping the hostname with the iOS version of the App)."_

Would you know a solution to be able to ping the thread device on Android? Is there another application that I could use?

I have a Mac so I have also tried to ping the device from my mac but I'm not lucky: My Mac is running "Monterey" whereas "Big Sur" is needed :o(

wgtdkp, 2021-12-09:
If you finish reading https://openthread.io/codelabs/openthread-border-router#6, you will find out that the Android app can ping IPv6 address directly but cannot ping the host name.

OlivierGre, 2021-12-09:
That was not clear to me.
I have tried again but the ping doesn't work. I have tried also to ping the IPV6 address of the OTBR (the one starting with OMR prefix fd00:dead:beef:cafe...) but it also fails.

OlivierGre, 2021-12-09:
Here is an extract of the ifconfig command on the Border Router:

```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.62  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::dea6:32ff:fe21:3e1e  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:21:3e:1e  txqueuelen 1000  (Ethernet)
        RX packets 5006  bytes 2248803 (2.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 12444  bytes 3094663 (2.9 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wpan0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1280
        inet6 fd80:6a9f:4948:7235:0:ff:fe00:fc00  prefixlen 64  scopeid 0x0<global>
        inet6 fd00:dead:beef:cafe:70eb:1d49:4a8e:e1d7  prefixlen 64  scopeid 0x0<global>
        inet6 fd80:6a9f:4948:7235:0:ff:fe00:fc10  prefixlen 64  scopeid 0x0<global>
        inet6 fd80:6a9f:4948:7235:0:ff:fe00:3000  prefixlen 64  scopeid 0x0<global>
        inet6 fd80:6a9f:4948:7235:9b9a:133d:7bda:5cf0  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::c41b:22f5:e1ad:25a0  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 2  bytes 112 (112.0 B)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 20  bytes 3580 (3.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
Form a mac, I can ping the IPV4 address 192.168.1.62 and the IPV6 address fe80::dea6:32ff:fe21:3e1e but I don't succeed to ping  fd00:dead:beef:cafe:70eb:1d49:4a8e:e1d7.

In my thread network, my Joiner device has the following addresses:
XXX

I don't succeed to ping the one starting with fd00:dead:beef:cafe.

wgtdkp, 2021-12-09:
**Can you confirm that you are following the https://openthread.io/codelabs/openthread-border-router guide to set up your BR?**  Except for using Thread 1.1.

> I have a Mac so I have also tried to ping the device from my mac but I'm not lucky: My Mac is running "Monterey" whereas "Big Sur" is needed :o(

I think you should expect that your Mac should not work.

Please help provide below information:

1. The output of `ip -6 route list` on your BR device
2. The output of `ot-ctl netdata show` on your BR device
3. The output of `netdata show` on your end device
4. The screenshot of failing to ping the IPv6 address
5. The version of your Android phone

OlivierGre, 2021-12-09:
> **Can you confirm that you are following the https://openthread.io/codelabs/openthread-border-router guide to set up your BR?** Except for using Thread 1.1.

You're raising a very good point! I haven't typed the commands to form the thread network because it would create a new dataset. I would like to keep my current dataset to have the same as my colleagues. I see in the "netdata show" command that I miss something (see below) because "routes" is empty. This is probably the reason why of my issue. Is there a way to keep my dataset but to enhance it with what is missing in routes?

> Please help provide below information:
> 
> 1. The output of `ip -6 route list` on your BR device
```
pi@raspberrypi:~ $ ip -6 route list 
::1 dev lo proto kernel metric 256 pref medium
64:ff9b::/96 dev nat64 metric 1024 pref medium
fd00:dead:beef:cafe::/64 dev wpan0 proto kernel metric 256 pref medium
fd80:6a9f:4948:7235::/64 dev wpan0 proto kernel metric 256 pref medium
fdaa:bb:1::2 dev nat64 proto kernel metric 256 pref medium
fe80::/64 dev nat64 proto kernel metric 256 pref medium
fe80::/64 dev wlan0 proto kernel metric 256 pref medium
fe80::/64 dev wpan0 proto kernel metric 256 pref medium

```

> 2. The output of `ot-ctl netdata show` on your BR device
```
netdata show
Prefixes:
fd00:dead:beef:cafe::/64 paros med 3000
Routes:
Services:
44970 5d fd806a9f494872359b9a133d7bda5cf0d11f s 3000
Done
```
> 3. The output of `netdata show` on your end device
I get: "Error 6: Parse"

> 4. The screenshot of failing to ping the IPv6 address
```
ping6 -I en0 fd00:dead:beef:cafe:70eb:1d49:4a8e:e1d7
PING6(56=40+8+8 bytes) fe80::1428:41d1:ff95:9b48%en0 --> fd00:dead:beef:cafe:70eb:1d49:4a8e:e1d7
ping6: sendmsg: No route to host
ping6: wrote fd00:dead:beef:cafe:70eb:1d49:4a8e:e1d7 16 chars, ret=-1
ping6: sendmsg: No route to host
ping6: wrote fd00:dead:beef:cafe:70eb:1d49:4a8e:e1d7 16 chars, ret=-1
ping6: sendmsg: No route to host

ping6 fd00:dead:beef:cafe:b0b3:d1e0:8158:4bc2 -I en0
PING6(56=40+8+8 bytes) fe80::1428:41d1:ff95:9b48%en0 --> fd00:dead:beef:cafe:b0b3:d1e0:8158:4bc2
ping6: sendmsg: No route to host
ping6: wrote fd00:dead:beef:cafe:b0b3:d1e0:8158:4bc2 16 chars, ret=-1
ping6: sendmsg: No route to host
ping6: wrote fd00:dead:beef:cafe:b0b3:d1e0:8158:4bc2 16 chars, ret=-1
ping6: sendmsg: No route to host
```

> 5. The version of your Android phone
Google Pixel 3 running Android 12

OlivierGre, 2021-12-09:
I have cleared the dataset of the BR and followed the instructions on https://openthread.io/codelabs/openthread-border-router#2  but, when I type the netdata show command, I have nothing in prefixes and routes:

jwhui, 2021-12-09:
@OlivierGre , what git commit are you using for building OTBR? It appears that the border routing feature is not enabled.

If you want the border router to automatically generate a routable prefixes for the Thread and Wi-Fi networks and advertise routes, you will need to enable the border routing feature with `BORDER_ROUTING=1` in the build command.

OlivierGre, 2021-12-10:
> @OlivierGre , what git commit are you using for building OTBR? It appears that the border routing feature is not enabled.
>
> If you want the border router to automatically generate a routable prefixes [...]

I have rebased it yesterday to get this fix https://github.com/openthread/ot-br-posix/pull/1141.
I'm on the latest commit of main branch:
```

commit a41fc11dc27bbc6c989ccfc92a78ddd684e6c1ad (HEAD -> main, origin/main, origin/HEAD)
Author: whd <7058128+superwhd@users.noreply.github.com>
Date:   Thu Dec 9 14:25:36 2021 +0800

    [border-router] start mDNS daemon before starting otbr-agent (#1141)
    
    This commit updates the .service file to start the mDNS daemon before
    starting otbr-agent.

```