openthread/openthread #7226: [cert] OTBR crashing intermittently in between 1.2 certification tests

suveshpratapa, 2021-12-09:
We notice this issue happen during 1.2 certification.  Originally it was uncovered when we ran a full SiLabs/SiLabs testbed, but we've also come across it in certain multi-vendor cases.

```

```

This OTBR crash can happen during ANY test.  For example, I have included traces and logs from two different tests where it was reproduced different times.  These were with the OTBR acting as Router DUT, but it can also be reproduced when it is a leader or other role.
- 5.10.16	MATN-TC-20: Automatic re-registration by Parent Router on behalf of MTD
- 5.10.18	MATN-TC-22: Use of low MLR timeout defaults to MLR_TIMEOUT_MIN

Basically, this issue seems to be induced whenever there are low power cases run as part of the suite.  If we run a set of tests (also including low power tests), then the issue can be reproduced frequently and easily.  If low power tests are omitted, then we don't see this happen (though, since the issue is intermittent, it's possible we just haven't run into it easier).

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

suveshpratapa, 2021-12-09:
@jwhui @simonlingoogle @JaneFromSilabs

simonlingoogle, 2021-12-09:
@jinran-google

simonlingoogle, 2021-12-09:
We encountered this same issue after ot-ctl reset. 
The issue remains even after BR reboots.

simonlingoogle, 2021-12-09:
Currently, we have two RCP reset types: STACK or SYSTEM and the default one is STACK. 

I think the certification build is using STACK because it's by default. 

If so, the issue seems to be unrelated to how the RCP device is reset. So, I think it's mostly like a Host side issue.

suveshpratapa, 2021-12-09:
@simonlingoogle Thanks for letting us know the issue is reproducible on your end.  Let us know if we can provide you with any debug information from our side. Are you able to get it to reproduce manually by hand?

Apparently this issue is not being hit on the Nordic platform in certification... they are using a commit of ot-br-posix on which we were able to hit it.  I know you said this is most likely a host side issue, but could it have something to do with the type of UART connection?  Are you hitting it with a UART/USB case?

simonlingoogle, 2021-12-10:
> Are you hitting it with a UART/USB case?

Recently we have this issue using SPI.

JaneFromSilabs, 2021-12-13:
I am seeing what I think is another version of this issue today showing up as SSH connection problems (preceded by the same timeout message).  The Test Harness is also unable to recover from this.
```
Dec 13 13:59:56 sil-rpi-1 otbr-agent[6693]: 00:00:41.136 [INFO]-PLAT----: RCP => [INFO]-PLAT----: Received 30 bytes
Dec 13 13:59:56 sil-rpi-1 otbr-agent[6693]: 00:00:41.164 [INFO]-MAC-----: Rx data poll, src:0xb802, qed_msgs:5, rss:-35, ack-fp:1
Dec 13 13:59:56 sil-rpi-1 otbr-agent[6693]: 00:00:41.241 [INFO]-PLAT----: RCP => [INFO]-PLAT----: Received 30 bytes
Dec 13 13:59:56 sil-rpi-1 otbr-agent[6693]: 00:00:41.274 [INFO]-MAC-----: Rx data poll, src:0xb802, qed_msgs:5, rss:-35, ack-fp:1
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:41.773 [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:146e:a00:0:c,0xb802)
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:42.136 [INFO]-PLAT----: RCP => [INFO]-PLAT----: Received 115 bytes
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:42.174 [INFO]-MAC-----: Received IPv6 UDP msg, len:113, chksum:e3f8, from:166e0a000000000c, sec:yes, prio:net, rss:-35.0
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:42.175 [INFO]-MAC-----:     src:[fe80:0:0:0:146e:a00:0:c]:19788
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:42.175 [INFO]-MAC-----:     dst:[fe80:0:0:0:146e:a00:0:1]:19788
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:42.175 [INFO]-MLE-----: Receive Child Update Request from child (fe80:0:0:0:146e:a00:0:c)
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:42.175 [INFO]-MLE-----: Child 0xb802 IPv6 address[1]=fd00:db8:0:0:4cb:4d31:6aaa:1ebf
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:42.175 [INFO]-MLE-----: Child 0xb802 IPv6 address[2]=fd68:d9e4:2936:14dc:5d3c:6b94:252e:5f0f
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:42.175 [INFO]-MLE-----: Child 0xb802 has 2 registered IPv6 addresses, 2 addresses stored
Dec 13 13:59:57 sil-rpi-1 otbr-agent[6693]: 00:00:42.175 [INFO]-MLE-----: Send Child Update Response to child (fe80:0:0:0:146e:a00:0:c,0xb802)
Dec 13 13:59:58 sil-rpi-1 otbr-agent[6693]: 00:00:42.776 [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:146e:a00:0:c,0xb802)
Dec 13 13:59:58 sil-rpi-1 otbr-agent[6693]: 00:00:42.978 [WARN]-PLAT----: radio tx timeout
Dec 13 14:00:27 sil-rpi-1 systemd[1]: Started Session 85 of user pi.
Dec 13 14:00:48 sil-rpi-1 systemd[1]: Started Session 86 of user pi.
```

JaneFromSilabs, 2021-12-13:
Here is the test harness logging from this SSH issue:
```
None - Dec 13 08:59:45 - ============> bash('sudo ot-ctl -- state',) returns ['detached', 'Done']
None - Dec 13 08:59:45 - Expecting [state]
None - Dec 13 08:59:45 - readline: state
None - Dec 13 08:59:45 - Expected [state]
None - Dec 13 08:59:45 - readline: detached
None - Dec 13 08:59:45 - readline: Done
None - Dec 13 08:59:45 - ========> __executeCommand('state',) returns ['detached', 'Done']
None - Dec 13 08:59:45 - ====> __isDeviceAttached returns False
None - Dec 13 08:59:45 - command: state
None - Dec 13 08:59:45 - ============> bash('sudo ot-ctl -- state',) returns ['leader', 'Done']
None - Dec 13 08:59:45 - Expecting [state]
None - Dec 13 08:59:45 - readline: state
None - Dec 13 08:59:45 - Expected [state]
None - Dec 13 08:59:45 - readline: leader
None - Dec 13 08:59:45 - readline: Done
None - Dec 13 08:59:45 - ========> __executeCommand('state',) returns ['leader', 'Done']
None - Dec 13 08:59:45 - ====> __isDeviceAttached returns True
None - Dec 13 08:59:45 - > joinNetwork(1,) returns True
Status --> Auto DUT Joining operation completed  [1]

Status --> Updating new channel on all testbed devices  [2]
```
...
```Completed: 'Test 5 3 1 SSED Attachment' at: 2021_12_13_09_00_05
[192.168.1.111:22] call reset
None - Dec 13 09:00:05 - command: factoryreset
SSH connection is lost, try reconnect after 1 second.
SSH connection is lost, try reconnect after 1 second.
```

romacdon, 2021-12-14:
Hi @simonlingoogle - have you had any luck debugging this issue further?  We (Silabs) are continuing our debug but have not made that much progress?

simonlingoogle, 2021-12-15:
@romacdon 

Some questions:
- Do you have a reliable way to reproduce this issue? 
- Currently otbr-agent is using `SPINEL_RESET_STACK`. Could you try using `SPINEL_RESET_PLATFORM` and see if the issue remains?

nykh2010, 2021-12-15:
@simonlingoogle 
I have the same issue. We use SiLabs efr32mg13 RCP as Border Router, the host is PC. It was very easy to reproduce when doing a multicast test for 60 nodes.

`Dec 15 10:53:07 mi otbr-agent[128354]: [INFO]-MLE-----: Receive Advertisement (fe80:0:0:0:649b:4992:e226:35f2,0xf800)
Dec 15 10:53:08 mi otbr-agent[128354]: [INFO]-MAC-----: Received IPv6 UDP msg, len:106, chksum:1c82, from:52837ac2eb2c9fa9, sec:no, prio:net, rss:-39.0, radio:15.4
Dec 15 10:53:08 mi otbr-agent[128354]: [INFO]-MAC-----:     src:[fe80:0:0:0:5083:7ac2:eb2c:9fa9]:19788
Dec 15 10:53:08 mi otbr-agent[128354]: [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
Dec 15 10:53:08 mi otbr-agent[128354]: [INFO]-MLE-----: Receive Advertisement (fe80:0:0:0:5083:7ac2:eb2c:9fa9,0xb000)
Dec 15 10:53:08 mi otbr-agent[128354]: [WARN]-PLAT----: radio tx timeout
Dec 15 10:53:08 mi otbr-agent[128354]: [WARN]-PLAT----: RCP failure detected
Dec 15 10:53:08 mi otbr-agent[128354]: [WARN]-PLAT----: Trying to recover (1/2)
Dec 15 10:53:09 mi otbr-agent[128354]: [INFO]-PLAT----: RCP reset: RESET_SOFTWARE
Dec 15 10:53:09 mi otbr-agent[128354]: [INFO]-CORE----: [settings] Read NetworkInfo {rloc:0x7000, extaddr:6e5e8349565181c0, role:Leader, mode:0x0f, version:3, keyseq:0x0, ...
Dec 15 10:53:09 mi otbr-agent[128354]: [INFO]-CORE----: [settings] ... pid:0x4e712432, mlecntr:0x7d5, maccntr:0x7d0, mliid:bf473701dd2c54bc}
Dec 15 10:53:09 mi otbr-agent[128354]: [WARN]-PLAT----: Error processing result: NotImplemented
Dec 15 10:53:09 mi otbr-agent[128354]: [WARN]-PLAT----: Error waiting response: NotImplemented
Dec 15 10:53:09 mi otbr-agent[128354]: [CRIT]-PLAT----: RestoreProperties() at /home/xulingfeng/Documents/matter/ot-br-posix/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2344: Failure
Dec 15 10:53:09 mi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=1/FAILURE
Dec 15 10:53:09 mi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Dec 15 10:53:14 mi systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 5012.`

romacdon, 2021-12-15:
Thanks @simonlingoogle 

> @romacdon
> 
> Some questions:
> 
> * Do you have a reliable way to reproduce this issue?

Unfortunately no. However, it is easily reproduced when running Thread 1.2 certification tests against the test harness but its unpredictable when it will occur.

> * Currently otbr-agent is using `SPINEL_RESET_STACK`. Could you try using `SPINEL_RESET_PLATFORM` and see if the issue remains?

@suveshpratapa / @JaneFromSilabs - can you guys give this a try?

JaneFromSilabs, 2021-12-15:
Thanks, @simonlingoogle, I'll give this a try.  While I cannot reproduce this on command, it does happen within about 10-15 tests for me, sometimes sooner.

JaneFromSilabs, 2021-12-15:
Hi @simonlingoogle ,  
I tried running with this line changed:
https://stash.silabs.com/projects/WMN_STACKS/repos/openthread/browse/src/lib/spinel/radio_spinel_impl.hpp#2325
to SPINEL_RESET_PLATFORM but this resulted in eventual failure.
```None - Dec 15 09:59:55 - > powerDown returns None
Status --> Waiting for 140 seconds .. [Waiting for BR_1_2 becomes BR_1]  [77]

[192.168.1.115:22] call getRloc
None - Dec 15 10:02:16 - command: ipaddr rloc
SSH connection is lost, try reconnect after 1 second.
SSH connection is lost, try reconnect after 1 second.
None - Dec 15 10:03:19 - FUNC bash failed:
None - Dec 15 10:03:19 - FUNC __executeCommand failed:
None - Dec 15 10:03:19 - command: ipaddr rloc
SSH connection is lost, try reconnect after 1 second.
SSH connection is lost, try reconnect after 1 second.```

suveshpratapa, 2021-12-16:
@simonlingoogle Can you confirm what version of raspberry pi you were using when reproducing this issue? Raspberry pi 3B/3B+/4?

JaneFromSilabs, 2021-12-16:
Hi @simonlingoogle, 
I was wondering if you could provide a little more detail on this comment:
```The issue remains even after BR reboots.```
What exactly is remaining after the reboot?

simonlingoogle, 2021-12-21:
@edmont do you experience the same issue on Nordic boards?

edmont, 2021-12-21:
Hi @simonlingoogle, we haven't observed this issue. Notice we use a [Nordic Connect SDK based RCP](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/openthread/coprocessor/README.html) and radio capabilities are different.

suveshpratapa, 2022-01-10:
@simonlingoogle Any success in finding out the root cause of this issue?  Enabling RCP restoration with `OT_RCP_RESTORATION_MAX_COUNT` seems to help things along some of the times.  We're still noticing our OTBR get into this weird state in the testbed from time to time, specific to certain raspberry pi hardware.

jwhui, 2022-09-13:
Closing stale issue.

closed 2022-09-13 by jwhui

vanthome, 2025-01-19:
@suveshpratapa we are facing this issue with the latest available SiSDK and OTBR and not only during certification but also during normal operation when large packets are transmitted. Why is Silabs not fixing this in the SDK/ firmware?

silabs-sebastien, 2025-01-30:
@vanthome I've contacted you through the Silabs support system to help investigate this issue with you.

vanthome, 2025-01-30:
@silabs-sebastien thanks for taking care of this. I have described the case here now with all details you need:

https://community.silabs.com/s/question/0D7Vm000006axZBKAY/detail?language=en_US&emkind=chatterCommentNotification&emtm=1738246970175&fromEmail=1&s1ext=0&s1nid=0DB1M000000k9ha&s1oid=00DA0000000L2kI&s1uid=0051M000008EHos&t=1738247522733