openthread/openthread #7284: RX window is not correct when not using OT_RADIO_CAPS_RECEIVE_TIMING

marius-preda, 2022-01-04:
Hello,

After the discussion from #7208 I changed the way I use otPlatRadioGetNow and decided that for NXP architecture is better and simpler to not use OT_RADIO_CAPS_RECEIVE_TIMING and to let the Open Thread stack control RX and sleep periods during CSL operation as is the case for normal polling case.  

After doing this change I have noticed, using a logic analyzer, that the receive window was not looking ok at all. I was testing with a CSL period of 3125(500 ms) and instead of a 6-7 ms RX window I was seeing a 490 ms window but the log was reporting a correct window size.

So I checked the code with @George-Stefan and we think there is a bug in `SubMac::HandleCslTimer` where mCslSampleTime
is incremented with the period before the timer is armed for the case where receive timing is not supported.

The code looks like this right now:

```
case kCslSleep:
    mCslSampleTime += periodUs;

    if (RadioSupportsReceiveTiming())
    {
        mCslTimer.FireAt(mCslSampleTime - timeAhead);
        timeAhead -= kCslReceiveTimeAhead;
    }
    else
    {
        mCslTimer.FireAt(mCslSampleTime + timeAfter);
        mCslState = kCslSample;
    }
```

But it should look something like this:
```
case kCslSleep:
    if (RadioSupportsReceiveTiming())
    {
        mCslSampleTime += periodUs;

        mCslTimer.FireAt(mCslSampleTime - timeAhead);
        timeAhead -= kCslReceiveTimeAhead;
    }
    else
    {
        mCslTimer.FireAt(mCslSampleTime + timeAfter);
        mCslState = kCslSample;

        mCslSampleTime += periodUs;
    }
```

In the case of "receive at" we want the timer to fire before the next RX window to program the exact RX activity in the radio and in the other case, the timer will fire when the current RX window will end to close the receiver and go to sleep. 

I can close #7208 as it was a misunderstanding on our side but we still have #7054 open and we can create a PR to fix both issues if the solutions proposed are ok.

Irving-cl, 2022-01-05:
Hi @marius-preda,

This makes sense to me. Thanks for finding the issue! Looks good to me to use such fix.

marius-preda, 2022-01-05:
Hi @Irving-cl,

Thanks for the fast reply! If you can fix it we are very happy. Thanks!

jwhui, 2022-01-05:
> Thanks for the fast reply! If you can fix it we are very happy. Thanks!

@marius-preda , can you clarify whether you would like to submit the PR(s) or one of us? Your prior comments seemed to indicate that you would like to submit the PRs. Either way is fine with us. Thanks!

marius-preda, 2022-01-06:
Hi @jwhui, 

Sorry for the confusion. I proposed to do the PR(s) but I misread @Irving-cl comment understanding that he wants to do the PR so I wanted to let him in this case. 

I will try to do it tomorrow. For #7054 the discussion seems not finalized. @George-Stefan proposed a change and we are not clear if you are ok with the proposal.

jwhui, 2022-01-06:
> I will try to do it tomorrow. For #7054 the discussion seems not finalized. @George-Stefan proposed a change and we are not clear if you are ok with the proposal.

Sounds good! I just responded in the other issue - https://github.com/openthread/openthread/issues/7054#issuecomment-1006978547.

closed 2022-01-11 by marius-preda