openthread/openthread #7286: Why the baudrate of the RCP firmware of the TBR is only 115200 when the network rate of Thread can reach 250kbps?

HeFeng1947, 2022-01-05:
According to the ARCHITECTURE of RCP mode, the Radio module contains only PHY layer and MAC layer, and other data needs to be reported to the host through the uart. So what rate of data needs to be reported to the host in the limit case? Is the baudrate of 115200 sufficient for normal communication in the limit case?

jwhui, 2022-01-05:
@HeFeng1947 , thanks for submitting this issue.

While the IEEE 802.15.4 2.4 GHz O-QPSK PHY supports 250 kbps, the actual data rate in practice is typically lower due to various overheads (e.g. MAC backoff, interframe spacing, RX-TX turnaround times, data-acknowledgment transmissions, etc).

That said, increasing serial baudrate can help reduce communication latency between host and RCP processors. I would recommend increasing the baudrate to 460800 if you can.

HeFeng1947, 2022-01-05:
> @HeFeng1947 , thanks for submitting this issue.
>
> While the IEEE 802.15.4 2.4 GHz O-QPSK PHY supports 250 kbps, the actual data rate in practice is typically lower due to various overheads (e.g. MAC [...]

Thank you for your reply! Your experience points have been invaluable to our OTBR development.
My concern mainly comes from the fact that in extreme conditions (for example, when multiple terminals need to transmit big data to the outside world at the same time), the baudrate of the serial port may cause data loss due to timely synchronization failure. 
Currently in my tests, some chips from well-known manufacturers do not support high baudrates. This performance bottleneck may not be well understood in the industry. Would OpenThread consider doing some testing and providing some theoretical values?

jwhui, 2022-01-05:
> Currently in my tests, some chips from well-known manufacturers do not support high baudrates.

Which chips are you working with? A number of modern 15.4 SoCs are capable of supporting 460800. That said, if SPI or native USB is an option, that would be even better.

> This performance bottleneck may not be well understood in the industry. Would OpenThread consider doing some testing and providing some theoretical values?

We've already had a number of reports indicating that 460800 resolves a number of issues with RCP. Some examples:
- https://github.com/openthread/ot-br-posix/issues/559#issuecomment-720614330
- https://groups.google.com/g/openthread-users/c/t08i-B6Iqwo/m/ZCMQtc34DAAJ
Again, the issue is not because of the raw 15.4 PHY data rate, but more because of the spinel serial protocol.

HeFeng1947, 2022-01-05:
> > Currently in my tests, some chips from well-known manufacturers do not support high baudrates.
> 
> Which chips are you working with? A number of modern 15.4 SoCs are capable of supporting 460800. That said, if SPI or native USB is an option, that would be even better.

**silabs - EFR32MG21** I have tested and communicated with FAE, and the highest baud rate can only reach 230400

jwhui, 2022-01-05:
> > Which chips are you working with? A number of modern 15.4 SoCs are capable of supporting 460800. That said, if SPI or native USB is an option, that would be even better.
> 
> **silabs - EFR32MG21** I have tested and communicated with FAE, and the highest baud rate can only reach 230400

FWIW, the [Silicon Labs codelab](https://openthread.io/codelabs/silabs-openthread-hardware#6) specifies 460800 baud rate when using BRD4166A, which includes an EFR32MG12 instead of EFR32MG21. And https://github.com/openthread/ot-br-posix/issues/559#issuecomment-720614330 was a comment from Silicon Labs.

@lmnotran @romacdon , can you help comment on EFR32MG21's support for 460800 baud rate?

HeFeng1947, 2022-01-05:
Unfortunately, due to the lack of core, we can't purchase the 1 series chips of Silabs. efr32mg21 seems to be the only option for the silabs solution at present.

HeFeng1947, 2022-01-05:
I would like to correct that the 230400 limit is derived from the table in silabs reference manual. Maybe I was wrong about what it meant. 
I tested the settings to 1M baud rate and RCP didn't work properly. But under the condition of less than 1M (maximum test 921600), it can work normally.

jwhui, 2022-01-05:
> I would like to correct that the 230400 limit is derived from the table in silabs reference manual. Maybe I was wrong about what it meant. I tested the settings to 1M baud rate and RCP didn't work properly. But under the condition of less than 1M (maximum test 921600), it can work normally.

Can you try with 460800?

HeFeng1947, 2022-01-06:
> Can you try with 460800?

yeah, it can work.

jwhui, 2022-01-06:
> > Can you try with 460800?
> 
> yeah, it can work.

@HeFeng1947 , great to hear! Is this issue resolved now?

HeFeng1947, 2022-01-07:
> > > Can you try with 460800?
> > 
> > 
> > yeah, it can work.
> 
> @HeFeng1947 , great to hear! Is this issue resolved now?

yeah， I think this issue could be closed now.

closed 2022-01-07 by HeFeng1947