openthread/openthread #7291: How to send an Anycast message

OlivierGre, 2022-01-06:
Anycast addresses are described on https://openthread.io/guides/thread-primer/ipv6-addressing

I have a Thread device freshly connected to a Thread network thanks to an Android phone commissioner. I would like to send an Anycast message to the Commissioner. It is said to use ALOC16 0xfc30 – 0xfc37 for the Commissioner but I don't know how to use it.
Would you have indications about how to send such Anycast message to the Commissioner?

Thank you

OlivierGre, 2022-01-07:
Hi Jonathan,
Thank you for your message.

In my case, the Commissioner is an Android app and doesn't have a CLI. Maybe that my first question was not clear enough: This is on the Joiner device (freshly connected to the thread network) that I want to send an anycast message to the Commissioner. Is there a way to get the Commissioner Anycast Address? I have seen this commit (https://github.com/openthread/openthread/pull/4707) which adds GetLocator(). Could it help me to retrieve the Commissioner Anycast Address? 

Thank you

jwhui, 2022-01-08:
> In my case, the Commissioner is an Android app and doesn't have a CLI. Maybe that my first question was not clear enough: This is on the Joiner device (freshly connected to the thread network) that [...]

As mentioned in https://github.com/openthread/openthread/issues/7291#issuecomment-1006761122, after attaching to the Thread network, a Thread device can determine the Commissioner Anycast Address using the Commissioner Session ID contained within the Thread Network Data.

> My Commissioner is in the Wifi network and not in the Thread network. Is it possible, in that case, to reach it with an Anycast Address?

Yes. Thread should route packets with a Commissioner Anycast destination to the Thread border router that the commissioner is attached to.

OlivierGre, 2022-01-10:
Ok, thank you Jonathan.

On the device that freshly joined the Thread network, is it possible to determine the Commissioner Anycast address? (programmatically)

> We could probably add an API to provide the Commissioner Anycast Address.

IMO, that would be useful.

OlivierGre, 2022-01-11:
> On the device that freshly joined the Thread network, is it possible to determine the Commissioner Anycast address? (programmatically)

I can explain the reason of my question:

When I have started working on Thread, I have focused on trying to echange data between the Android phone (the commissioner) and the freshly commissioned device. Thanks to the help obtained on this forum, I have been able to configure the BR and to send UDP packets between the Android phone (connected to the wifi) and the Thread device.
For the moment, I'm doing the commissioning, I then use the CLI to read the IP address matching the OMR prefix and I give this address to the Android application. It can then send UDP packet to the thread device.
I'm now looking how they could identify each other automatically. 

In another issue, you advised me to implement a SRP service. I will not get that in a short time so I was looking for an alternate solution. From the Thread device, is it possible to determine the Commissioner Anycast address? 

I have thought to a workaround that seems to work. On the thread device, I'm allocating a static IPV6 Address starting from OMR prefix and finishing by some bytes arbitrary chosen. By this way the Android phone can guess the IP Address of the Thread device. They can then talk to each other and possibly switch to the IP address dynamically allocated.
This is just for a demo (and not a real product) so that seems an acceptable workaround until I get the SRP service, isn't it? 

In Openthread Android Commissining App, is there a way to determine the OMR prefix?

Thank you

wgtdkp, 2022-01-12:
>When I have started working on Thread, I have focused on trying to echange data between the Android phone (the commissioner) and the freshly commissioned device.

You can exchange data between an Android phone and a commissioned device without a commissioner. You can use [Android socket](https://developer.android.com/reference/java/net/Socket) to send/recv IP traffics as if you are talking to a WiFi device, as long as you have the IPv6 address of the end device.

Even if you can get the Commissioner Anycast address, packets sent to this address is not routed directly to the mobile phone but to the border router and then relayed to the commissioner app via the DTLS connection between the commissioner app and the border router. That said, the packets sent from the end device will not be received by other network applications on the mobile phone but as I said, you don't need (and shouldn't use) the commissioner app to exchange data between mobile phone and Thread device.

The SRP service solution is the recommended solution to discover a specific Thread device from WiFi network. On your Android phone, you just need to subscribe to and resolve your registered SRP/mDNS service to get the IPv6 address of your Thread device with [NsdManager](https://developer.android.com/reference/android/net/nsd/NsdManager).

> I'm now looking how they could identify each other automatically.
>
>In another issue, you advised me to implement a SRP service. I will not get that in a short time so I was looking for an alternate solution

Setting up a SRP service on a Thread end device seems pretty simple per https://openthread.io/codelabs/openthread-border-router#4 and you can make a script on the end device to automate this process for demo purpose.

Could you share the specific blocker that stops you from using SRP?

OlivierGre, 2022-01-12:
> You can exchange data between an Android phone and a commissioned device without a commissioner. You can use [Android socket](https://developer.android.com/reference/java/net/Socket) to send/recv IP [...]

Yes, I have that working. I'm able to send UDP packets between the Android phone and the commissioned device. It is working both ways. I indeed need the IPV6 address of the end device. 

> Setting up a SRP service on a Thread end device seems pretty simple per https://openthread.io/codelabs/openthread-border-router#4 and you can make a script on the end device to automate this process for demo purpose.
> 
> Could you share the specific blocker that stops you from using SRP?

The HW that I use is under development and SRP is not available yet. 
The end device uses a version of the OpenThread lib that is old. 
I had the same problem with the RCP of the Border Router. I have been able to update it to the latest version and it works (I'm compiling it for Thread 1.1.1 though).
I have seen that it is not straighforward to update the openthread lib of my end device (due to dependencies with the rest of the SW). And then I should see how to configure the SRP service. I'm not sure that I will succeed to do that...