openthread/openthread #7292: Border Router Ingress Firewall feature doesn't  account for NAT64 prefixes

gabekassel, 2022-01-06:
The OTBR ingress firewall feature added in #7043 doesn't account for NAT64 prefixes added to network data as a route. 

In testing, if I hand add the prefix to the `allow` ipsets, NAT64 pinging then works.

simonlingoogle, 2022-01-07:
I wonder what the issue is. #7043 implements ingress filtering, where certain destinations and sources are not allowed for ingress traffic. 
But an NAT64 address should always be the source for ingress traffic so it should not be blocked? @gabekassel

gabekassel, 2022-01-07:
I think the issue is that we, separately, have a generic DROP rule at the end of all chains in our system. I'm trying to find a good solution. I realized I can't just add to the ipset manually since 1) there can be contention with otbr trying to "swap" the sets at the same time and 2) that otbr calls `flush` at launch

superwhd, 2022-01-10:
@gabekassel Are the NAT64 prefixes dynamic or static? If they are static, maybe you can insert a new subchain in front of `OTBR_FORWARD_INGRESS`, containing rules that accept NAT64 prefixes.

Also, I'm not very sure why the packets are processed by the `FORWARD` chain. Could you share some more details (e.g., topology, source/destination address types) about your use case?

gabekassel, 2022-01-19:
I root caused this to hop count in the ping command and nothing to do with firewall.

closed 2022-01-19 by gabekassel