openthread/openthread #7302: TCP is being built and linked unconditionally

edmont, 2022-01-10:
TCP sources are being built unconditionally here: https://github.com/openthread/openthread/blob/main/third_party/CMakeLists.txt#L33
Same for linking, i.e. here: https://github.com/openthread/openthread/blob/main/src/core/ftd.cmake#L46

Shouldn't this be conditioned to `OPENTHREAD_CONFIG_TCP_ENABLE`?

jwhui, 2022-01-10:
@edmont , I raised this question in #6885 at https://github.com/openthread/openthread/pull/6885#discussion_r683141586. See @samkumar's response in https://github.com/openthread/openthread/pull/6885#discussion_r684571465.

If you have further thoughts, feel free to share them in this issue (rather than in #6885).

jwhui, 2022-01-28:
Resolved by #7306.

closed 2022-01-28 by jwhui