openthread/openthread #7304: nanoleaf essentials

ajdezigns, 2022-01-11:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.
looking to control nanoleaf essentials in Home Assistant ( https://github.com/home-assistant/core )

**Describe the solution you'd like** A clear and concise description of what you want to happen.
this project ( https://github.com/roysjosh/nanoleaf-ltpdu ) looks like it would make that possible

ndyck14, 2022-01-12:
Its great to see the demand for this integration, although my guess is the request may get better visibility in the home-assistant repo, not OpenThread. Just so you know, we haven't published documentation for our Essentials APIs, so any integration work carriees risks that APIs may change without notice. We watch the market closely though!

roysjosh, 2022-01-13:
@ajdezigns please follow the progress here: https://community.home-assistant.io/t/homekit-accessory-protocol-hap-over-coap-udp-was-nanoleaf-essentials-bulb-via-thread-coap/335167

I hope to have a functional integration using HomeKit within a week or two. It will likely take time for HA to review as the UDP/CoAP HomeKit underlay is completely new to both the HA project and the aiohomekit project AFAICT.

Stefan-Schmidt, 2022-02-18:
@ndyck14 I have a question on Nanoleaf and OpenThread though. I have trouble to get a Nanoleaf essential on the OpenThread Border Router (OTBR). And from what I have found so far I am not the only one (https://forum.nanoleaf.me/forum/feature-requests/homekit-compatible-thread-border-router-other-than-homepod-mini).

Is this something you are testing internally? Running Nanoleaf devices against OTBR (as kind of reference implementation)?
I would, maybe naively, expect this combination to work, given both a certified components. I f you have an idea what the culprit might be and how to fix this with OTBR, I would be all ears.

ndyck14, 2022-02-28:
Yes, our products work fine on OpenThread Border Routers, although perhaps the question is how to put the devices onto a network you are creating? For now our App supports adding Essentials to Apple Thread networks (HomePod Mini / Apple TV 4th Gen) or to our own Border Routers (Shapes, Elements, Lines). For our own networks, the credentials are exposed through the App and you can use them for OTBR for additional redundancy. Although if you are looking for a solution exlusively with OTBR, this is not possible with publicly available tools today.

Stefan-Schmidt, 2022-02-28:
Yeah, the question was more targeted regarding support for the onboarding itself. Like MeshCoP support.

I also did some more testing on my side and from what I can see the essentials do not implement MeshCoP and the whole onboarding story for Nanoloeaf goes through the mobile app and links to partners. I my case I used the app /account linking to onboard an essential bulb onto an Eero Thread network. With the than exposed network details and credentials in the nanoleaf app I was able to join my OTBR into the same network.

I was, perhaps naively, expecting MeshCoP support was part of the Thread certification process and all devices would support it. Reality has me shown differently. Not sure if you can comment on that or not, but I for one would be interested if adding MeshCoP support on the nodes as well as the border routers of nanoleaf are on the roadmap and/or if this is something more planned to be handled by Matter on top (to onboard and hand over credentials to the Thread devices).

Jc2k, 2022-02-28:
@ndyck14 Coming back to Nanoleaf and Home Assistant, we are able to control Essentials devices using HomeKit over CoAP if we setup the mesh out of band (e.g. with Nanoleaf App). We should also be able to connect to Nanoleaf devices using HomeKit over IP and BLE too. For HomeKit, are you able to comment on whether it is possible to manage the mesh connection parameters with just HomeKit characteristics (e.g. the thread control point that takes tlv8 structures)? (Not neccesarily looking for guidance on how).

ndyck14, 2022-02-28:
Thread certification requires MeshCoP, and Essentials support it, although the mode is not enabled by default and not easily triggered by users for the moment unfortunately.

For HomeKit, yes everything is possible using the standardized characteristics over BLE and/or Thread.

Stefan-Schmidt, 2022-02-28:
> Thread certification requires MeshCoP, and Essentials support it, although the mode is not enabled by default and not easily triggered by users for the moment unfortunately.

I see. Happy to hear its actually there. I will educate myself in patience and hope that a future firmware update enables it by default. And thanks a lot for taking the time for community communications.

closed 2022-03-01 by jwhui

clstb, 2023-08-30:
For anyone stumbling upon this, I got an OTBR created Thread network to show in the Nanoleaf app using the following settings:
[image]