openthread/openthread #7312: leader respons with ack after calling thread stop

lmaciejonczyk, 2022-01-13:
Topology:
leader ---- SED/SSED

Git OpenThread commit:
3c260bd7c7444b9bf446429de3fe46fab09162b3
Wed Jan 12 15:25:21 2022

Reproduction steps:
1. Form the topology
2. On leader side, call: thread stop 
3. Observe that leader still respons with ack on child Data Request
4. On leader side, call: thread start
5. Try ping devices
6. Observe that the leader-child communication is lost

**Expected behavior** A clear and concise description of what you expected to happen.
After call "thread stop" the leader should not response with ack so that child reattach the network.
[pcap_thread_stop_and_still_ack.zip](https://github.com/openthread/openthread/files/7861970/pcap_thread_stop_and_still_ack.zip)

jwhui, 2022-01-13:
@lmaciejonczyk , I believe this is currently expected behavior. `thread stop` only stops Thread-related protocols, such as MLE and TMF. `ifconfig down` will actually disable the radio.

The Thread 1.1.1 specification does state the following:
```
Rx-off-when-idle Children
SHOULD also periodically (at time intervals indicated by the application) resynchronize by
sending a Child Update Request to their Parent to avoid cases when the MAC Data Request
is being acknowledged erroneously by a receiver that is not a valid Parent.
```

You can also take a look at OpenThread's [Periodic Parent Search](https://openthread.io/guides/build/features/child-supervision) feature that also addresses this issue.

lmaciejonczyk, 2022-01-14:
I've been missing it. It makes sense to me now, thanks.

closed 2022-01-14 by lmaciejonczyk