openthread/openthread #7317: otCoapSendRequest return OT_ERROR_INVALID_STATE

bhumikaajain, 2022-01-14:
Hi,
I am developing an application on an openthread protocol-based network.
In my setup, I have one Rpi-based border router and openthread nodes are built upon nRF_Connect_SDK.
Development Platform: nRF52840
SDK: nRF_Connect_SDK(v1.8.0)
My CoAP server is on the border router and the CoAP_Client node is trying to send a message to it using "otCoapSendRequest".
My request is getting failed and I am receiving error 13 (OT_ERROR_INVALID_STATE).
I have seen in the definition of "otCoapSendRequest" and among direct return errors, OT_ERROR_INVALID_STATE is not present.

Can you help me in debugging the reason for this error in the "otCoapSendRequest"

jwhui, 2022-01-14:
A quick search for `kErrorInvalidState` in `src/core/coap` shows that it
may be returned when the UDP socket is not yet bound - see
src/core/coap/coap.cpp
<https://github.com/openthread/openthread/blob/7a1c843ffd07e59b1b0ffe907cee1f9418eebdec/src/core/coap/coap.cpp#L1697>
.

Can you check whether the UDP socket is bound?

I would suggest reviewing the CoAP Quick Start
<https://github.com/openthread/openthread/blob/7a1c843ffd07e59b1b0ffe907cee1f9418eebdec/src/core/coap/coap.cpp#L1697>
guide
and ensure you are calling the APIs in the same way.

bhumikaajain, 2022-01-14:
Hi,

Can you please share the quick start guide link again?
In the above answer, both links are the same.

jwhui, 2022-01-14:
https://github.com/openthread/openthread/blob/main/src/cli/README_COAP.md#quick-start

bhumikaajain, 2022-01-20:
I have reviewed the quick start guide, and I am following all steps.
Could you suggest a way so that I can find UDP socket is bound or not?

jwhui, 2022-01-20:
> I have reviewed the quick start guide, and I am following all steps. Could you suggest a way so that I can find UDP socket is bound or not?

Have you validated that CoAP works with the CLI? If so, the CLI just makes OpenThread API calls, so you should be able to follow that.

Note that [`otCoapStart()`](https://openthread.io/reference/group/api-coap#otcoapstart) should bind the UDP socket for you: https://github.com/openthread/openthread/blob/0dccf0547348e62967cb080238ff4f6b5b1f7079/src/core/coap/coap.cpp#L1661

bhumikaajain, 2022-01-21:
@jwhui Thank you.
My issue is resolved.

closed 2022-01-21 by jwhui