openthread/openthread #7318: RadioSpinelNoResponse exception when call txpower by UBUS RPC CALL

LancerSky, 2022-01-14:
**Describe the bug** A clear and concise description of what the bug is.
Hi, I have add the txpower command to my ubus service. So I can get the txpower from my RPC interface as follow
```
else if (!strcmp(aAction, "txpower"))
{
    int8_t  power;
    otPlatRadioGetTransmitPower(mController->GetInstance(), &power);
    blobmsg_add_u32(&mBuf, "TXPower", power);
}
---------------------------
# ubus call otbr txpower
{
        "TXPower": 8,
        "Error": 0
}
```
But I got an error when I called `ubus call otbr txpower` multiple times. And the otbr-agent will restart.
Here is the exception log,
```
Fri Jan 14 09:31:22 2022 user.debug otbr-agent[17438]: 00:02:59.990 [DEBG]-PLAT----: Wait response: tid=7 key=37
Fri Jan 14 09:31:23 2022 user.debug otbr-agent[17438]: 00:03:00.116 [DEBG]-PLAT----: Trying to get RCP time offset
Fri Jan 14 09:31:23 2022 user.debug otbr-agent[17438]: 00:03:00.117 [DEBG]-PLAT----: Wait response: tid=8 key=2050
Fri Jan 14 09:31:37 2022 user.info otbr-agent[17438]: 00:03:14.197 [INFO]-MLE-----: Send Advertisement (ff02:0:0:0:0:0:0:1)
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.199 [DEBG]-MAC-----: Request to start operation "TransmitDataDirect"
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.201 [DEBG]-MAC-----: Starting operation "TransmitDataDirect"
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.203 [DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.204 [DEBG]-MAC-----: RadioState: CsmaBackoff -> Transmit
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.213 [DEBG]-MAC-----: RadioState: Transmit -> Receive
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.214 [DEBG]-MAC-----: =========================[TX seqno=132 len=069]=========================
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.214 [DEBG]-MAC-----: | 41 D8 84 07 C0 FF FF 5A | 4C 61 BF B2 09 8B AA 7F | AX..@..ZLa?2..*.
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.215 [DEBG]-MAC-----: | 3B 01 F0 4D 4C 4D 4C 43 | 19 00 15 0E 00 00 00 00 | ;.pMLMLC........
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.215 [DEBG]-MAC-----: | 00 00 00 01 D8 C8 5C 61 | 57 26 39 A9 F7 1B F2 E0 | ....XH\aW&9)w.r`
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.215 [DEBG]-MAC-----: | 2A 14 45 D7 18 2F 2D 70 | 91 96 55 DE A3 13 2C BA | *.EW./-p..U^#.,:
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.215 [DEBG]-MAC-----: | C2 92 66 00 00 .. .. .. | .. .. .. .. .. .. .. .. | B.f.............
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.215 [DEBG]-MAC-----: ------------------------------------------------------------------------
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.216 [DEBG]-MAC-----: Finishing operation "TransmitDataDirect"
Fri Jan 14 09:31:37 2022 user.info otbr-agent[17438]: 00:03:14.216 [INFO]-MAC-----: Sent IPv6 UDP msg, len:90, chksum:4319, to:0xffff, sec:no, prio:net
Fri Jan 14 09:31:37 2022 user.info otbr-agent[17438]: 00:03:14.216 [INFO]-MAC-----:     src:[fe80:0:0:0:a88b:9b2:bf61:4c5a]:19788
Fri Jan 14 09:31:37 2022 user.info otbr-agent[17438]: 00:03:14.216 [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
Fri Jan 14 09:31:37 2022 user.debug otbr-agent[17438]: 00:03:14.216 [DEBG]-MAC-----: Idle mode: Radio receiving on channel 22
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.172 [DEBG]-MAC-----: Rx security - frame counter 256
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.175 [DEBG]-MAC-----: =========================[RX seqno=198 len=078]=========================
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.177 [DEBG]-MAC-----: | 09 D8 C6 FF FF FF FF 37 | 66 08 AA 41 47 9B DB 6C | .XF....7f.*AG.[l
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.183 [DEBG]-MAC-----: | 0E 15 00 01 00 00 FF FF | FF FF FF 7F 3B 01 F0 4D | ............;.pM
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.187 [DEBG]-MAC-----: | 4C 4D 4C BD 82 00 15 FF | 02 00 00 00 00 00 00 01 | LML=............
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.191 [DEBG]-MAC-----: | A1 D9 C8 31 80 BE EC A8 | 5A C0 69 8C B9 6D 61 CA | !YH1.>l(Z@i.9maJ
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.193 [DEBG]-MAC-----: | AC 7C 9E 2D 2A 4A 62 E4 | 20 BD 51 3A 37 ED .. .. | ,|.-*Jbd =Q:7m..
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.196 [DEBG]-MAC-----: ------------------------------------------------------------------------
Fri Jan 14 09:31:43 2022 user.info otbr-agent[17438]: 00:03:20.196 [INFO]-MAC-----: Received IPv6 UDP msg, len:83, chksum:bd82, from:0e6cdb9b4741aa08, sec:yes, prio:net, rss:-37.0
Fri Jan 14 09:31:43 2022 user.info otbr-agent[17438]: 00:03:20.196 [INFO]-MAC-----:     src:[fe80:0:0:0:c6c:db9b:4741:aa08]:19788
Fri Jan 14 09:31:43 2022 user.info otbr-agent[17438]: 00:03:20.196 [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.196 [DEBG]-MLE-----: Receive UDP message
Fri Jan 14 09:31:43 2022 user.warn otbr-agent[17438]: 00:03:20.196 [WARN]-MLE-----: Failed to process UDP: Security
Fri Jan 14 09:31:43 2022 user.debug otbr-agent[17438]: 00:03:20.197 [DEBG]-MAC-----: Idle mode: Radio receiving on channel 22
Fri Jan 14 09:31:55 2022 user.debug otbr-agent[17438]: 00:03:32.778 [DEBG]-PLAT----: Wait response: tid=10 key=37
Fri Jan 14 09:31:56 2022 user.debug otbr-agent[17438]: 00:03:33.382 [DEBG]-PLAT----: Wait response: tid=11 key=37
Fri Jan 14 09:32:04 2022 user.debug otbr-agent[17438]: 00:03:41.066 [DEBG]-PLAT----: Wait response: tid=12 key=37
Fri Jan 14 09:32:04 2022 user.debug otbr-agent[17438]: 00:03:41.546 [DEBG]-PLAT----: Wait response: tid=13 key=37
Fri Jan 14 09:32:04 2022 user.debug otbr-agent[17438]: 00:03:42.022 [DEBG]-PLAT----: Wait response: tid=14 key=37
Fri Jan 14 09:32:05 2022 user.debug otbr-agent[17438]: 00:03:42.778 [DEBG]-PLAT----: Wait response: tid=15 key=37
Fri Jan 14 09:32:07 2022 user.crit otbr-agent[17438]: 00:03:44.785 [CRIT]-PLAT----: HandleRcpTimeout() at ot-br-posix-2022-1-5/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2218: RadioSpinelNoResponse
Fri Jan 14 09:32:12 2022 user.info otbr-agent[20095]: [INFO]-UTILS---: Running thread-reference-20200818-1424-ge736d2488
Fri Jan 14 09:32:12 2022 user.info otbr-agent[20095]: [INFO]-UTILS---: Thread version: 1.2.0
Fri Jan 14 09:32:12 2022 user.info otbr-agent[20095]: [INFO]-UTILS---: Thread interface: wpan0
Fri Jan 14 09:32:12 2022 user.info otbr-agent[20095]: [INFO]-UTILS---: Backbone interface:
Fri Jan 14 09:32:12 2022 user.info otbr-agent[20095]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0?uart-baudrate=460800
```
**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
I use the commit https://github.com/openthread/ot-br-posix/commit/a965a7535dad33303585d2080437f3d796e54a81
I have found a normal commit https://github.com/openthread/ot-br-posix/commit/ae90b332891874df4200cfdf27f082d05a2cf7c7
But failed after https://github.com/openthread/ot-br-posix/commit/92059387fc5bc253cda65a75293d511b58383b38
2. IEEE 802.15.4 hardware platform
nRF52840USB Dongle
3. Build steps
add compile option `-DOTBR_OPENWRT=ON`
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

jwhui, 2022-01-18:
@LancerSky , thanks for raising this issue.

Given that you are using an nRF52840 USB Dongle as your RCP, I wonder if this may be the same issue as https://github.com/openthread/ot-nrf528xx/issues/237.

/cc/ @MarekPorwisz

LancerSky, 2022-01-19:
Hi Jonathan，
Here is my patch in OpenWrt.
[0001-openwrt-add-txpower-command.patch.txt](https://github.com/openthread/openthread/files/7896314/0001-openwrt-add-txpower-command.patch.txt)

For testing the problem, I use the following shell command,
```
for i in $(seq 1 100)
do
	ubus call otbr txpower
	echo "count" $i
	sleep 1
done
```
It will failed after call txpower by ubus in many times.
But no problem with ot-ctl,
```
for i in $(seq 1 100)
do
	ot-ctl txpower
	echo "count" $i
	sleep 1
done
```
As I described above,
I use the commit openthread/ot-br-posix@a965a75 for testing, and it failed.
I found a normal commit openthread/ot-br-posix@ae90b33,
But failed after openthread/ot-br-posix@9205938
Thanks.

jwhui, 2022-01-19:
> As I described above, I use the commit [openthread/ot-br-posix@a965a75](https://github.com/openthread/ot-br-posix/commit/a965a75) for testing, and it failed. I found a normal commit [...]

@LancerSky , can you clarify what you mean by "normal commit"? Are you saying that the failure does not occur with https://github.com/openthread/ot-br-posix/commit/ae90b33, but the failure does occur with https://github.com/openthread/ot-br-posix/commit/9205938?

LancerSky, 2022-01-20:
Yeah! The failure does occur with openthread/ot-br-posix@9205938 and failure after this commit.

jwhui, 2022-01-25:
> Yeah! The failure does occur with [openthread/ot-br-posix@9205938](https://github.com/openthread/ot-br-posix/commit/9205938) and failure after this commit.

@LancerSky , thanks for clarifying.

@zhanglongxia , any thoughts on why https://github.com/openthread/ot-br-posix/commit/9205938 may be causing this issue?

LancerSky, 2022-05-09:
Hi, @jwhui ,
I think it's a task scheduling problem. I try to modify the following code and test again. It work great.
```
index 7e5f1bf0bd..760a6baca9 100644
--- a/src/agent/application.cpp
+++ b/src/agent/application.cpp
@@ -147,14 +147,18 @@ otbrError Application::Run(void)
         FD_ZERO(&mainloop.mWriteFdSet);
         FD_ZERO(&mainloop.mErrorFdSet);
 
-        MainloopManager::GetInstance().Update(mainloop);
+        // MainloopManager::GetInstance().Update(mainloop);
+        mNcp.Update(mainloop);
+        mUbusAgent.Update(mainloop);
 
         rval = select(mainloop.mMaxFd + 1, &mainloop.mReadFdSet, &mainloop.mWriteFdSet, &mainloop.mErrorFdSet,
                       &mainloop.mTimeout);
 
         if (rval >= 0)
         {
-            MainloopManager::GetInstance().Process(mainloop);
+            // MainloopManager::GetInstance().Process(mainloop);
+            mUbusAgent.Update(mainloop);
+            mNcp.Update(mainloop);
         }
         else if (errno != EINTR)
         {
```

LancerSky, 2022-05-09:
please ignore my message, there are other issues as well.

jwhui, 2022-09-13:
Closing stale issue.

closed 2022-09-13 by jwhui