openthread/openthread #7322: POSIX build failing

j-noel-longhand, 2022-01-14:
**Describe the bug** A clear and concise description of what the bug is.

The POSIX-compliant build fails.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id

    Using source as of commit 169fec2be49d1018a94be6742834edaa78550dab
    (have not tried earlier to find when this broke).

2. IEEE 802.15.4 hardware platform

     n/a

3. Build steps

     make -f src/posix/Makefile-posix DAEMON=1

4. Network topology

       n/a

**Expected behavior** A clear and concise description of what you expected to happen.

    Successful build suitable for use in the 'Simulating a Thread network with OpenThread' (aka 'Simulation with Toolchain') Codelab

**Console/log output** If applicable, add console/log output to help explain your problem.

```
                /home/jdonald/devwork/thread/openthread/src/posix/../../src/posix/platform/netif.cpp: In function ‘otError DeleteExternalRoute(const otIp6Prefix&)’:
                /home/jdonald/devwork/thread/openthread/src/posix/../../src/posix/platform/netif.cpp:609:58: error: ‘NLM_F_NONREC’ was not declared in this scope
                     req.header.nlmsg_flags = NLM_F_REQUEST | NLM_F_ACK | NLM_F_NONREC;
                                                                          ^~~~~~~~~~~~
                  CXX      libopenthread_posix_a-virtual_time.o
                Makefile:1021: recipe for target 'libopenthread_posix_a-netif.o' failed
                make[5]: *** [libopenthread_posix_a-netif.o] Error 1
                make[5]: *** Waiting for unfinished jobs....
                Makefile:812: recipe for target 'all-recursive' failed
                make[4]: *** [all-recursive] Error 1
                Makefile:434: recipe for target 'all-recursive' failed
                make[3]: *** [all-recursive] Error 1
                Makefile:587: recipe for target 'all-recursive' failed
                make[2]: *** [all-recursive] Error 1
                Makefile:513: recipe for target 'all' failed
                make[1]: *** [all] Error 2
                src/posix/Makefile-posix:326: recipe for target 'do-build-posix' failed
                make: *** [do-build-posix] Error 2

```

**Additional context** Add any other context about the problem here.

Using standard build tools on Debian Stretch running on Lenovo Intel Corei5 x64 hardware.

superwhd, 2022-01-17:
@j-noel-longhand Could you share the output of the following commands? I suspect the version of `libnl` or the kernel is not high enough.
```bash
/sbin/ldconfig -p -v | grep libnl
uname -a
```

j-noel-longhand, 2022-01-17:
Sure thing:

```
➜   /sbin/ldconfig -p -v | grep libnl
/sbin/ldconfig	libnl-3.so.200 (libc6,x86-64) => /lib/x86_64-linux-gnu/libnl-3.so.200
	libnl-route-3.so.200 (libc6,x86-64) => /usr/lib/x86_64-linux-gnu/libnl-route-3.so.200
	libnl-genl-3.so.200 (libc6,x86-64) => /lib/x86_64-linux-gnu/libnl-genl-3.so.200
➜   uname -a
unameLinux parfmaen 5.10.0-10-amd64 #1 SMP Debian 5.10.84-1 (2021-12-08) x86_64 GNU/Linux
```

j-noel-longhand, 2022-01-17:
I just ran a full system update and did a fresh pull and posix build and it worked, no errors. Thank you for looking into this issue, I am closing it now.