openthread/openthread #7333: [github-actions] ot-commissioner workflow fails intermittently

simonlingoogle, 2022-01-19:
[logs_111023.zip](https://github.com/openthread/openthread/files/7896762/logs_111023.zip)

wgtdkp, 2022-01-19:
Looks like there is something broken with the new Commissioner enhancement tests. But not clear why it starts fail now, after running stably for nearly half a year:
```
2022-01-19T11:13:40.8845002Z ++ grep -q '1111111111111111": false'
2022-01-19T11:13:40.8845330Z ++ send_command_to_commissioner 'sessionid --nwk other' 'DEAD00BEEF00CAFE": \d*'
2022-01-19T11:13:40.8845621Z ++ local 'command=sessionid --nwk other'
2022-01-19T11:13:40.8845821Z ++ local expect_error=
2022-01-19T11:13:40.8846184Z ++ '[' 2 == 2 ']'
2022-01-19T11:13:40.8846423Z ++ expect_error='DEAD00BEEF00CAFE": \d*'
2022-01-19T11:13:40.8846597Z ++ local result
2022-01-19T11:13:40.8846890Z +++ /usr/local/bin/commissioner_ctl.py execute 'sessionid --nwk other'
2022-01-19T11:13:40.8847148Z ++ result='[32mnull
2022-01-19T11:13:40.8847325Z [done][0m
2022-01-19T11:13:40.8847479Z thread2'
2022-01-19T11:13:40.8847654Z ++ echo '[32mnull
2022-01-19T11:13:40.8847842Z [done][0m
2022-01-19T11:13:40.8847980Z thread2'
2022-01-19T11:13:40.8848160Z [32mnull
2022-01-19T11:13:40.8848329Z [done][0m
2022-01-19T11:13:40.8848483Z thread2
2022-01-19T11:13:40.8848670Z ++ '[' -n '[32mnull
2022-01-19T11:13:40.8848842Z [done][0m
2022-01-19T11:13:40.8849026Z thread2' ']'
2022-01-19T11:13:40.8849239Z ++ '[' -z 'DEAD00BEEF00CAFE": \d*' ']'
2022-01-19T11:13:40.8849455Z ++ echo '[32mnull
2022-01-19T11:13:40.8849805Z [done][0m
2022-01-19T11:13:40.8849964Z thread2'
2022-01-19T11:13:40.8850185Z ++ grep -q 'DEAD00BEEF00CAFE": \d*'
2022-01-19T11:13:40.8850387Z ++ return 1
```

Will take a closer look tomorrow.

simonlingoogle, 2022-01-19:
> Looks like there is something broken with the new Commissioner enhancement tests. But not clear why it starts fail now, after running stably for nearly half a year:
>
> ``` [...]

I think it has been unstable for quite a while.

jwhui, 2022-01-19:
Duplicate of #6616

closed 2022-01-19 by jwhui