openthread/openthread #7339: CoAP: Setter for the message code

wohdo, 2022-01-20:
There is a `otCoapMessageGetCode` for getting the status code, but setting it is only possible when initializing the response message.

To change just the code, a little helper would be nice to have. Something like

```C
/**
 * This function sets the Code value.
 *
 * @param[in]  aMessage  A pointer to the CoAP message.
 * @param[in] code CoAP message code.
 */
void otCoapMessageSetCode(const otMessage *aMessage, otCoapCode code);
```

jwhui, 2022-01-20:
We typically use [`otCoapMessageInit()`](https://openthread.io/reference/group/api-coap#otcoapmessageinit) and [`otCoapMessageInitResponse()`](https://openthread.io/reference/group/api-coap#otcoapmessageinitresponse) to set the CoAP message code. We haven't had a use case for changing the CoAP message code after that so far.

Can you provide your use case for needing to change the CoAP message code?

wohdo, 2022-01-21:
There is no strong necessity for having it.

It however makes it more ergonomic to pass the Message around with using the Code to track the status or add it in one of the sequential function calls, while already having a Message to work with, e.g. to add append the payload. 
For my usage, the response handler would query various sub-modules to add some TLVs to the payload, while using the status to signal issues with it, but I would like to send the payload either way.
 
Having the option to have a symmetrical access to the status seemed generally nice to provide.  

Would you accept a PR providing this? Or does this fall under a special, not intended usage?

closed 2022-01-27 by jwhui