openthread/openthread #7359: Router stores 'valid' link to (former-parent) router even though MLE link was never established

EskoDijk, 2022-01-26:
**Describe the bug** 
During OT-NS simulations I've seen cases where a Router A stores a link as 'valid' to a peer Router B. However there was never a  MLE Link Request / Link Accept process successfully  completed between A and B. 

Topology: (L = Leader)
```
A  <--->  B <---> L
```

This case happens when  A initially attaches as child of B, and then upgrades to router role. Then A stores a 'valid' router link state in its router table per below code:

https://github.com/openthread/openthread/blob/main/src/core/thread/mle_router.cpp#L3768
```
    // Keep link to the parent in order to respond to Parent Requests before new link is established.
    *router = mParent;
    router->SetState(Neighbor::kStateValid);
    router->SetNextHop(kInvalidRouterId);
    router->SetCost(0);
```

Normally the above is fine when the new MLE link between A and B is established shortly after this above code is executed. However in some particular cases this MLE link is never established - see below. That implies that A will indefinitely consider itself "connected" to B and thus to the Thread Network Partition, while in reality it is unconnected because B does not consider itself linked to A. B stores 'invalid' as link state to A as long as there has not yet been an MLE connection setup that was completed.

Reasons the MLE link is never established:
* B rejects the initial MLE Link Request of A because it doesn't consider A yet to be a Router. The MLE leader data update from L to B takes some time to propagate and A sends its link request already before that. So at the time that B receives the link request from A, the RLOC of A isn't present yet in the router table of B.
* Also subsequent MLE Advertisements of A or B do not trigger the link setup if the link margin between A and B is below OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN defined in https://github.com/openthread/openthread/blob/main/src/core/config/mle.h#L203. Then both A and B won't initiate a next link request.

So aboveleads to the behaviour that node A remains forever part of the Partition, without having any usable links to other Partition nodes. Once in a while the MLE 'network id timeout' value never reaches >= 120, in which case loss of connectivity to Leader is detected, but then A attaches again as child to B, it immediately upgrades to Router, and the same issue starts again.

**To Reproduce** 
FYI below the simplest way to reproduce.  It's easier to see what happens with -DOT_FULL_LOGS=ON however that can only work in a modified OT-NS (not included here, a PR for the future ;) )

1. Git commit id: used 137c40136cab33eaa32066223f0a1b4838ed225d with minor delta's applied to enable low-link-margin simulation; see branch https://github.com/EskoDijk/openthread/tree/low-link-margin-simulation for the changes. 

2. OT-NS with simulation platform is used with above (branch) node code for ot-cli-ftd.

3. Build steps: 

```
./script/cmake-build simulation -DOT_PLATFORM=simulation -DOT_OTNS=ON -DOT_SIMULATION_VIRTUAL_TIME=ON -DOT_SIMULATION_VIRTUAL_TIME_UART=ON \
 -DOT_SIMULATION_MAX_NETWORK_SIZE=999 -DOT_COMMISSIONER=ON -DOT_JOINER=ON -DOT_BORDER_ROUTER=ON -DOT_SERVICE=ON -DOT_COAP=ON \
 -DOT_FULL_LOGS=OFF
```

4. Network topology

Topology: (L = Leader)
```
A  <--->  B <---> L
```

In OT-NS, first place one node L and let it become Leader. Then place B and let it become Router. Then place A so it can connect only to B. A becomes child of B, then upgrades, sends link request to B, B does not answer, then A's routing table lists a valid link to B while B does not list a valid link to A in its routing table.

**Expected behavior** 
What is expected:  one or more of below for Router A
*  It should consider the link as 'invalid' as long as the MLE router link procedure between A and B is not finished yet. See below in 'solutions'.
* It  could retry the initial MLE link request when it knows for sure that its former Parent B has received the latest assigned Router IDs information from the Leader. This seems easiest if triggered by an MLE Advertisement from B, even in low-link-margin conditions - see below in 'solutions'.
* Or, perhaps it should not upgrade to Router in the first place if its link margin is below OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN and if it sees no other Routers nearby to make links to. In such case staying a Child is better.
* Or, perhaps if it  has zero links valid and no MLE Advertisements with link margin above OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN  come in, then the Router A should be less picky and just select any Router to make a link to.
* Or, maybe it should become Leader of a new Partition since it knows B is too far way ( link margin too low i.e. < OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN) to make a Router link to it.

**Possible solutions **
One possible solution could be to set the state of router 'Peer' in the table, i.e. of the former parent, to invalid like this:
```
    router->SetState(Neighbor::kStateInValid);
```

or a special new state could be introduced to be able to handle any remaining Parent's Parent-Request messages before new link is established:
```
    router->SetState(Neighbor::kStateFormerParent);
```
this state should then not persist indefinitely. E.g. after some timeout it's changed.

Another solution is to let Router A not apply the link margin threshold OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN in case it has 0 actual links to Routers. That is, for the first MLE link to be established to another Router it won't apply this threshold. And optionally in case an (N+1)-th Router candidate is found and it only has space for N links, it could drop any existing link with link margin < OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN.

A quick solution is to lower OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN, e.g. to 3.

**Console/log output** If applicable, add console/log output to help explain your problem.
Can be included if desired (including all OT node debug logs / Wireshark traces)

**Additional context** 
To consider for automated testing: various cases are only exhibited when different than default RSSI values are used in the simulation platform. Adding RSSI capability to OT-NS is already in progress: https://github.com/openthread/ot-ns/issues/169

jwhui, 2022-01-26:
@EskoDijk , thanks for raising this issue.

I think you're on to something here. However, I think the issue is more fundamental and extends beyond the specific scenario you highlighted above.

Wireless communication links are time varying and can be asymmetric. In the case of two routers that already have established links with each other, it is possible for one router to invalidate that link and result in the same issue raised above.

Thread's current solution to this is that the Leader monitors the route cost to each active router. If the route cost becomes infinite, then the leader will remove the router after some timeout. You can observe this behavior in your reproducer scenario, where the child will become a router and then reattach as a child again after receiving indication that its router id has been revoked. Unfortunately, the device will continuously switch between child and router if the link remains asymmetric. An undesirable behavior we should address.

I like the suggestion of not upgrading to a router if it does not currently see any neighboring routers that satisfy the minimum link margin requirement. In scenarios where the router needs to upgrade to provide additional connectivity, it means we may orphan some devices. However, I don't think it's good for routers to come and go due to flaky links.

Thoughts?

simonlingoogle, 2022-01-27:
> I like the suggestion of not upgrading to a router if it does not currently see any neighboring routers that satisfy the minimum link margin requirement. 

+1 

Does it also mean the device won't respond to a Parent Request for REED?

simonlingoogle, 2022-01-27:
> * It  could retry the initial MLE link request when it knows for sure that its former Parent B has received the latest assigned Router IDs information from the Leader. This seems easiest if triggered by an MLE Advertisement from B, even in low-link-margin conditions - see below in 'solutions'.

+1 

This seems a long-standing issue that the initial Link Request failed to establish the expected links. Currently, we are calling `SendLinkRequest` right in `HandleAddressSolicitResponse` which appears to be too early. 
I think we should delay the initial Link Request to allow neighbors to learn the new Router ID from Leader.
The new Router can send Link Request if any one of the conditions below is satisfied:
- 5 seconds has elapsed since `HandleAddressSolicitResponse`
- The parent (or any other Router) has triggered MLE Advertisement containing the new Router ID

Thoughts?

EskoDijk, 2022-01-27:
@simonlingoogle Maybe the 'fast' initial Link Request is not that bad - it will work in case a nearby recipient is a Leader. But trying again after 5 seconds could be added to that. (Task for that could be scheduled)

@jwhui Thanks for the additional context of asymmetric links. Indeed the node A at some point gets the 'router id sequence' timeout. 

Not upgrading to Router seems indeed like the best strategy for A, if no other Routers seen >= OPENTHREAD_CONFIG_MLE_LINK_REQUEST_MARGIN_MIN.   (And that also means not being available for other nodes that need a Parent, ignoring the parent request for REEDs.)

simonlingoogle, 2022-01-28:
I did some experiment on this issue using the same topology to see what is the possible enhancements that can make the initial link request more effective:
1. I tried `SendAdvertisement` on A before sending `Link Request`. The purpose was to allow the latest Route TLV be advertised by A to B. However, it didn't work because B rejects to process the Advertisement. 
2. I tried `SendAdvertisement` on L before sending `AddressSolicitResponse`. The purpose was to allow the latest Route TLV be advertised by L to B, so that B can handle the initial Link Request from A. It did work and A established a Link to B within the initial Link Request. 

I am not sure if `2` is a good idea. Obviously any change to MLE behavior should be carefully weighted. Without 2, only the Leader is able to accept initial Link Requests. With `2`, Leader and its neighbor Routers are able to accept initial Link Requests.

simonlingoogle, 2022-01-28:
> But trying again after 5 seconds could be added to that. (Task for that could be scheduled)

The problem I can see with this strategy is that the Router should have already established links, triggered by MLE Advertisements, after 5 seconds. So, sending multicast Link Request later may not be that useful by then.

EskoDijk, 2022-01-28:
> The problem I can see with this strategy is that the Router should have already established links, triggered by MLE Advertisements, after 5 seconds. So, sending multicast Link Request later may not be that useful by then.

Agree, it's only useful I think if there are no established links yet 5 seconds after having upgraded to Router role. If not doing this, then waiting until the next MLE advertisement after having upgraded is the alternative (not sure how long this would take on average - it's topology-dependent also.)

simonlingoogle, 2022-01-28:
> Agree, it's only useful I think if there are no established links yet 5 seconds after having upgraded to Router role. If not doing this, then waiting until the next MLE advertisement after having [...]

Whenever a new Router is added to the Router ID set, the trickle timer for triggering MLE Advertisements will be reset so the next MLE advertisement will be triggered within 1 second for Leader and A, but later for other Routers since they need to reset the trickle timer too after receiving new MLE Advertisements.

I think strategy `2` of https://github.com/openthread/openthread/issues/7359#issuecomment-1023802954 effectively reduce the delay of the first MLE advertisement from <1s to 0s on Leader, which should be a mild change. It should be helpful for many home nets where MLE Advertisement of Leader can be received by most other Routers within the house. 

However, this strategy is not effective if the network is large and the neighbor Routers of the new Router can't receive MLE Advertisement from the Leader. Another strategy (say strategy `3`) that might be useful is that the new Router can delay the multicast Link Request until it receives the first updated MLE Advertisement from neighbor Routers. When the new Router receives the MLE advertisement and decides to send the first Link Request (after becoming Router), it sends multicast Link Request instead of unicast Link Request(current behavior). 

Combining strategy `2` and `3` may also be helpful.

Thoughts? @EskoDijk @jwhui

jwhui, 2022-05-24:
@EskoDijk @simonlingoogle , I submitted #7745 to implement (2) and (3) above.

simonlingoogle, 2022-05-25:
Does https://github.com/openthread/openthread/pull/7745 implement (2) ?

I think (2) is 

```
I tried SendAdvertisement on L before sending AddressSolicitResponse. 
The purpose was to allow the latest Route TLV be advertised by L to B, 
so that B can handle the initial Link Request from A. 
It did work and A established a Link to B within the initial Link Request.
```

jwhui, 2022-05-25:
Yeah, I was confused on (2) and (3) - I typically prefer not to use indexes to refer to previous text in these forums.

What was implemented is what was proposed in https://github.com/openthread/openthread/issues/7359#issuecomment-1022814840:

- 5 seconds has elapsed since HandleAddressSolicitResponse
- The parent (or any other Router) has triggered MLE Advertisement containing the new Router ID

simonlingoogle, 2022-05-25:
Thanks for the clarification. I think implementing https://github.com/openthread/openthread/issues/7359#issuecomment-1022814840 should be good enough. 

--- 

https://github.com/openthread/openthread/issues/7359#issuecomment-1023802954 suggested to `SendAdvertisesment` before sending `AddressSolicitResponse`, which further reduces the delay when the new Router can receive the updated RouterIdSet. But it may not be worth the complexity.

jwhui, 2022-06-17:
Resolved by #7745 and #7765.

closed 2022-06-17 by jwhui