openthread/openthread #7366: SEED gets stuck with wrong RLOC address

TPolzer, 2022-01-27:
I'm running on nrf52840 dongles, built from openthread commit ac49c91d091ea3627f18831307ffd63451762edf (but I was having the same issue with a ~1 year older openthread version previously).

I have some sleepy end devices that need to be reset regularly, otherwise they get stuck in a state without connectivity.
Looking at the cli state (while stuck/unreachable), there seems to be some confusion about its own Rloc:
```
> version
OPENTHREAD/thread-reference-20200818-1458-g10f618064; NRF52840; Jan 24 2022 23:42:41
Done
> state
child
Done
> parent info
Ext Addr: 0641c47a4faa91c6
Rloc: 5000
Link Quality In: 3
Link Quality Out: 3
Age: 128
Done
> ipaddr
fdbf:e492:40f3:de5a:0:ff:fe00:d403
2a02:**secret**
fdbf:e492:40f3:de5a:10ca:4911:ba0e:25d6
fe80:0:0:0:7455:cb77:8391:7ed9
Done
> mode
-
Done
> pollperiod
10000
Done
> parent info
Ext Addr: 0641c47a4faa91c6
Rloc: 5000
Link Quality In: 3
Link Quality Out: 3
Age: 40
Done
```
It's clearly keeping some communication with its "parent", but the Rloc based IPv6 address doesn't match its parent's Rloc.  
Routers with the same openthread build don't have this issue.

jwhui, 2022-01-27:
@TPolzer , thanks for raising this issue. I'm not yet sure what could cause this issue.

Can you provide the output of the `rloc16` CLI command?

TPolzer, 2022-02-17:
It took a while for me to reproduce this, got bit by https://github.com/openthread/ot-nrf528xx/issues/237 inbetween.

One device is now stuck again:
- Mesh Local Prefix: fdbf:e492:40f3:de5a::/64
- Mesh local address: fdbf:e492:40f3:de5a:0:ff:fe00:100b
- ```
  > rloc16
  100b
  Done
  ```
- ```
  > parent info
  Ext Addr: 9ea1180198677fef
  Rloc: 9000
  Link Quality In: 3
  Link Quality Out: 3
  Age: 197
  Done
  ```

simonlingoogle, 2022-02-17:
How do you reproduce this bug? What's the topology?

TPolzer, 2022-02-17:
I reproduce this by waiting, it seems to happen in bursts (might be me imagining patterns in a random distribution though).

The topology is ~10 routers, nearly fully meshed (-> max distance 2 hops), and a few child nodes, 2x rx always on, 2x pollperiod of 3s and 2x pollperiod of 10s.

This specific node where I can observe the issue right now has a pollperiod of 10s, but I've seen it happen to all the sleepy end devices.

simonlingoogle, 2022-02-17:
I tried to reproduce this issue with mleid_connectivity.py stress test of OTNS. 
And the test did fail. 
I found that Parents could send a `Child Id Response` with inconsistent RLOC16 and Child RLOC16.

```
Mesh Link Establishment
    Security Suite: 802.15.4 Security (0x00)
    Auxiliary Security Header
    Command: Child ID Response (12)
    TLV (Source Address = 94:00)
        Type: Source Address (0)
        Length: 2
        Address: 9400
    TLV (Leader Data)
        Type: Leader Data (11)
        Length: 8
        Partition ID: 0x7f8f3d3f
        Weighting: 64
        Data Version: 47
        Stable Data Version: 97
        Leader Router ID: 8
    TLV (Active Timestamp)
        Type: Active Timestamp (22)
        Length: 8
        Active Timestamp: Jan  1, 1970 08:00:00.000000000 CST
    TLV (Address16 = 98:b1)
        Type: Address16 (10)
        Length: 2
        Address16: 98b1
    TLV (Network Data)
        Type: Network Data (12)
        Length: 6
        Thread Network Data
    TLV (Route64)
        Type: Route64 (9)
        Length: 26
        ID Sequence: 213
        Assigned Router ID Mask: 55c8018204890288
        Routing Table Entry: 0x00 (1)
        Routing Table Entry: 0x00 (3)
        Routing Table Entry: 0x00 (5)
        Routing Table Entry: 0xf1 (7)
        Routing Table Entry: 0x03 (8)
        Routing Table Entry: 0x00 (9)
        Routing Table Entry: 0x00 (12)
        Routing Table Entry: 0x00 (23)
        Routing Table Entry: 0x00 (24)
        Routing Table Entry: 0x00 (30)
        Routing Table Entry: 0x01 (37)
        Routing Table Entry: 0x00 (40)
        Routing Table Entry: 0x00 (44)
        Routing Table Entry: 0x00 (47)
        Routing Table Entry: 0x00 (54)
        Routing Table Entry: 0x00 (56)
        Routing Table Entry: 0x00 (60)
```

So, the parent seems to be using 0x9400 but it gave the Child RLOC16 0x98b1. 

Did a brief look at the `SendChildIdResonse` code and seems the Parent always use `Get<Mac::Mac>().GetShortAddress()` to get the RLOC16 for both SourceAddressTLV and Child RLOC16. 

So, not sure why it happened, but maybe this can be a breakthrough to see why a Parent could send such an inconsistent packet.

simonlingoogle, 2022-02-17:
BTW, the child is actually a FTD

simonlingoogle, 2022-02-17:
Some observation for this issue:
- The parent was a REED and just became Router because of the ChildIdRequest from the Child
- The parent can find the Child in ChildTable, but the Child's RLOC16 was mismatched with Router ID
- When the parent became Router after ChildIdRequest, the Child was in `kStateChildIdRequest` with incorrect RLOC16, but parent failed to fix the RLOC16, neither does the parent remove the Child in `SetStateRouter`.
- When the parent send ChildIdResponse, it didn't check if the Child's RLOC16 matches Router ID or not, but simply use it without check. 

One possible fix may be iterating all Child in `kStateChildIdRequest` and change RLOC16 to 0 if it mismatches the Router ID. 

But it's not totally clear why such Child exist in the first place. Maybe it was an old Child when the Parent was using another Router ID. 

Thoughts? @jwhui @abtink

jwhui, 2022-02-18:
> To be safer, I think we should update the code to always remove children whenever there is a router ID mismatch.

After thinking about this some more, we probably don't want to just remove the child if it is in `kStateChildIdRequest`. Setting the child's RLOC16 value to zero in this case is probably better.

Thoughts?

simonlingoogle, 2022-02-18:
> After thinking about this some more, we probably don't want to just remove the child if it is in `kStateChildIdRequest`. 

Agree. 

> Setting the child's RLOC16 value to zero in this case is probably better.

I have actually verified that such a change can fix the issue. 

We can also fix mismatched RLOC16 in `SendChildIdResponse` when the Child's state becomes `Valid`, such that the invariant is simpler: Children in `Valid` or `Restored` state should always have matching RLOC16, but not so for Children in other states. 

I found that a Child can enter `kStateChildIdRequest` from invalid states in `HandleChildIdRequest`, so we can't guarantee that the Child RLOC16 will match the REED's future RLOC16. I am not sure if we can also have such issue for Router when sending ChildIdResponse. 

Thoughts? 

--- 
It may also be a good idea for Children to ignore such inconsistent `ChildIdResponse`?

jwhui, 2022-02-18:
@simonlingoogle , I think all of these fixes are worth considering. That said, do we have a definitive root cause for the issue?

For example, I can see from the code that the following is possible:
1. REED has a child table entry in restoring state corresponding to neighbor X.
2. REED receives a Child ID Request from neighbor X.
3. REED upgrades to a router and does not remove the child table entry for neighbor X because it is in the Child ID Request state.
4. Router now sends a Child ID Response with the wrong RLOC16.

Can you say whether the above is what's happening in your reproducer test? If not, can we pin down the exact scenario that is causing the Router IDs to be out of sync?

abtink, 2022-02-18:
Thanks @TPolzer for reporting this and thanks @simonlingoogle for finding the cause.

Setting rloc16 of all children in `kStateChildIdRequest` to zero from `SetStateRouter()` can work. But I agre it may be better/easier to check for this in `SendChildIdResponse()` here:
https://github.com/openthread/openthread/blob/564982c818666787096e5ad7a81a2d2f036aba36/src/core/thread/mle_router.cpp#L3081-L3086

Replacing the `if` with 
```c++
    if ((aChild.GetRloc16() == 0) || !RouterIdMatch(aChild.GetRloc16(), GetRloc16()))
```
This way, parent will not send incorrect rloc (no matter what).

simonlingoogle, 2022-02-18:
> Thanks @TPolzer for reporting this and thanks @simonlingoogle for finding the cause.
> 
> 
> 
> Setting rloc16 of all children in `kStateChildIdRequest` to zero from `SetStateRouter()` can work. But I agree it may be better/easier to check for this in `SendChildIdResponse()` here:
> 
> https://github.com/openthread/openthread/blob/564982c818666787096e5ad7a81a2d2f036aba36/src/core/thread/mle_router.cpp#L3081-L3086
> 
> 
> 
> Replacing the `if` with 
> 
> ```c++
> 
>     if ((aChild.GetRloc16() == 0) || !RouterIdMatch(aChild.GetRloc16(), GetRloc16()))
> 
> ```
> 
> This way, parent will not send incorrect rloc (no matter what).
> 
> 

I am buying this fix too. 

Will verify if it works by stress tests.

simonlingoogle, 2022-02-18:
> For example, I can see from the code that the following is possible:
> 
> 1. REED has a child table entry in restoring state corresponding to neighbor X.

I think the child table entry was in `Invalid` state rather than `restoring` state. 

> 2. REED receives a Child ID Request from neighbor X.
> 3. REED upgrades to a router and does not remove the child table entry for neighbor X because it is in the Child ID Request state.
> 4. Router now sends a Child ID Response with the wrong RLOC16.

I can verify more on the initial state of the child table entry.

simonlingoogle, 2022-02-18:
> * REED has a child table entry in restoring state corresponding to neighbor X.
> * REED receives a Child ID Request from neighbor X.
> * REED upgrades to a router and does not remove the child table entry for neighbor X because it is in the Child ID Request state.
> * Router now sends a Child ID Response with the wrong RLOC16.

The process may be more complex than that because of follow:
- When REED restores a Child, the Child's RLOC16 must be matching
- When the REED receives a AddressSolicitResponse with a different Router ID, it will remove the restored Child to be Invalid. 
- When the Child send `Parent Request` and  `Child ID Request` again, the RLOC16 will be initialized to 0, so there won't be an issue. 

I wonder if the following process can cause issue:
1. A Router has a `Valid` Child (either restored or normally attached) with matching RLOC16
2. The Child lost the Parent (Router) and sends `Parent Request` and  `Child ID Request`, so the Child table entry on the Router enters `kStateChildIdRequest` state. 
3. The Router becomes a REED again for any reason, and the Child table entry remains in `kStateChildIdRequest` state
4. The Router (now REED) becomes Router again when it receives AddressSolicitResponse with a different Router ID. The Child table entry on the Router is not changed because of `kStateChildIdRequest` state. 
5. The Child reattaches by sending  `Parent Request` and  `Child ID Request`, and the Child table entry remains in `kStateChildIdRequest` state during this process
6. Thus the issue. 

Thoughts? @jwhui @abtink

simonlingoogle, 2022-02-18:
@jwhui 

With a second thought, I think your process for reproducing this issue should be correct. 

A REED can have Child in `Valid` state. And the Child will enter `kStateChildIdRequest` state in `HandleChildIdRequest` with a non-zero RLOC16. If later the REED becomes a Router with a different Router ID, the issue happens. 

The debugging seems to be telling the same story.

simonlingoogle, 2022-02-18:
> Replacing the `if` with
> 
> ```c++
>     if ((aChild.GetRloc16() == 0) || !RouterIdMatch(aChild.GetRloc16(), GetRloc16()))
> ```
> 
> This way, parent will not send incorrect rloc (no matter what).

Verified that it fixes the issue in stress tests.

abtink, 2022-02-18:
Just submitted https://github.com/openthread/openthread/pull/7428

TPolzer, 2022-02-18:
Thanks! Will try it out once I get around to flashing all the routers.

Since I control all devices on my network, I can do that. For other people it might be nice to have a solution that fixes the problem "locally", if that's feasible.

jwhui, 2022-02-19:
> For other people it might be nice to have a solution that fixes the problem "locally", if that's feasible.

I submitted #7430 that should cause a SED to restart the attach process if the Router IDs to not match.

jwhui, 2022-03-25:
Resolved by #7428 and #7430

closed 2022-03-25 by jwhui