openthread/openthread #7375: When using "simulation" to simulate in Linux, you cannot continuously simulate, and keep prompting "program received signal sig34, real time event 34" information

tuopu, 2022-01-29:
**Additional context** Add any other context about the problem here.

When simulating with Linux, use the CLI mode. When you enter "thread start", the system prompts "program received signal sig34, real time event 34

In __ GI___ select (nfds=7, readfds=0x7fffffffe370, writefds=0x7fffffffe3f0, exceptfds=0x7fffffffe470, timeout=0x7fffffffe360) at ../ sysdeps/unix/sysv/linux/select. c: 69 () ", and then run for 1-2 seconds, constantly prompt the above information, and the simulation can no longer be carried out.

jwhui, 2022-01-29:
Can you provide more detail about your setup? Are you trying to use GDB?

Note that simulation makes use of linux timers: https://github.com/openthread/openthread/blob/ce77ab3c1d7ad91b284615112ae38c08527bf73e/examples/platforms/simulation/alarm.c#L130

tuopu, 2022-01-30:
Yes, I use GDB emulation, and Linux uses the latest distribution Ubuntu 21.0,

use  “cmake -G"CodeBlocks - Ninja" -DOT_PLATFORM=simulation ..”
The command generates a simulation project in the build directory and uses CodeBlocks software for simulation. GDB can be used for normal simulation in the past, but the latest thread version will get the above information and cannot be simulated normally.

jwhui, 2022-01-30:
Have you tried configuring GDB with the following?
```
handle SIG34 pass nostop noprint
```

tuopu, 2022-01-30:
After testing, it can run normally. Can you help point out the reason for this message.

jwhui, 2022-01-30:
As mentioned in https://github.com/openthread/openthread/issues/7375#issuecomment-1024973316, it is caused by a timer that generates a signal and interrupts the process.

closed 2022-01-30 by tuopu