openthread/openthread #7387: how can  set and adjust thread to make it suitable for PHY chip

tuopu, 2022-02-02:
Because the chip I use is 433-868 / 915MHz phys and supports air speed of 20KB / s-1000kb / s, what parameters do I need to set and adjust thread to make it suitable for PHY chip,thank you.

jwhui, 2022-02-02:
Note that the Thread Specification currently only specifies operation over IEEE 802.15.4 2.4 GHz O-QPSK PHY. While it may be possible to adapt OpenThread to a different PHY, it would not be conformant to the current Thread Specification.

If you want to adapt OpenThread to support a different PHY that has longer communication delays, you may need to adjust various protocol timeouts. The most important ones are likely in MLE (see [src/core/thread/mle_types.hpp](https://github.com/openthread/openthread/blob/main/src/core/thread/mle_types.hpp)) and the MAC (see [src/core/mac/sub_mac.hpp](https://github.com/openthread/openthread/blob/main/src/core/mac/sub_mac.hpp)).

tuopu, 2022-02-03:
Thank you very much for your reply. I understand and will study further.

closed 2022-02-03 by tuopu