openthread/openthread #7388: How to update remote resources with put and post commands

tuopu, 2022-02-02:
I can use the command "coap resource test resource" to create resources, use "coap set testing123" to change resources, and use "coap get FDDE: ad00: beef: 0:2780:9423:166c: 1aac test resource" to obtain resources, but I can't update resource values with put and post. The commands are "coap post FDDE: ad00: beef: 0:2780:9423:166c: 1aac test resource con payload" and "coap post FDDE: ad00: beef: 0:2780:9423:166c: 1aac test resource block-1024 10", and I don't understand the role of "payload" in "coap post FDDE: ad00: beef: 0:2780:9423:166c: 1aac test resource con payload", Is it the content of the "test resource" resource?

jwhui, 2022-02-02:
> I can use the command "coap resource test resource" to create resources, use "coap set testing123" to change resources, and use "coap get FDDE: ad00: beef: 0:2780:9423:166c: 1aac test resource" to [...]

The OpenThread CLI is intended to serve as an example of how to use the OpenThread APIs. The current CoAP CLI implementation does not handle the CoAP put and post primitives - see implementation of `Coap::HandleRequest()`: https://github.com/openthread/openthread/blob/3b79cb0d084edc00c7ec8a29f97b0e8a40bbe18c/src/cli/cli_coap.cpp#L603

> I don't understand the role of "payload" in "coap post FDDE: ad00: beef: 0:2780:9423:166c: 1aac test resource con payload", Is it the content of the "test resource" resource?

"payload" specifies the CoAP message payload.

tuopu, 2022-02-03:
Thank you very much for your reply. I understand and will study further.

closed 2022-02-03 by tuopu