openthread/openthread #7418: Building Simulation Example(s) to Get Started...

silver2row, 2022-02-15:
**Describe the bug** A clear and concise description of what the bug is. 

` When I build the required packaged for simulation from this site, https://openthread.io/codelabs/openthread-simulation-posix/index.html?index=..%2F..index#1 , I then receive some warnings as errors. `

**To Reproduce** Information to reproduce the behavior, including:
1. ` main `
2. ` BBGG and cc1352R `
3. Build steps
a. git clone https://github.com/openthread/openthread
b. cd openthread 
c. git checkout origin/master
d. make -f examples/Makefile-simulation // This works!
e. make -f src/posix/Makefile-posix DAEMON=1 // This does not work and this is where I end w/ warnings as errors.
4. Network topology : ` Home Router w/ Extender but w/ the board, WiFi 2.4 Ghz, wpan, 6lowpan, Zigbee, and Sub 1-Ghz. `

**Expected behavior** A clear and concise description of what you expected to happen. ` I figured the commands listed would work without the correct installation of the simulation example (for whatever reason). `

` I am incorrect and could not use each command as stated. `

**Console/log output** If applicable, add console/log output to help explain your problem. 

```

Making all in posix
Making all in platform
  CXX      libopenthread_posix_a-netif.o
/home/debian/src/openthread/src/posix/../../src/posix/platform/netif.cpp: In function ‘void AddRtAttr(nlmsghdr*, uint32_t, uint8_t, const void*, uint8_t)’:
/home/debian/src/openthread/src/posix/../../src/posix/platform/netif.cpp:527:21: error: cast from ‘char*’ to ‘rtattr*’ increases required alignment of target type [-Werror=cast-align]
  527 |     rta           = (struct rtattr *)((char *)(aHeader) + NLMSG_ALIGN((aHeader)->nlmsg_len));
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[5]: *** [Makefile:1051: libopenthread_posix_a-netif.o] Error 1
make[4]: *** [Makefile:821: all-recursive] Error 1
make[3]: *** [Makefile:434: all-recursive] Error 1
make[2]: *** [Makefile:589: all-recursive] Error 1
make[1]: *** [Makefile:515: all] Error 2
make: *** [src/posix/Makefile-posix:326: do-build-posix] Error 2

```

**Additional context** Add any other context about the problem here.

1. armhf board w/ 2.4Ghz and BLE.
2. add-on board w/ cc1352R multi-band and M4F chip.
3. It seems I may be running out of onboard memory for openthread but I cannot be sure as of now.
4. Testing will ensue. Please try to read through the example and give or take. Thank you.

Seth

P.S. If you need additional info. from me, please ask or let me know exactly what you need.

superwhd, 2022-02-17:
Looks it's similar to https://github.com/openthread/openthread/issues/5647. And there was a fix https://github.com/openthread/openthread/pull/7237, not merged yet, unfortunately. 

You can also try adding the compile option ` -Wcast-align`.

silver2row, 2022-02-25:
Hello,

Thank you.

Seth

closed 2022-03-09 by silver2row