openthread/openthread #7435: networkdiagnostic exception

LancerSky, 2022-02-24:
**Describe the bug** A clear and concise description of what the bug is.
We have 100 End-devices attach to a Router at the same time. 
[childtable.log](https://github.com/openthread/openthread/files/8129034/childtable.log)
But only 84 were queried through the networkdiagnostic command. 
`ot-ctl networkdiagnostic get ff03::2 0 1 2 3 4 5 6 7 8 9 14 15 16 17 18 19`
[networkdiagnostic.log](https://github.com/openthread/openthread/files/8129051/networkdiagnostic.log)
[diagnostics.log](https://github.com/openthread/openthread/files/8129122/diagnostics.log)

I review the source code and found that the mTable variable is only assigned 84. It is right? 
Can I change the const variable OT_NETWORK_BASE_TLV_MAX_LENGTH or redefine it via cmake?

```
/**
 * Maximum value length of Thread Base TLV.
 */
#define OT_NETWORK_BASE_TLV_MAX_LENGTH 254

/**
 * Size of Network Diagnostic Child Table entry.
 */
#define OT_NETWORK_DIAGNOSTIC_CHILD_TABLE_ENTRY_SIZE 3

/**
 * This structure represents a Network Diagnostic TLV.
 *
 */
typedef struct otNetworkDiagTlv
{
    /**
     * The Network Diagnostic TLV type.
     */
    uint8_t mType;

    union
    {
        otExtAddress              mExtAddress;
        uint16_t                  mAddr16;
        otLinkModeConfig          mMode;
        uint32_t                  mTimeout;
        otNetworkDiagConnectivity mConnectivity;
        otNetworkDiagRoute        mRoute;
        otLeaderData              mLeaderData;
        otNetworkDiagMacCounters  mMacCounters;
        uint8_t                   mBatteryLevel;
        uint16_t                  mSupplyVoltage;
        uint32_t                  mMaxChildTimeout;
        struct
        {
            uint8_t mCount;
            uint8_t m8[OT_NETWORK_BASE_TLV_MAX_LENGTH];
        } mNetworkData;
        struct
        {
            uint8_t      mCount;
            otIp6Address mList[OT_NETWORK_BASE_TLV_MAX_LENGTH / OT_IP6_ADDRESS_SIZE];
        } mIp6AddrList;
        struct
        {
            uint8_t mCount;
            otNetworkDiagChildEntry
                mTable[OT_NETWORK_BASE_TLV_MAX_LENGTH / OT_NETWORK_DIAGNOSTIC_CHILD_TABLE_ENTRY_SIZE];
        } mChildTable;
        struct
        {
            uint8_t mCount;
            uint8_t m8[OT_NETWORK_BASE_TLV_MAX_LENGTH];
        } mChannelPages;
    } mData;
} otNetworkDiagTlv;
```
**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id https://github.com/openthread/ot-br-posix/commit/a965a7535dad33303585d2080437f3d796e54a81
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

wgtdkp, 2022-02-24:
@LancerSky Thanks for reporting this issue.

> Can I change the const variable OT_NETWORK_BASE_TLV_MAX_LENGTH or redefine it via cmake?

No, you shouldn't do that. The value `OT_NETWORK_BASE_TLV_MAX_LENGTH` is defined by specification and used by almost all Thread TLVs, changing it's value will break the entire Thread stack.

It's a limitation of the current Thread spec that Network Diagnostic TLV supports only value length of `254` which is not enough for the Child Table. I will work to extend max length of Child Table TLV to `65535` and request change to Thread specification.

LancerSky, 2022-02-24:
Great to hear that. 
And I found that there is duplicate data in ChildTable,  for example, 'ChildId: 0x0016' is repeated 3 times. Is this right?

EskoDijk, 2022-02-24:
Agree that OT_NETWORK_BASE_TLV_MAX_LENGTH can never be changed as such - when the data doesn't fit into the Base TLV then the Extended TLV format should be used that has a 16-bit length field.

@wgtdkp The Thread spec already supports the Extended TLV types for all TMF TLVs since v1.2.0 - see Section 10.4.3. So this applies to all TMF messages. Back in v1.1.1 it was also supported, however it was then unclear whether the extended format could be used for most messages because these messages did not list that extended format in the figure.  (To save work on spec editing it was decided in the v1.2.0 editing to not to explicitly show the extended format in all the TLV descriptions. Usually only one of both formats is shown in the figure.)

Another thing to note was that CoAP Block2 Option support was also added in the v1.2.1 spec to support the case of many Children (e.g. 511). That implies the returned TLV will be large and the Router's CoAP server needs to split the TMF response up into 2 or more COAP Blocks (depending on its chosen block size).

wgtdkp, 2022-02-24:
@EskoDijk Thanks for the information! Looks like we only need to update OpenThread implementation here now, great!

jwhui, 2022-11-07:
Resolved by #8368

closed 2022-11-07 by jwhui