openthread/openthread #7444: An incorrect size passed to sha256.Update in test_ecdsa

disconnect3d, 2022-02-28:
Hey, in the code below we can see a `sha256.Update` call that takes a `sizeof(kMessage)-1`. However, since `kMessage` is a pointer (`const char *kMessage = ...`) the sizeof operation returns the size of the pointer - and so e.g. 8 bytes (on 64-bit architectures). As a result, the sha is likely performed on a 7 or 8 bytes prefix of the kMessage string instead of on all of it.

https://github.com/openthread/openthread/blob/41409d4b60641a008c23f9a9c807b2d782eed070/tests/unit/test_ecdsa.cpp#L132-L155

It seems to me that either the `kMessage` declaration should be changed to `const char kMessage[] = "...";` or a `strlen(kMessage)` should be used instead of `sizeof(kMessage)`.

abtink, 2022-02-28:
Thanks for finding this.
- Submitted https://github.com/openthread/openthread/pull/7446

closed 2022-02-28 by jwhui

abtink, 2022-03-01:
@disconnect3d I know it is off-topic, but seeing the "night elf" icon reminded me of good memories of playing Warcraft 3 almost 20 years ago 😄