openthread/openthread #7460: Thread JIRAs/changes and OT implementation

abtink, 2022-03-09:
Created this issue to track list of Thread group JIRAs and changes being discussed/proposed/implemented and their related status and implementation in OpenThread:

- [SPEC-999](https://threadgroup.atlassian.net/browse/SPEC-999) Link Layer Retry Delay on no-ack
   - PR: https://github.com/openthread/openthread/pull/7346
- [SPEC-1000](https://threadgroup.atlassian.net/browse/SPEC-1000) Extra delay when packet is to be forwarded
- [SPEC-1036](https://threadgroup.atlassian.net/browse/SPEC-1036) MLE Parent Request Retries
- [SPEC-1033](https://threadgroup.atlassian.net/jira/core/projects/SPEC/issues/SPEC-1033) Becoming router

_Note: Access Thread JIRA links require Thread membership and Thread group login/account._

edmont, 2022-03-28:
Hi @abtink, could you consider adding [SPEC-799](https://threadgroup.atlassian.net/browse/SPEC-799) - https://github.com/openthread/openthread/pull/7472 to the list?

edmont, 2022-04-19:
[SPEC-1062](https://threadgroup.atlassian.net/browse/SPEC-1062) to be added?

simonlingoogle, 2022-06-23:
Need to track [JIRA] (SPEC-1077) Resolving on-link prefix contention on AIL ? @abtink

jwhui, 2022-06-23:
> Need to track [JIRA] (SPEC-1077) Resolving on-link prefix contention on AIL ? @abtink

Added to https://github.com/orgs/openthread/projects/1/views/1

closed 2025-07-17 by abtink