openthread/openthread #7478: [CLI] handling of a full sTxBuffer

thcu-gp, 2022-03-14:
Issues in `CliUartOutput` lines 333 to 350
* A platform flush is implemented here, but it is an internal CLI buffer that is full. There is no guarantee that the underlying uart buffer is the issue
* Additionally if the flush failed, adding the output of `otLogWarnPlat(...)` of line 347 will not help, it ends up in the same buffer

bukepo, 2022-04-21:
* Yes, the internal CLI buffer is for formatting the output, and it's possible the underlying uart buffer is not full. But we have to format the output first before we can deliver it to the underlying buffer. And I'm not aware of a way to partial format and continue later.
* The logging doesn't have to be the same buffer, OpenThread supports other log facility like syslog.

thcu-gp, 2022-04-22:
While the logging doesn't have to be the same buffer, it is possible.
When it happens the code gets in a loop.