openthread/openthread #7493: [test] MIC computation incorrect on extended address mode src addr

vitprajzler, 2022-03-18:
**Describe the bug**

The macFrame.parse function in `test/scripts/thread-cert/mac802154.py` is trimming encrypted payloads and computes incorrect MICs for payloads with extended address mode source address. 

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id: 0a74d66fedd3c9558ccf298055305a0eef5a6896
2. IEEE 802.15.4 hardware platform: nRF52840
3. Build steps: no build needed
4. Network topology: no network topology needed

For example, the following 802.15.4 frame is incorrectly parsed: 
`6998483412003802380dc50b000001a1ecdb4dcb8ccd43b7c29a465c832b5625aecf532a7fc4b1c1ebf5d89408e2`

with the `DEFAULT_NETWORK_KEY` ( `[ 0x00, 0x11, 0x22, 0x33, 0x44, 0x55, 0x66, 0x77, 0x88, 0x99, 0xaa, 0xbb, 0xcc, 0xdd, 0xee, 0xff ]`)
and RLOC16 mapping as follows
`mac802154.DeviceDescriptors.add(0x3802, common.MacAddress(bytearray([ 0xa6, 0x28, 0xe1, 0x18, 0x17, 0xca, 0x95, 0x67 ]), common.MacAddressType.LONG))`

parses the payload into plaintext
`7a773a8000e1a6004202680517aa3200000000000000000000`
and MIC `74b9f59c`

**Expected behavior** 

Payload should be 
`7a773a8000e1a6004202680517aa32000000000000000000000000`
and MIC computed as per the original frame, `d89408e2`

**Console/log output** N/A

vitprajzler, 2022-03-18:
Adding `+ 2` to the following line fixes the immediate problem

https://github.com/openthread/openthread/blob/687e04f6d460ec48b7b5b88216ed4b6d7348b81c/tests/scripts/thread-cert/mac802154.py#L277

Irving-cl, 2022-03-27:
@vitprajzler 

Sorry for the late response and really thanks for pointing it out! The code **was** correct. It parsed from the end: the last 2 bytes was parsed as the FCS, the 4 bytes before it was parsed as the **MIC**, and that position is regarded as the end of the payload. That explained why the payload is 2-bytes less from expected. You are also right because I just found the FCS somehow has gone in the MacFrame.

A previous capture:
[image]

A recent capture:
[image]

The first capture comes from previous Thread tests in 2020 Feb. And the second one is from SiLabs when developing CSL feature (#6981, a few weeks ago) Not sure when and why FCS was removed from the Mac Frame. Do you know something about this? @jwhui @simonlingoogle 

If FCS should be there, we need to fix it. If it's intentionally removed, we can update the test scripts and do not parse FCS.

jwhui, 2022-04-07:
I believe this is a difference in tooling. When running CI tests, we capture packets with the FCS. However, many sniffer tools do not include the FCS.

If we want to support both models, then we'll need a way to determine whether the FCS was included or not.

jwhui, 2022-08-23:
Closing stale issue.

closed 2022-08-23 by jwhui