openthread/openthread #7501: About global macro definitions

tuopu, 2022-03-22:
Ask the expert for a question. I want to put the global macro definition of thread into a header file. Which file is more appropriate, such as "openthread_ftd = 1".

jwhui, 2022-03-22:
The build system typically defines this value depending on what it is trying to build.

Some examples:

- https://github.com/openthread/openthread/blob/237c91b9398bfe55426c45cbe5bcb5713c178ba3/Android.mk#L53
- https://github.com/openthread/openthread/blob/237c91b9398bfe55426c45cbe5bcb5713c178ba3/BUILD.gn#L46
- https://github.com/openthread/openthread/blob/237c91b9398bfe55426c45cbe5bcb5713c178ba3/src/core/ftd.cmake#L32

tuopu, 2022-03-25:
Hello, expert, thank you for your reply. I'm learning to apply thread to FreeRTOS. When referring to the example, I found that the thread in OT RTOS can't be upgraded to the latest version, and it can't be compiled successfully in Linux. Please ask your latest submitted thread version to apply in FreeRTOS. Do you have any good suggestions?

tuopu, 2022-03-25:
One more question，Under the world mechanism of thead, how can we know that the thread is about to arrive, so as to use semaphores and other system functions to inform the task to continue running?

jwhui, 2022-03-29:
> One more question，Under the world mechanism of thead, how can we know that the thread is about to arrive, so as to use semaphores and other system functions to inform the task to continue running?

Platform drivers handle events (e.g. timer fires, radio packets received, etc.) and those drivers are responsible for scheduling OpenThread execution at the appropriate time depending on the platform's implementation.