openthread/openthread #7505: otbr-agent crash

nykh2010, 2022-03-22:
https://github.com/openthread/openthread/blob/687e04f6d460ec48b7b5b88216ed4b6d7348b81c/src/lib/spinel/radio_spinel_impl.hpp#L2226

in csl mode, when SSED send child update request, otbr-agent always crash at this. 
`Mar 22 16:47:37 mi otbr-agent[77754]: 00:00:06.790 [I] BorderRouter--: Evaluating routing policy
Mar 22 16:47:37 mi otbr-agent[77754]: 00:00:06.790 [I] BorderRouter--: Send OMR prefix fd18:a7f5:5aec:1::/64 in RIO (valid lifetime = 1800 seconds)
Mar 22 16:47:37 mi otbr-agent[77754]: 00:00:06.790 [I] BorderRouter--: Sent Router Advertisement on interface 2
Mar 22 16:47:37 mi otbr-agent[77754]: 00:00:06.791 [I] BorderRouter--: Router advertisement scheduled in 16 seconds
Mar 22 16:47:37 mi otbr-agent[77754]: 00:00:06.791 [I] BorderRouter--: Start evaluating routing policy, scheduled in 16000 milliseconds
Mar 22 16:47:37 mi otbr-agent[77754]: 00:00:06.791 [I] BorderRouter--: Received Router Advertisement from fe80:0:0:0:e405:4a65:2b26:7f68 on interface 2
Mar 22 16:47:39 mi otbr-agent[77754]: 00:00:08.934 [I] BorderRouter--: Router solicitation times out
Mar 22 16:47:39 mi otbr-agent[77754]: 00:00:08.934 [I] BorderRouter--: Received Router Solicitation from fe80:0:0:0:e405:4a65:2b26:7f68 on interface 2
Mar 22 16:47:39 mi otbr-agent[77754]: 00:00:08.934 [I] BorderRouter--: Start evaluating routing policy, scheduled in 856 milliseconds
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.790 [I] BorderRouter--: Evaluating routing policy
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.790 [I] BorderRouter--: Send OMR prefix fd18:a7f5:5aec:1::/64 in RIO (valid lifetime = 1800 seconds)
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.791 [I] BorderRouter--: Sent Router Advertisement on interface 2
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.791 [I] BorderRouter--: Router advertisement scheduled in 209 seconds
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.791 [I] BorderRouter--: Start evaluating routing policy, scheduled in 209000 milliseconds
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.791 [I] BorderRouter--: Received Router Advertisement from fe80:0:0:0:e405:4a65:2b26:7f68 on interface 2
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.803 [I] Mac-----------: Frame rx failed, error:Drop, len:120, seqnum:87, type:Data, src:aa929392c2e906f2, dst:72bc91b0e3d9e227, sec:yes, ackreq:yes
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.932 [I] BorderRouter--: Vicarious router solicitation time out
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.934 [I] BorderRouter--: Router solicitation times out
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:09.934 [I] BorderRouter--: Start evaluating routing policy, scheduled in 2857 milliseconds
Mar 22 16:47:40 mi otbr-agent[77754]: 00:00:10.151 [I] Mle-----------: Child CSL synchronization expired
**Mar 22 16:47:41 mi otbr-agent[77754]: 00:00:10.953 [W] Platform------: radio tx timeout
Mar 22 16:47:41 mi otbr-agent[77754]: 00:00:10.953 [C] Platform------: BSD TCP function() at radio_spinel_impl.hpp:2226:** RadioSpinelNoResponse
Mar 22 16:47:41 mi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=6/NOTCONFIGURED
Mar 22 16:47:41 mi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Mar 22 16:47:46 mi systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 56.
Mar 22 16:47:47 mi otbr-agent[77901]: [INFO]-UTILS---: Running 0.3.0-3e501c4efd
`

jwhui, 2022-03-22:
@nykh2010 , can you provide more detail about your setup?

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

nykh2010, 2022-03-24:
commit id is 3e501c4efd3ac740524a6d6fc546308311749ccb in main branch.
RCP and SSED platform are EFR32MG21 of Silabs. 
ot-br-posix is build ./script/setup directly. 
SSED connect to Border Router directly.

jwhui, 2022-03-24:
@romacdon @suveshpratapa , can you help take a look?

romacdon, 2022-03-24:
@nykh2010  can you provide additional info on your RCP environment?  Where are you building this from? ot-efr32?  If ot-efr32 what git commit are you using?  If you are using Simplicity Studio what GSDK version are you using?

nykh2010, 2022-03-29:
RCP is build by Simplicity Studio 5, the detail of SDK version is  "Gecko SDK Suite: Amazon, Bluetooth 3.3.0, Bluetooth Mesh 2.2.0, EmberZNet 7.0.0.0, Flex 3.3.0.0, MCU 6.2.0.0, Micrium OS Kernel, OpenThread 2.0.0.0 (GitHub-9dedd1869), Platform 4.0.0.0, USB 1.0.0.0, Wi-SUN 1.2.0.0, Z-Wave SDK 7.17.0.0 (v4.0.0)". The hardware is a USB stick of mg21.

suveshpratapa, 2022-03-29:
Hi @nykh2010!

It sounds like you're using our [major SDK release 2.0.0.0](https://www.silabs.com/documents/public/release-notes/open-thread-release-notes-2.0.0.0.pdf), tagged with the following compatible GitHub versions. Please check both of them:
[openthread @ ](https://github.com/openthread/openthread/commits/9dedd1869)9dedd1869
[ot-br-posix @](https://github.com/openthread/ot-br-posix/commits/58d09bee8) 58d09bee8

Note however, that we have a number of bug fixes we added around the low power feature in [patch release 2.0.2.0](https://www.silabs.com/documents/public/release-notes/open-thread-release-notes-2.0.2.0.pdf), with the following GitHub commits.  We suggest you use the latest 2.0.2.0 SDK if possible:
[openthread @ ](https://github.com/openthread/openthread/commits/3b79cb0d084)3b79cb0d084
[ot-br-posix @](https://github.com/openthread/ot-br-posix/commits/5be35949c6f) 5be35949c6f

nykh2010, 2022-03-30:
@suveshpratapa ok, thank you! I will use the newest SDK and try again.

jwhui, 2022-08-23:
Closing stale issue.

closed 2022-08-23 by jwhui