openthread/openthread #7513: Host crashes with otPlatRadioEnableSrcMatch

dismirlian, 2022-03-23:
We have observed frequent ot-daemon crashes in a topology with two/three REEDs (all RCP+posix -based), and many SEDs (eg. 70), under heavy traffic.

The ot-daemon hosts always crash with the following error message:

`Mar 22 17:23:12 GW0087 /usr/sbin/ot-daemon[27084]: [CRIT]-PLAT----: otPlatRadioEnableSrcMatch() at ../git/src/posix/platform/radio.cpp:274: Failure`

Browsing the code, the daemon dies here:
```
void otPlatRadioEnableSrcMatch(otInstance *aInstance, bool aEnable)
{
    OT_UNUSED_VARIABLE(aInstance);
    SuccessOrDie(sRadioSpinel.EnableSrcMatch(aEnable));
}
```

Apparently, Spinel runs out of transaction IDs when trying to enable the source match table after adding many children (maybe 15 or more), and when a `SPINEL_PROP_STREAM_RAW` frame appears in between. This can be seen in this portion of the logs ([srcmatch.log](https://github.com/openthread/openthread/files/8335470/srcmatch.log)):

```
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: Starting operation "TransmitDataDirect"
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: RadioState: Receive -> CsmaBackoff
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: RadioState: CsmaBackoff -> Transmit
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=13, key=113, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MLE-----: network id timeout = 6
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:a88d:ac97:dee0:e452,0xf01e)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:18ec:ab48:6dcd:6d54,0xf01f)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MAC-----: Prepping indir tx IPv6 UDP msg, len:81, chksum:b72b, to:0xf020, sec:yes, prio:normal
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:c057:eb86:6e77:5957,0xf020)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:d4bf:6d8f:7fc5:b592,0xf023)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=14, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=14 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Clearing short addr: 0xf022 -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:2c03:8113:4c10:fc9,0xf022)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=15, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=15 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Clearing short addr: 0xf024 -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:e080:fdb9:4470:cf03,0xf024)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=1, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=1 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Clearing short addr: 0xf025 -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:108a:cf7:280f:5413,0xf025)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=2, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=2 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Clearing short addr: 0xf02a -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:46a:8ce7:c62d:1a41,0xf02a)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=3, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=3 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Clearing short addr: 0xf029 -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:bc9b:caf8:89d0:eaed,0xf029)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=4, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=4 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Clearing short addr: 0xf028 -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:8413:3e7:5a94:692b,0xf028)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=5, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=5 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Clearing short addr: 0xf02b -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:3c75:43d4:b37d:cee9,0xf02b)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=6, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=6 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Clearing short addr: 0xf01d -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:780a:6483:48a:7499,0xf01d)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=7, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=7 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Clearing short addr: 0xf02c -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MLE-----: Send Child Update Request to child (fe80:0:0:0:b06e:3fa4:7284:d049,0xf02c)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MAC-----: Prepping indir tx IPv6 UDP msg, len:237, chksum:0880, to:0xf020, sec:no, prio:net
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=8, key=4868, wait=1
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: Wait response: tid=8 key=4868
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-MAC-----: SrcAddrMatch - Adding short addr: 0xf023 -- OK (0)
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [INFO]-MAC-----: Prepping indir tx IPv6 UDP msg, len:237, chksum:65ae, to:0xf023, sec:no, prio:net
Mar 23 16:48:16 GW0087 /usr/sbin/ot-daemon[21286]: [DEBG]-PLAT----: RequestV: tid=9, key=4868, wait=1
[... 45 lines cut ...]
    VerifyOrExit(!aWait || tid > 0, error = OT_ERROR_BUSY);

    error = SendCommand(command, aKey, tid, aFormat, aArgs);
    SuccessOrExit(error);
    
    .....

}

void otPlatRadioEnableSrcMatch(otInstance *aInstance, bool aEnable)
{
    OT_UNUSED_VARIABLE(aInstance);
    auto err = sRadioSpinel.EnableSrcMatch(aEnable);
    if (err != OT_ERROR_NONE) {
        otLogCritPlat("otPlatRadioEnableSrcMatch: %s", otThreadErrorToString(err));  // ADDED
    }
    SuccessOrDie(err);
}
```

If I understand correctly, the command that enables the source match table (and the ones that manipulate the table) are all synchronous (`RequestV` waits for the response before returning, so the `tid` is freed immediately), but the radio TX (`SPINEL_PROP_STREAM_RAW`) is not, so the `tid` is marked busy. If it is not freed before 15 commands are executed, then the 16th command will fail because of unavailable `tids`. The Add/Clear short address commands, if failed, don't cause the daemon to `exit()`, but the Enable command does (that's why it always crashes in the same location, though I suspect we may be running out of TIDs more often). Actually, there *are* `tids` available, but *not* the one that is still in use by the radio TX operation. Can we reuse a `tid` that has been recently freed? Maybe I'm missing something; I don't have intimate knowledge of the code.

**To Reproduce** Information to reproduce the behavior, including:
1. Git commit id: 78c6c90c6ac3d7bf3641a373631412c63e91d8c6, but I think the same behavior would be observed with the latest.
2. IEEE 802.15.4 hardware platform: EFR32

abtink, 2022-03-23:
Thanks @dismirlian . Something does not feel correct about the logs.

My expectation is that commands are handled one by one and synchronously on host when interacting with RCP (from what I recall `radio-spinel` should wait till it gets the spinel reply from RCP before returning from any of radio API calls including `AddSrcMatch`, `ClearSrcMatch`). So there should be only one waiting `tid` and should not run out of ids.

@bukepo can you please comment on this and maybe take a look? thanks.

dismirlian, 2022-03-23:
Hi @abtink thanks for your answer.

In `RadioSpinel::RequestV` you can see that the host doesn't wait for radio TX (`SPINEL_PROP_STREAM_RAW`) to finish, here:

https://github.com/openthread/openthread/blob/f81c0b50f0a3d6f51fb09162bf2511fc62f6e6fe/src/lib/spinel/radio_spinel_impl.hpp#L1741-L1771

Actually, if there is a radio TX in progress and the host tries to send 15 or more commands to the RCP (not radio TX of course, but for example Add/Clear src match table), then the host *thinks* it runs out of TIDs (because it tries to reuse the TID from the ongoing radio TX, which is still busy).

As a proof-of-concept fix, I modified `RadioSpinel::GetNextTid` to "search" for the first free TID:
```C++
template <typename InterfaceType, typename ProcessContextType>
spinel_tid_t RadioSpinel<InterfaceType, ProcessContextType>::GetNextTid(void)
{
    spinel_tid_t tid           = 0;
    spinel_tid_t freeCandidate = 1;
    uint16_t     mask          = (1 << freeCandidate);

    for (; freeCandidate < 16; freeCandidate++, mask <<= 1)
    {
        if ((mask & mCmdTidsInUse) == 0)
        {
            tid = freeCandidate;
            mCmdTidsInUse |= mask;
            break;
        }
    }

    return tid;
}
```

This seems to fix the issue. What do you think?

tlifschitz, 2022-03-23:
Hi @abtink and @bukepo!

Just for the record. The current implementation of `RadioSpinel::GetNextTid` increments tid by one assuming the next time it will be free. That is not the case in the situation @dismirlian mentions, in which more than 14 commands are sent while is an ongoing radio packet transmission:

https://github.com/openthread/openthread/blob/09dd21abcad7f55e511ab71c8061c7145e2d5bd8/src/lib/spinel/spinel.h#L4828

https://github.com/openthread/openthread/blob/d79d0a99ac2e3ceb5e5a0755da9f39b1c1e26219/src/lib/spinel/radio_spinel_impl.hpp#L1699-L1712

abtink, 2022-03-23:
Thanks guys!

I agree. The root cause seems to be the `GetNextTid()`. The fix to loop over TIDs seems reasonable to me.  I would suggest starting with `mCmdNextTid` to try to keep them in sequence. How about something like this:
```c++
spinel_tid_t RadioSpinel<InterfaceType, ProcessContextType>::GetNextTid(void)
{
    spinel_tid_t tid = mCmdNextTid;

    while ((1 << tid) & mCmdTidsInUse) != 0)
    {
        tid = SPINEL_GET_NEXT_TID(tid);

        if (tid == mCmdNextTid)
        {
            ExitNow(tid = 0);
        }
    }

    mCmdTidsInUse |= (1 << tid);
    mCmdNextTid = SPINEL_GET_NEXT_TID(tid);

exit:
    return tid;
}
```

Reading the code, I think basically we can have at most two "in-use" TIDs, one for ongoing TX (waiting for TX done) and one for a spinel command (waiting for its response) associated with a radio platform API call.

----

Unrelated to this issue, but reading the radio-spinel, I always feel the code is more complex that it needs to be - at some point it may be good to re-design and simplify this, e.g., the fact that radio tx API can hold to a TID and tx done event is signaled using the same TID by RCP and as  a `PROP_LAST_STATUS` property is not the best design in my view.

abtink, 2022-03-23:
I can submit a PR with above change.

abtink, 2022-03-24:
Submitted PR https://github.com/openthread/openthread/pull/7516 to update `GetNextTid()`.

Thanks again for finding and debugging this issue and finding the root-cause.

dismirlian, 2022-03-24:
Hi @abtink, thanks for your answer and the PR! Your solution would work perfectly, actually @tlifschitz had exactly the same idea on how to fix this issue.

I agree that the Spinel code is overly complex. For this case, for instance, we could use fixed TIDs (one for radio TX and another for every other command).

We'll test it tomorrow and provide feedback. The proof of concept fix I proposed has been working well for several hours now, while the original implementation crashed in few minutes, so I think your PR should fix this issue.

Thanks again!

bukepo, 2022-03-25:
@dismirlian is it verified that #7516 fixed the issue?

dismirlian, 2022-03-25:
Hi @bukepo, ot-daemon has been running for several hours without issue, so I thing this is fixed. Thanks!

closed 2022-03-25 by dismirlian

jwhui, 2022-03-25:
@dismirlian @tlifschitz , thanks for reporting this issue and verifying the fix!