openthread/openthread #7518: Can't access local TMF from host stack

dismirlian, 2022-03-24:
On posix host, we were able to access the TMF running on the local stack, using commit 78c6c90c6ac3d7bf3641a373631412c63e91d8c6.

For example, we could retrieve the local network diagnostics from the linux command line:
```
# printf "\x12\x3\x0\x5\x16" | coap-client -v6 -mpost coap://[fdd4:3333:2222:1111:0:ff:fe00:3c00]:61631/d/dg -f -
v:1 t:CON c:POST i:9b5e {} [ Uri-Port:61631, Uri-Path:d, Uri-Path:dg ] :: '\x12\x03\x00\x05\x16'
v:1 t:ACK c:2.04 i:9b5e {} [ ] :: '\x00\x08\xB6\xA5"\x0C5@p\xA3\x05\x0Cz\x00\x01\x00\x81\x00\x00\x00\x00\x01\xF1\xF1'
```

`fdd4:3333:2222:1111:0:ff:fe00:3c00` is the RLOC of the local host stack. 

Also, we could query a remote device:
```
root@GW0087:~# printf "\x12\x3\x0\x5\x16" | coap-client -v6 -mpost coap://[fdd4:3333:2222:1111:0:ff:fe00:3c1a]:61631/d/dg -f -
v:1 t:CON c:POST i:4afc {} [ Uri-Port:61631, Uri-Path:d, Uri-Path:dg ] :: '\x12\x03\x00\x05\x16'
v:1 t:ACK c:2.04 i:4afc {} [ ] :: '\x00\x08\x12\x8A\x0C\xF7(\x0FT\x13'
```

However, using the latest version (119df62cbddce9d3e10b0a56d7fbf9d53c7fa218), we *can* query remote devices but not the local stack:

```
# printf "\x12\x3\x0\x5\x16" | coap-client -v6 -mpost coap://[fdd4:3333:2222:1111:0:ff:fe00:3c00]:61631/d/dg -f -
v:1 t:CON c:POST i:0e0e {} [ Uri-Port:61631, Uri-Path:d, Uri-Path:dg ] :: '\x12\x03\x00\x05\x16'
Mar 24 13:45:08.548 WARN coap_network_read: unreachable
```

We get an ICMP network unreachable error. Is this by design? How could we restore this feature? Thanks!

simonlingoogle, 2022-03-24:
I guess this is because of the change of https://github.com/openthread/openthread/pull/5910, after which the TMF port is no longer accessible from Host side because a socket that binds to wpan0 interface is not created. 

I think there is no hardcore reason why TMF UDP should not have a socket binding to wpan0.
I recall one reason why we changed TMF UDP to not bind to wpan0 is because that binding to wpan0 changes the order of TMF and MLE message transmissions sometimes, which breaks Harness Thread 1.1 Certification (but it should be a Harness issue).

dismirlian, 2022-03-24:
@simonlingoogle thanks for your answer!

Ok, so it should be possible to restore the socket binding to wpan0. What would be the neatest way of doing this? Maybe an opt-in configuration, so that the Harness continues to work?

simonlingoogle, 2022-03-25:
> @simonlingoogle thanks for your answer!
> 
> Ok, so it should be possible to restore the socket binding to wpan0. What would be the neatest way of doing this? Maybe an opt-in configuration, so that the Harness continues to work?

I think we can add configuration to alter the behavior, and leave the current behavior as the default.

jwhui, 2022-08-23:
Closing stale issue.

closed 2022-08-23 by jwhui