openthread/openthread #7525: CC1352   OpenThread  Sub-1GHz    868MHz

MichalPoterek, 2022-03-25:
I would like to experiment with the OpentThread network in the Sub-1GHz band to explore what kind of coverage and performance I can get. 

I'm using Texas Instrument CCS with TI  OpenThread SDK.

I found out that going to the 868MHz band will be not as easy as configuring the appropriate frequency in the configurator. I was looking in google is there are any pieces of information about similar ideas realized for TI  chipsets. Unfortunately, I haven't any good explanation for TI.

Based on a comment to code below let me know if I'm going in the right way and if these are all places that I need to modify. I will be appreciated any help  with that :)

`###################################
#######  openthread-core-default-config.h
###################################

#ifndef OPENTHREAD_CONFIG_RADIO_915MHZ_OQPSK_SUPPORT
#ifndef OPENTHREAD_CONFIG_RADIO_2P4GHZ_OQPSK_SUPPORT
#define OPENTHREAD_CONFIG_RADIO_915MHZ_OQPSK_SUPPORT 1    //this I think need to be changed to 1
#define OPENTHREAD_CONFIG_RADIO_2P4GHZ_OQPSK_SUPPORT 0    //this I think need to be changed to 0
#endif
#endif

#ifndef OPENTHREAD_CONFIG_DEFAULT_CHANNEL
#if OPENTHREAD_CONFIG_RADIO_2P4GHZ_OQPSK_SUPPORT
#define OPENTHREAD_CONFIG_DEFAULT_CHANNEL 11
#else
#if OPENTHREAD_CONFIG_RADIO_915MHZ_OQPSK_SUPPORT
#define OPENTHREAD_CONFIG_DEFAULT_CHANNEL 1
#endif // OPENTHREAD_CONFIG_RADIO_915MHZ_OQPSK_SUPPORT
#endif // OPENTHREAD_CONFIG_RADIO_2P4GHZ_OQPSK_SUPPORT
#endif // OPENTHREAD_CONFIG_DEFAULT_CHANNEL

####################################
######### radio.c
####################################

/**
 * Function documented in platform/radio.h
 */
otError otPlatRadioEnable(otInstance *aInstance)
{
    otError error = OT_ERROR_BUSY;
    RF_Params rfParams;
    (void)aInstance;

    if (sState == platformRadio_phyState_Sleep)
    {
        error = OT_ERROR_NONE;
    }
    else if (sState == platformRadio_phyState_Disabled)
    {
        RF_Params_init(&rfParams);

        rfParams.pErrCb         = rfCoreErrorCallback;
        rfParams.nID            = RF_STACK_ID_THREAD;    //here I think it should be changed to RF_STACK_ID_154  or RF_STACK_ID_CUSTOM 
        sRfHandle = RF_open(&sRfObject, &RF_prop,
                (RF_RadioSetup *)&sRadioSetupCmd, &rfParams);

        otEXPECT_ACTION(sRfHandle != NULL, error = OT_ERROR_FAILED);
        sState = platformRadio_phyState_Sleep;

        error = OT_ERROR_NONE;
    }

exit:
    if (error == OT_ERROR_FAILED)
    {
        sState = platformRadio_phyState_Disabled;
    }

    return error;
}

[... 244 lines cut ...]
    .endTime = 0x00000000
};

// CMD_IEEE_RX_ACK
// IEEE 802.15.4 Receive Acknowledgement Command
const rfc_CMD_IEEE_RX_ACK_t RF_cmdIeeeRxAck =
{
    .commandNo = 0x2C03,
    .status = 0x0000,
    .pNextOp = 0,
    .startTime = 0x00000000,
    .startTrigger.triggerType = 0x0,
    .startTrigger.bEnaCmd = 0x0,
    .startTrigger.triggerNo = 0x0,
    .startTrigger.pastTrig = 0x0,
    .condition.rule = 0x0,
    .condition.nSkip = 0x0,
    .seqNo = 0x00,
    .endTrigger.triggerType = 0x0,
    .endTrigger.bEnaCmd = 0x0,
    .endTrigger.triggerNo = 0x0,
    .endTrigger.pastTrig = 0x0,
    .endTime = 0x00000000
};
`

jwhui, 2022-04-05:
The existing sub-GHz implementation is a prototype and not anything that is currently specified in Thread. The existing prototype implements IEEE 802.15.4 O-QPSK in the sub-GHz bands.

I'm not sure if the CC1352 supports IEEE 802.15.4 O-QPSK in sub-GHz bands. @srickardti , can you help provide some insight?

srickardti, 2022-04-06:
Yes, setting the Sub-1GHz O-QPSK modulation is not supported on the CC1352. The device has not been characterized for 868MHz or 915MHz operation as defined by IEEE 802.15.4-2015 chapter 12. You can find the characterized PHYs in chapter 8 sections 10 to 23 of the [CC1352R Datasheet](https://www.ti.com/lit/gpn/cc1352r). Unfortunately, adding a PHY is not just tuning the transmitter and receiver to a lower frequency.

Furthermore, to enable the other PHY options for the radio, the commands will need to be translated from the IEEE command set to the PROP command set. To achieve lower power operation the current implementation relies on the radio core for some of the low level MAC operations like CSMA-CA, frame filtering, and ack matching. These are not available with the other physical layers. So the OpenThread software MAC will need to take up the difference.

MichalPoterek, 2022-04-07:
Thank you for your help and deep explanation. Looks like probably better to use another vendor chip which provides already SDK for this purpose. Do you have any experience with that? Which vendor might be best suited to use with thread?

srickardti, 2022-04-07:
I believe this feature was used as a proof of concept for one vendor's silicon offering. Currently there is no standard, interoperable, way to run a Thread network in Sub-1GHz, however the Thread Group is currently working on this.

I would caution running this feature in a shipped product. Since this question calls out the 868MHz band, I assume this is operating in a region following ETSI regulations. Consider ERC 70-03 for the spectrum allocation and EN 300 220-1 for the definition of polite spectrum access. This feature in OpenThread seems to be running the 2.4GHz O-QPSK MAC at Sub-1GHz frequencies with minimal changes. This has the potential to easily run afoul of the ETSI regulations.

jwhui, 2022-04-07:
Yes, as I mentioned in https://github.com/openthread/openthread/issues/7525#issuecomment-1089151244, the current sub-GHz implementation is a proof-of-concept and not something that's included in the current Thread Specification.