openthread/openthread #7529: otbr-agent crash when receiving 20 udp broadcasts per second "radio tx timeout"

HeFeng1947, 2022-03-28:
**otbr-agent crash when receiving 20 udp broadcasts per second**

**To Reproduce** 
1. Git commit id: **otbr-posix version: 2022-1-5 a965a7535dad33303585d2080437f3d796e54a81** 
2. IEEE 802.15.4 hardware platform: **ipq4018/ qca9531** (host) + **efr32mg21** (rcp) ; **460800** bardrate ; connect with **ch340g** usb2uart with **hardware flow control**
3. Build steps: ot-br-posix openwrt version; openwrt 19.07.7 ; host rx buffer add to 2048
4. Network topology: otbr as leader, 20 mtd device(nrf52840 dongle) connect directly. Child send a **COAP PUT** packet of one byte to **FF03 ::1** every one second

otbr-agent log output:
```
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.939 [NOTE]-CLI-----: Input: bufferinfo
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.939 [NOTE]-CLI-----: Output: total: 2048
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.939 [NOTE]-CLI-----: Output: free: 1740
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.940 [NOTE]-CLI-----: Output: 6lo send: 153 306
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.940 [NOTE]-CLI-----: Output: 6lo reas: 0 0
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.940 [NOTE]-CLI-----: Output: ip6: 0 0
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.940 [NOTE]-CLI-----: Output: mpl: 1 2
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.940 [NOTE]-CLI-----: Output: mle: 0 0
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.940 [NOTE]-CLI-----: Output: arp: 0 0
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.941 [NOTE]-CLI-----: Output: coap: 0 0
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.941 [NOTE]-CLI-----: Output: coap secure: 0 0
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.941 [NOTE]-CLI-----: Output: application coap: 0 0
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:46.941 [NOTE]-CLI-----: Output: Done
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.058 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:8ce4:dc46:c30:b57d PUT with payload: 03
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.119 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:6738:b2ee:25a5:182a PUT with payload: 0b
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.129 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:d838:f8b1:10a5:2c71 PUT with payload: 05
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.133 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:c709:59e0:2a1a:d2a8 PUT with payload: 01
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.184 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:b198:efaf:9d81:79c7 PUT with payload: 0e
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.195 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:159b:f725:6fde:dbe5 PUT with payload: 12
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.197 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:be79:8241:e1fb:729d PUT with payload: 09
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.240 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:6600:5182:e6dd:13f6 PUT with payload: 11
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.253 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:8b51:404f:9905:9dc1 PUT with payload: 06
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.272 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:3d1c:2369:589f:8ced PUT with payload: 0f
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.287 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:bdcd:a8ac:3e7a:12ec PUT with payload: 08
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.322 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:4bb6:ebe8:be5e:e701 PUT with payload: 0c
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.393 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:86c7:2b94:828a:791d PUT with payload: 0a
Mon Feb 15 22:53:12 2021 user.notice otbr-agent[6850]: 00:06:47.433 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:9bbe:901f:5823:8e37 PUT with payload: 07
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:47.577 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:50f1:d7ef:43c2:4dd3 PUT with payload: 14
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:47.721 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:aaf0:861e:1153:5821 PUT with payload: 02
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:47.776 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:76b6:3d39:8d2c:dbb9 PUT with payload: 10
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:47.800 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:cd16:189a:2358:ac3e PUT with payload: 04
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:47.816 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:e266:2f95:78a5:83fa PUT with payload: 13
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:47.887 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:6720:7303:3bbd:1147 PUT with payload: 0d
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:48.052 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:8ce4:dc46:c30:b57d PUT with payload: 03
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:48.120 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:6738:b2ee:25a5:182a PUT with payload: 0b
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:48.131 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:d838:f8b1:10a5:2c71 PUT with payload: 05
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:48.134 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:c709:59e0:2a1a:d2a8 PUT with payload: 01
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:48.187 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:159b:f725:6fde:dbe5 PUT with payload: 12
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:48.195 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:be79:8241:e1fb:729d PUT with payload: 09
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:48.202 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:b198:efaf:9d81:79c7 PUT with payload: 0e
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:48.240 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:6600:5182:e6dd:13f6 PUT with payload: 11
Mon Feb 15 22:53:13 2021 user.notice otbr-agent[6850]: 00:06:48.258 [NOTE]-CLI-----: Output: coap request from fdde:ad00:beef:0:8b51:404f:9905:9dc1 PUT with payload: 06
Mon Feb 15 22:53:13 2021 user.warn otbr-agent[6850]: 00:06:48.263 [WARN]-PLAT----: radio tx timeout
Mon Feb 15 22:53:13 2021 user.warn otbr-agent[6850]: 00:06:48.263 [WARN]-PLAT----: RCP failure detected
Mon Feb 15 22:53:13 2021 user.warn otbr-agent[6850]: 00:06:48.263 [WARN]-PLAT----: Trying to recover (1/5)
Mon Feb 15 22:53:13 2021 user.warn otbr-agent[6850]: 00:06:48.296 [WARN]-PLAT----: Error processing result: NotImplemented
Mon Feb 15 22:53:13 2021 user.warn otbr-agent[6850]: 00:06:48.296 [WARN]-PLAT----: Error waiting response: NotImplemented
Mon Feb 15 22:53:13 2021 user.crit otbr-agent[6850]: 00:06:48.296 [CRIT]-PLAT----: RestoreProperties() at ot-br-posix-2022-1-5/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp:2370: Failure
Mon Feb 15 22:53:18 2021 user.info otbr-agent[7165]: [INFO]-UTILS---: Running thread-reference-20200818-1424-ge736d2488
Mon Feb 15 22:53:18 2021 user.info otbr-agent[7165]: [INFO]-UTILS---: Thread version: 1.2.0
Mon Feb 15 22:53:18 2021 user.info otbr-agent[7165]: [INFO]-UTILS---: Thread interface: wpan0
Mon Feb 15 22:53:18 2021 user.info otbr-agent[7165]: [INFO]-UTILS---: Backbone interface: wlan0
Mon Feb 15 22:53:18 2021 user.info otbr-agent[7165]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyUSB0?uart-baudrate=460800&uart-flow-control
Mon Feb 15 22:53:18 2021 user.debug otbr-agent[7165]: 49d.19:26:54.254 [DEBG]-PLAT----: Wait response: tid=0 key=0
Mon Feb 15 22:53:18 2021 user.info otbr-agent[7165]: 49d.19:26:54.258 [INFO]-PLAT----: RCP reset: RESET_POWER_ON
[... cut ...]
Mon Feb 15 22:53:18 2021 user.debug otbr-agent[7165]: 49d.19:26:54.262 [DEBG]-PLAT----: Wait response: tid=2 key=2
Mon Feb 15 22:53:18 2021 user.debug otbr-agent[7165]: 49d.19:26:54.266 [DEBG]-PLAT----: Wait response: tid=3 key=8
Mon Feb 15 22:53:18 2021 user.debug otbr-agent[7165]: 49d.19:26:54.269 [DEBG]-PLAT----: Wait response: tid=4 key=5
```

This crash can steadily reproduce. I have tried various methods mentioned in the previous issue, such as improving baudrate to 460800, increasing hardware flow control and increasing rx buffer on the host side, but none of them had any effect.

HeFeng1947, 2022-04-04:
Update some interesting test  results:
We did the same test on the old version. From the test results, the stability of the old version（openthread/ot-br-posix@ae90b332891874df4200cfdf27f082d05a2cf7c7） is **much stronger** than that of the new version.

I'm not really sure, but maybe it's partly related to this issue #7318

MichalPoterek, 2022-05-02:
We use more than 50 devices in network, recenly we upgrade BR to Thread 1.2 and we got some problems. It looks like problem exist with both TexasInstruments and  Nordig RCP dongle. 
[image]

jwhui, 2022-05-03:
@MichalPoterek , can you provide the full logs in text form? The snippet you provided does not seem to indicate any errors from OpenThread itself. The image is also very difficult to read.

MichalPoterek, 2022-05-04:
Hi, below please find full messages file and I attached as well network charts where is seen when devices was disconnected from network. Attached PB3 - Ping Latency.csv  is  exported PB3 chart.

[Logs.zip](https://github.com/openthread/openthread/files/8626925/Logs.zip)

LancerSky, 2022-05-12:
Hi @jwhui 
We are developing OTBR base on OpenWrt. And I also got the same exception in version https://github.com/openthread/ot-br-posix/commit/5be35949c6f1fd4cc3a8e63169cec6a2211135a4. 
The RCP firmware version was built from https://github.com/openthread/ot-efr32/commit/31e9fc8606d9a68cd79868d2b3169992168a7398.
The end device firmware version was built from https://github.com/openthread/ot-nrf528xx/commit/6533ff751f1c5e05ca48bda15f7b7bc3419fc8e2.
And I didn't find 'PROP 43' in [Internet-Draft for the Spinel Host-Controller Protocol](https://tools.ietf.org/html/draft-rquattle-spinel-unified), Is this the problem?
```
Thu May 12 05:47:09 2022 user.info otbr-agent[3232]: 00:06:24.745 [INFO]-MAC-----: Received IPv6 HopOpts msg, len:83, chksum:0000, from:0x7000, sec:yes, prio:normal, rss:-40.0
Thu May 12 05:47:09 2022 user.info otbr-agent[3232]: 00:06:24.745 [INFO]-MAC-----:     src:[fdd2:b178:8ba5:0:6600:5182:e6dd:13f6]
Thu May 12 05:47:09 2022 user.info otbr-agent[3232]: 00:06:24.745 [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.991 [DEBG]-MAC-----: Received frame from short address 0x7000
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: Rx security - frame counter 37487
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: =========================[RX seqno=189 len=078]=========================
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: | 49 98 BD 32 3D FF FF 00 | 70 0D 6F 92 00 00 01 7C | I.=2=...p.o....|
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: | 5A 3F 50 F1 D7 EF 43 C2 | 4D D3 03 00 00 01 E1 06 | Z?PqWoCBMS....a.
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.992 [DEBG]-MAC-----: | 6D 04 40 E2 70 01 F0 9C | 95 16 33 D8 2C 58 03 82 | m.@bp.p...3X,X..
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.993 [DEBG]-MAC-----: | 0B 76 12 4A 10 5F 73 2C | FC BC 6E 65 74 77 6F 72 | .v.J._s,|<networ
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.993 [DEBG]-MAC-----: | 6B 5F 74 65 73 74 FF 14 | AB DD 77 4F 66 10 .. .. | k_test..+]wOf...
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:24.993 [DEBG]-MAC-----: ------------------------------------------------------------------------
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:24.993 [INFO]-MAC-----: Received IPv6 HopOpts msg, len:83, chksum:0000, from:0x7000, sec:yes, prio:normal, rss:-40.0
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:24.993 [INFO]-MAC-----:     src:[fdd2:b178:8ba5:0:50f1:d7ef:43c2:4dd3]
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:24.993 [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.038 [DEBG]-MAC-----: Received frame from short address 0x7000
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.039 [DEBG]-MAC-----: Rx security - frame counter 37488
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.041 [DEBG]-MAC-----: =========================[RX seqno=190 len=078]=========================
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.041 [DEBG]-MAC-----: | 49 98 BE 32 3D FF FF 00 | 70 0D 70 92 00 00 01 7C | I.>2=...p.p....|
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.041 [DEBG]-MAC-----: | 5A 3F 50 F1 D7 EF 43 C2 | 4D D3 03 00 00 01 E1 06 | Z?PqWoCBMS....a.
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.043 [DEBG]-MAC-----: | 6D 04 40 E2 70 01 F0 9C | 95 16 33 D8 2C 58 03 82 | m.@bp.p...3X,X..
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.043 [DEBG]-MAC-----: | 0B 76 12 4A 10 5F 73 2C | FC BC 6E 65 74 77 6F 72 | .v.J._s,|<networ
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.043 [DEBG]-MAC-----: | 6B 5F 74 65 73 74 FF 14 | 2E B2 9B 6F 66 10 .. .. | k_test...2.of...
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.044 [DEBG]-MAC-----: ------------------------------------------------------------------------
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:25.044 [INFO]-MAC-----: Received IPv6 HopOpts msg, len:83, chksum:0000, from:0x7000, sec:yes, prio:normal, rss:-40.0
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:25.045 [INFO]-MAC-----:     src:[fdd2:b178:8ba5:0:50f1:d7ef:43c2:4dd3]
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:25.045 [INFO]-MAC-----:     dst:[ff03:0:0:0:0:0:0:1]
Thu May 12 05:47:10 2022 user.warn otbr-agent[3232]: 00:06:25.046 [WARN]-PLAT----: radio tx timeout
Thu May 12 05:47:10 2022 user.warn otbr-agent[3232]: 00:06:25.046 [WARN]-PLAT----: RCP failure detected
Thu May 12 05:47:10 2022 user.warn otbr-agent[3232]: 00:06:25.046 [WARN]-PLAT----: Trying to recover (1/5)
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.047 [DEBG]-PLAT----: Wait response: tid=0 key=0
Thu May 12 05:47:10 2022 user.info otbr-agent[3232]: 00:06:25.083 [INFO]-PLAT----: RCP reset: RESET_POWER_ON
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.083 [DEBG]-PLAT----: Wait response: tid=1 key=32
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.084 [DEBG]-PLAT----: Wait response: tid=2 key=54
Thu May 12 05:47:10 2022 user.debug otbr-agent[3232]: 00:06:25.120 [DEBG]-PLAT----: Wait response: tid=3 key=53
[... cut ...]
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-ADPROXY-: Started
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-DPROXY--: started
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-BA------: Start Thread Border Agent: OK
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-UBUS----: Connected as 81469bc6
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-UBUS----: Uloop run
Thu May 12 05:47:16 2022 user.info otbr-agent[11593]: [INFO]-APP-----: Border router agent started.
```

jwhui, 2022-05-12:
@LancerSky , as FYI, the following log line is due to the issue raised in #7552 and fixed in #7570.

The radio tx timeout issue seems similar to what was reported in https://github.com/openthread/ot-nrf528xx/issues/210. Please post there as well.

Note that the spinel internet draft is not active and should not be used for reference.

LancerSky, 2022-05-13:
Hi @jwhui ,
Because of the prompt `[netif] Failed to transmit, error:NoBufs`, so I set `OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS` to 4096 or even bigger. 
Does this parameter have any effect on this? Do you have any suggestions about this parameter? Thanks.

lancersky, 2022-06-06:
Hi @jwhui ,
I pushed the patch https://github.com/openthread/openthread/commit/e83ecab39587e854492d0184d0b610dfe93da1d8 into https://github.com/openthread/ot-br-posix/commit/5be35949c6f1fd4cc3a8e63169cec6a2211135a4

```
$ quilt diff
Index: ot-br-posix-2022-02-02/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp
===================================================================
--- ot-br-posix-2022-02-02.orig/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp
+++ ot-br-posix-2022-02-02/third_party/openthread/repo/src/lib/spinel/radio_spinel_impl.hpp
@@ -1671,18 +1671,11 @@ otError RadioSpinel<InterfaceType, Proce
     do
     {
         uint64_t now;
-        uint64_t remain;
 
         now = otPlatTimeGet();
-        if (end <= now)
-        {
-            HandleRcpTimeout();
-            ExitNow(mError = OT_ERROR_NONE);
-        }
-        remain = end - now;
-
-        if (mSpinelInterface.WaitForFrame(remain) != OT_ERROR_NONE)
+        if ((end <= now) || (mSpinelInterface.WaitForFrame(end - now) != OT_ERROR_NONE))
         {
+            otLogWarnPlat("Wait for response timeout");
             HandleRcpTimeout();
             ExitNow(mError = OT_ERROR_NONE);
         }
@@ -2299,6 +2292,7 @@ void RadioSpinel<InterfaceType, ProcessC
     }
 
     --mRcpFailureCount;
+    otLogNotePlat("RCP recovery is done");
 
 exit:
     return;
```

And the log show as following, hope that helps.
```
Fri Oct  3 21:42:50 2025 user.warn otbr-agent[564]: 00:12:42.515 [WARN]-PLAT----: radio tx timeout
Fri Oct  3 21:42:50 2025 user.crit otbr-agent[564]: 00:12:42.515 [CRIT]-PLAT----: HandleRcpTimeout() at radio_spinel_impl.hpp:2220: RadioSpinelNoResponse
```

lancersky, 2022-06-06:
I added the following code to check whether the host can communicate with RCP after `radio tx timeout`. 

```
 void RadioSpinel<InterfaceType, ProcessContextType>::HandleRcpTimeout(void)
 {
+    otExtAddress tIeeeEui64;
+    otError error = Get(SPINEL_PROP_HWADDR, SPINEL_DATATYPE_EUI64_S, tIeeeEui64.m8);
+    if (error == OT_ERROR_NONE)
+    {
+        otLogWarnPlat("Host can communicate with RCP!");
+    }
+    else
+    {
+        otLogWarnPlat("Host cannot communicate with RCP!");
+    }
```
```
Fri Oct  3 22:26:53 2025 user.warn otbr-agent[12557]: 00:03:58.538 [WARN]-PLAT----: radio tx timeout
Fri Oct  3 22:26:53 2025 user.warn otbr-agent[12557]: 00:03:58.562 [WARN]-PLAT----: Host can communicate with RCP!
Fri Oct  3 22:26:53 2025 user.crit otbr-agent[12557]: 00:03:58.564 [CRIT]-PLAT----: HandleRcpTimeout() at radio_spinel_impl.hpp:2229: RadioSpinelNoResponse
```

jwhui, 2022-06-07:
Maybe @edmont or @LuDuda can share some thoughts on the issue reported in https://github.com/openthread/openthread/issues/7529#issuecomment-1147283390 ?

lancersky, 2022-06-18:
Hi @jwhui ,
I had change my RCP firmware which built from ot-efr32 to Simplicity Studio. The long test did not appear 'radio tx timeout' again.
But I ran into another problem. https://github.com/openthread/ot-efr32/issues/331

lancersky2017, 2022-06-23:
We chage the GSDK to v4.0.1, and it solve our problem. thanks.

closed 2022-06-23 by jwhui