openthread/openthread #7552: TCP malforms the logging information.

canisLupus1313, 2022-03-31:
**Describe the bug** A clear and concise description of what the bug is.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id 5eb973ce00d3eec5898d6cc63fd4a20f9e914111
2. IEEE 802.15.4 hardware platform nRF52840dk RCP + ot daemon
3. Build steps build rcp + ot daemon 
4. Network topology two boards Leader + router

During RCP crash the log outputted in system journal from ot daemon show incorrect function. `BSD TCP function`. Looks like this macro is overloaded in https://github.com/openthread/openthread/blob/main/third_party/tcplp/bsdtcp/tcp.h#L46

```
/jenkins_workspace/workspace/st-fw-nrfconnect-thread_master_4/thread_test/test_objects/rcp_host_app_1_1/bin/ot-daemon[46278]: 00:01:03.614 [C] Platform------: Too many rcp failures, exiting
/jenkins_workspace/workspace/st-fw-nrfconnect-thread_master_4/thread_test/test_objects/rcp_host_app_1_1/bin/ot-daemon[46278]: 00:01:03.614 [C] Platform------: BSD TCP function() at radio_spinel_impl.hpp:2249: Failure
```

Excluding TCP with -DOT_EXCLUDE_TCPLP_LIB=ON  DOT_CONFIG=my-posix-config.h doesn't seem to have any effect.

my-posix-config.h content
```
#include <openthread-core-posix-config.h>

/**
* @def OPENTHREAD_CONFIG_TCP_ENABLE
*
* Define as 1 to enable TCP.
*
*/
#ifndef OPENTHREAD_CONFIG_TCP_ENABLE
#define OPENTHREAD_CONFIG_TCP_ENABLE 0
#endif
```

canisLupus1313, 2022-04-04:
@samkumar Can You take a look?

samkumar, 2022-04-05:
Thanks for bringing this to my attention. This indeed looks like an issue --- I put in that `#define` statement in the original TCPlp code since it was necessary for the original platform I was developing it on. But I can absolutely see that it could be harmful in OpenThread. Tomorrow, I'll investigate this and determine the best way to remove this.

samkumar, 2022-04-05:
(Also, sorry for the delay --- I just saw this now!)

marcorussi, 2022-04-05:
Hello all, sorry to chip in but I came across the exact same problem while testing RCP sample from Nordic nrfconnect SDK. After many experiments I finally got to a point where that error happens when I send _dataset init new_ from **ot-ctl** to the **ot-daemon** connected to my nRF52840 RCP board. 
Thank you.

canisLupus1313, 2022-04-05:
@marcorussi just to keep things separate please create separate issue, and i will be glad to help you.

@samkumar This also highlights one more problem this code was supposed to be cut out from compilation and somehow there are few artifacts left.

PR #7570 (2022-04-07): [tcp] remove extraneous #define and restrict inclusion of TCPlp headers
This pull request fixes #7552.

First, this pull requests changes `third_party/tcplp/bsdtcp/tcp.h` so that it no longer does `#define __func__`. This line was necessary on one of the original platforms for which TCPlp was written. However, OpenThread assumes that `__func__` is appropriately defined by the toolchain, so there is no need to define it again.

The above change fixes issue #7552, but #7552 illustrates a broader problem, which is that global-scope artifacts (variables and macros) defined by TCPlp can affect othe [... cut]
closed 2022-04-07 by jwhui