openthread/openthread #7567: [github-actions] test_firewall.py consistently failing

jwhui, 2022-04-05:
CI checks on all PRs are failing on this test.

```
======================================================================
ERROR: test (__main__.Firewall)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/thread_cert.py", line 302, in tearDown
    self._verify_packets(test_info_path)
  File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/thread_cert.py", line 331, in _verify_packets
    self.verify(pv)
  File "./tests/scripts/thread-cert/border_router/test_firewall.py", line 239, in verify
    _pkt = pkts.filter_eth_src(vars['Host_ETH']).filter_ipv6_dst(
  File "/home/runner/work/ot-br-posix/ot-br-posix/third_party/openthread/repo/tests/scripts/thread-cert/pktverify/packet_filter.py", line 234, in must_next
    raise errors.PacketNotFound(self.index, self._stop_index)
pktverify.errors.PacketNotFound: ((561, 561), (770, 770))

----------------------------------------------------------------------
```

https://github.com/openthread/ot-br-posix/runs/5837544063?check_suite_focus=true

simonlingoogle, 2022-04-06:
Do we have firewall related change recently ? @superwhd

superwhd, 2022-04-06:
We don't have recent firewall changes. 

The test failed when verifying that HOST sends a ping packet to BR while the destination is a mesh local address or a link local address. That requires adding a route before the ping. I'm checking if the failure was due to insufficient delay.

superwhd, 2022-04-06:
Another observation is that this test case only fails when TREL is disabled. When TREL is enabled, the test is passed. However, I cannot reproduce the test failure locally when TREL is disabled.

superwhd, 2022-04-06:
I'll disable this step as for now and look for a long term solution later.

closed 2022-04-06 by jwhui