openthread/openthread #7572: Iperf3.7 work error in new commit

HeFeng1947, 2022-04-06:
**Describe the bug** Iperf3 work error in new commit. I have tested that it can work normally at ot-br-posix@a965a7535dad33303585d2080437f3d796e54a81, but it cannot work at ot-br-posix@5be35949c6f1fd4cc3a8e63169cec6a2211135a4. I don't currently have time to validate each commit between the two, so I haven't been able to determine which commit cause the issue.

**To Reproduce** Dev A create a thread network and work as iperf server, Dev B joined thread network and work as iperf client.
Dev A output:
```
# iperf3 -V -s -6 -B fd11:22:0:0:1471:928d:29d0:d09a
iperf 3.7
Linux GL-S200 4.14.221 #0 Mon Feb 15 15:22:37 2021 mips
-----------------------------------------------------------
Server listening on 5201
```
Dev B output:
```
# iperf3 -u -6 -c fd11:22:0:0:1471:928d:29d0:d09a
iperf3: error - unable to connect to server: Connection refused
```

Normally working output: 
Dev B:
```
# iperf3 -u -6 -c fd01:ed8:8750:9ae3:c227:eeb2:fadc:e479 -b 10k -l 1k
Connecting to host fd01:ed8:8750:9ae3:c227:eeb2:fadc:e479, port 5201
[  5] local fde9:350c:f299:6d1e:a626:187b:9125:3354 port 41451 connected to fd01:ed8:8750:9ae3:c227:eeb2:fadc:e479 port 5201
[ ID] Interval           Transfer     Bitrate         Total Datagrams
[  5]   0.00-1.00   sec  2.00 KBytes  16.4 Kbits/sec  2  
[  5]   1.00-2.00   sec  1.00 KBytes  8.19 Kbits/sec  1  
[  5]   2.00-3.00   sec  1.00 KBytes  8.19 Kbits/sec  1  
[  5]   3.00-4.00   sec  1.00 KBytes  8.19 Kbits/sec  1  
[  5]   4.00-5.00   sec  2.00 KBytes  16.4 Kbits/sec  2  
[  5]   5.00-6.00   sec  1.00 KBytes  8.19 Kbits/sec  1  
[  5]   6.00-7.00   sec  1.00 KBytes  8.19 Kbits/sec  1  
[  5]   7.00-8.00   sec  1.00 KBytes  8.19 Kbits/sec  1  
[  5]   8.00-9.00   sec  1.00 KBytes  8.19 Kbits/sec  1  
[  5]   9.00-10.00  sec  2.00 KBytes  16.4 Kbits/sec  2  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.00  sec  13.0 KBytes  10.6 Kbits/sec  0.000 ms  0/13 (0%)  sender
[  5]   0.00-10.59  sec  13.0 KBytes  10.1 Kbits/sec  86602.764 ms  0/13 (0%)  receiver
```

jwhui, 2022-04-07:
@HeFeng1947 , are you performing this test between two OTBR devices?

Can you provide the output of `route -6` command for both?

HeFeng1947, 2022-04-08:
> are you performing this test between two OTBR devices?

Yes.

> Can you provide the output of `route -6` command for both?

## On Server Side
```
# ot-ctl ipaddr
fda0:678:6bec:f424:0:ff:fe00:fc11
fda0:678:6bec:f424:0:ff:fe00:fc38
fd04:269c:2e05:a238:513c:c6db:4a2f:34e7
fda0:678:6bec:f424:0:ff:fe00:fc10
fda0:678:6bec:f424:0:ff:fe00:fc00
fda0:678:6bec:f424:0:ff:fe00:e000
fda0:678:6bec:f424:c2af:aed6:415b:5388
fe80:0:0:0:2897:ac37:aa14:346f
Done

# route -A inet6
Kernel IPv6 routing table
Destination                                 Next Hop                                Flags Metric Ref    Use Iface
64:ff9b::%2012282880/96                     ::%2012282880                           U     1024   1        0 tayga-nat64
2001:db8:1::1%2012282880/128                ::%2012282880                           U     1024   1        0 tayga-nat64
fd04:269c:2e05:a238::%2012282880/64         ::%2012282880                           U     256    2       40 wpan0   
fd24:8a95:a2f1::%2012282880/64              ::%2012282880                           U     512    1        0 wpan0   
fd37:bc33:4ad6:10::%2012282880/64           ::%2012282880                           U     1024   1        0 br-lan  
fd37:bc33:4ad6::%2012282880/48              ::%2012282880                           !n    2147483647 1        0 lo      
fda0:678:6bec:f424::%2012282880/64          ::%2012282880                           U     256    1        0 wpan0   
fe80::%2012282880/64                        ::%2012282880                           U     256    1        0 tayga-nat64
fe80::%2012282880/64                        ::%2012282880                           U     256    1        0 br-lan  
fe80::%2012282880/64                        ::%2012282880                           U     256    1        0 wlan0   
fe80::%2012282880/64                        ::%2012282880                           U     256    1        0 wpan0   
::%2012282880/0                             ::%2012282880                           !n    -1     1     1059 lo      
::1%2012282880/128                          ::%2012282880                           Un    0      3      121 lo      
fd04:269c:2e05:a238::%2012282880/128        ::%2012282880                           Un    0      2        0 wpan0   
fd04:269c:2e05:a238:513c:c6db:4a2f:34e7%2012282880/128 ::%2012282880                           Un    0      3       25 wpan0   
fd37:bc33:4ad6:10::%2012282880/128          ::%2012282880                           Un    0      2        0 br-lan  
fd37:bc33:4ad6:10::1%2012282880/128         ::%2012282880                           Un    0      2        0 br-lan  
fda0:678:6bec:f424::%2012282880/128         ::%2012282880                           Un    0      2        0 wpan0   
fda0:678:6bec:f424:0:ff:fe00:e000%2012282880/128 ::%2012282880                           Un    0      2        0 wpan0   
fda0:678:6bec:f424:0:ff:fe00:fc00%2012282880/128 ::%2012282880                           Un    0      2        0 wpan0   
fda0:678:6bec:f424:0:ff:fe00:fc10%2012282880/128 ::%2012282880                           Un    0      2        0 wpan0   
fda0:678:6bec:f424:0:ff:fe00:fc11%2012282880/128 ::%2012282880                           Un    0      2        0 wpan0   
fda0:678:6bec:f424:0:ff:fe00:fc38%2012282880/128 ::%2012282880                           Un    0      2        0 wpan0   
fda0:678:6bec:f424:c2af:aed6:415b:5388%2012282880/128 ::%2012282880                           Un    0      3       32 wpan0   
fe80::%2012282880/128                       ::%2012282880                           Un    0      2        0 tayga-nat64
fe80::%2012282880/128                       ::%2012282880                           Un    0      2        0 br-lan  
fe80::%2012282880/128                       ::%2012282880                           Un    0      2        0 wlan0   
fe80::%2012282880/128                       ::%2012282880                           Un    0      2        0 wpan0   
fe80::2897:ac37:aa14:346f%2012282880/128    ::%2012282880                           Un    0      2        0 wpan0   
fe80::9683:c4ff:fe18:59fe%2012282880/128    ::%2012282880                           Un    0      2        0 br-lan  
fe80::9683:c4ff:fe18:59ff%2012282880/128    ::%2012282880                           Un    0      2        0 wlan0   
fe80::deb8:9ed4:9d61:4ed4%2012282880/128    ::%2012282880                           Un    0      2        0 tayga-nat64
ff00::%2012282880/8                         ::%2012282880                           U     256    1        0 tayga-nat64
ff00::%2012282880/8                         ::%2012282880                           U     256    1        0 br-lan  
ff00::%2012282880/8                         ::%2012282880                           U     256    2      291 wlan0   
ff00::%2012282880/8                         ::%2012282880                           U     256    1        0 wpan0   
::%2012282880/0                             ::%2012282880                           !n    -1     1     1059 lo   
   
# ot-ctl netdata show
Prefixes:
fd04:269c:2e05:a238::/64 paros med e000
Routes:
fd85:b01b:8b0a:0::/64 s med e000
fd24:8a95:a2f1:0::/64 s med 9c00
Services:
44970 5d fda006786becf424c2afaed6415b5388d11f s e000
44970 5d fda006786becf42414b9bfdfea0bde27d11f s 9c00
44970 01 6e04b000000e10 s e000
[... 206 lines cut ...]
Thu Apr  7 10:13:01 2022 user.debug otbr-agent[3267]: 00:12:46.783 [DEBG]-MAC-----: Received frame from short address 0xe000
[... cut ...]
Thu Apr  7 10:13:03 2022 user.info otbr-agent[3267]: 00:12:48.038 [INFO]-MAC-----:     src:[fe80:0:0:0:201c:ad61:39a9:afb1]:19788
Thu Apr  7 10:13:03 2022 user.info otbr-agent[3267]: 00:12:48.039 [INFO]-MAC-----:     dst:[ff02:0:0:0:0:0:0:1]:19788
Thu Apr  7 10:13:03 2022 user.debug otbr-agent[3267]: 00:12:48.040 [DEBG]-PLAT----: Wait response: tid=12 key=33
```
> Can you also provide a pcap file?
[key-13fd46b652ad445d2ab38a3a3dda19bd-announce.zip](https://github.com/openthread/openthread/files/8448433/key-13fd46b652ad445d2ab38a3a3dda19bd-announce.zip)

silabs-JimL, 2022-04-09:
I met this issue as well.

I used two OTBRs to make the test.
The first one is a raspberry Pi 4 (Ubuntu 20.04) + EFR32 as the RCP.
The second one is a PC (Ubuntu 20.04) + EFR32 as the RCP.

- RCP commit cef0a4c on ot-efr32
- OTBR commit 355a415db0 on ot-br-posix

The TCP connection is always reset immediately.

**TCP Server:**
jim@jbuntu:~$ sudo ot-ctl
> state
leader
Done
> ipaddr
fdf9:8738:26e:fd9c:0:ff:fe00:fc11
fdf9:8738:26e:fd9c:0:ff:fe00:fc10
fdf9:8738:26e:fd9c:0:ff:fe00:fc38
fd3a:91f3:d261:1:cf90:c8b3:fbb0:753b
fdf9:8738:26e:fd9c:0:ff:fe00:fc00
fdf9:8738:26e:fd9c:0:ff:fe00:8000
fdf9:8738:26e:fd9c:cc74:a31a:676e:af19
fe80:0:0:0:c74:b180:5fc9:57db
Done
> tcp init
Done
> tcp listen fd3a:91f3:d261:1:cf90:c8b3:fbb0:753b 30000
Done
> Accepted connection from [fd3a:91f3:d261:1:d4c6:1212:4d82:2f2b]:49152
TCP: Connection established
TCP: Connection reset

**TCP Client:**
```
ubuntu@jim-pi4:~$ sudo ot-ctl
> state
child
Done
> ipaddr
fdf9:8738:26e:fd9c:0:ff:fe00:fc11
fd3a:91f3:d261:1:d4c6:1212:4d82:2f2b
fdf9:8738:26e:fd9c:0:ff:fe00:8001
fdf9:8738:26e:fd9c:49fd:cedd:df06:9e5
fe80:0:0:0:6ce5:bf64:7794:8d59
Done
> ping fd3a:91f3:d261:1:cf90:c8b3:fbb0:753b
16 bytes from fd3a:91f3:d261:1:cf90:c8b3:fbb0:753b: icmp_seq=1 hlim=64 time=75ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 75/75.0/75 ms.
Done
> tcp init
Done
> tcp connect fd3a:91f3:d261:1:cf90:c8b3:fbb0:753b 30000
Done
> TCP: Connection established
TCP: Connection reset
```

This is the capture made on the PC over the wpan0 interface.
[tcp-connection-reset.zip](https://github.com/openthread/openthread/files/8456201/tcp-connection-reset.zip)

jwhui, 2022-04-09:
@HeFeng1947 @silabs-JimL , can you try disabling OpenThread's internal TCP implementation by setting `OPENTHREAD_CONFIG_TCP_ENABLE=0` when building?

silabs-JimL, 2022-04-09:
@jwhui Thanks. iperf3 works after disabling the OpenThread's internal TCP implementation.

jwhui, 2022-04-09:
> @jwhui Thanks. iperf3 works after disabling the OpenThread's internal TCP implementation.

Thanks for validating! I just submitted a fix in https://github.com/openthread/ot-br-posix/pull/1336 .

HeFeng1947, 2022-04-11:
There seems to be some issues in my environment. But the test results are already available
ubuntu 21.04 + efr32mg12 (brd4161a)
raspi 4B + efr21mg21 (brd4180a)
RCP commit cef0a4c on ot-efr32
OTBR commit 355a415db0 on ot-br-posix

iperf3 server
```
pi@raspberrypi:~ $ iperf3  -V -s -6 -B fda0:678:6bec:f424:4864:a1e9:3f90:279f
iperf 3.9
Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Time: Mon, 11 Apr 2022 08:11:39 GMT
Accepted connection from fd39:cd4b:1a7:1:313b:4e77:691e:a674, port 39090
      Cookie: pi53ug4qx3ebkhwis45z3s2fdvy2iioscmdz
      Target Bitrate: 250000
[  5] local fda0:678:6bec:f424:4864:a1e9:3f90:279f port 5201 connected to fd39:cd4b:1a7:1:313b:4e77:691e:a674 port 51919
Starting Test: protocol: UDP, 1 streams, 1024 byte blocks, omitting 0 seconds, 10 second test, tos 0
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-1.00   sec  1.00 KBytes  8.19 Kbits/sec  0.000 ms  0/1 (0%)  
[  5]   1.00-2.00   sec  2.00 KBytes  16.4 Kbits/sec  56.987 ms  0/2 (0%)  
[  5]   2.00-3.00   sec  2.00 KBytes  16.4 Kbits/sec  110.782 ms  0/2 (0%)  
[  5]   3.00-4.00   sec  2.00 KBytes  16.4 Kbits/sec  152.362 ms  0/2 (0%)  
[  5]   4.00-5.00   sec  3.00 KBytes  24.6 Kbits/sec  195.778 ms  0/3 (0%)  
[  5]   5.00-6.00   sec  2.00 KBytes  16.4 Kbits/sec  229.227 ms  0/2 (0%)  
[  5]   6.00-7.00   sec  2.00 KBytes  16.4 Kbits/sec  253.013 ms  0/2 (0%)  
[  5]   7.00-8.00   sec  2.00 KBytes  16.4 Kbits/sec  283.613 ms  0/2 (0%)  
[  5]   8.00-9.00   sec  2.00 KBytes  16.4 Kbits/sec  298.660 ms  0/2 (0%)  
[  5]   9.00-10.00  sec  2.00 KBytes  16.4 Kbits/sec  315.426 ms  0/2 (0%)  
[  5]  10.00-11.00  sec  2.00 KBytes  16.4 Kbits/sec  327.681 ms  0/2 (0%)  
[  5]  11.00-12.00  sec  2.00 KBytes  16.4 Kbits/sec  351.547 ms  0/2 (0%)  
[  5]  12.00-13.00  sec  2.00 KBytes  16.4 Kbits/sec  362.126 ms  0/2 (0%)  
[  5]  13.00-14.00  sec  2.00 KBytes  16.4 Kbits/sec  366.717 ms  0/2 (0%)  
[  5]  14.00-15.00  sec  3.00 KBytes  24.6 Kbits/sec  369.409 ms  0/3 (0%)  
[  5]  15.00-16.00  sec  2.00 KBytes  16.4 Kbits/sec  372.810 ms  0/2 (0%)  
[  5]  16.00-17.00  sec  2.00 KBytes  16.4 Kbits/sec  377.283 ms  0/2 (0%)  
[  5]  17.00-18.00  sec  3.00 KBytes  24.6 Kbits/sec  380.411 ms  0/3 (0%)  
[  5]  18.00-19.00  sec  2.00 KBytes  16.4 Kbits/sec  385.857 ms  0/2 (0%)  
[  5]  19.00-20.00  sec  2.00 KBytes  16.4 Kbits/sec  388.773 ms  0/2 (0%)  
[  5]  20.00-21.00  sec  2.00 KBytes  16.4 Kbits/sec  398.031 ms  0/2 (0%)  
[  5]  21.00-22.00  sec  3.00 KBytes  24.6 Kbits/sec  397.659 ms  0/3 (0%)  
[  5]  22.00-23.00  sec  2.00 KBytes  16.4 Kbits/sec  400.998 ms  0/2 (0%)  
[  5]  23.00-24.00  sec  2.00 KBytes  16.4 Kbits/sec  404.928 ms  0/2 (0%)  
[  5]  24.00-25.00  sec  2.00 KBytes  16.4 Kbits/sec  405.660 ms  0/2 (0%)  
[  5]  25.00-26.00  sec  2.00 KBytes  16.4 Kbits/sec  403.401 ms  0/2 (0%)  
[  5]  26.00-27.00  sec  3.00 KBytes  24.6 Kbits/sec  405.433 ms  0/3 (0%)  
[  5]  27.00-28.00  sec  2.00 KBytes  16.4 Kbits/sec  409.754 ms  0/2 (0%)  
[  5]  28.00-29.00  sec  2.00 KBytes  16.4 Kbits/sec  410.468 ms  0/2 (0%)  
[  5]  29.00-30.00  sec  2.00 KBytes  16.4 Kbits/sec  410.369 ms  0/2 (0%)  
[  5]  30.00-31.00  sec  3.00 KBytes  24.6 Kbits/sec  409.345 ms  0/3 (0%)  
[  5]  31.00-32.00  sec  2.00 KBytes  16.4 Kbits/sec  412.272 ms  0/2 (0%)  
[  5]  32.00-33.00  sec  2.00 KBytes  16.4 Kbits/sec  412.892 ms  0/2 (0%)  
[  5]  33.00-34.00  sec  2.00 KBytes  16.4 Kbits/sec  418.203 ms  0/2 (0%)  
[  5]  34.00-35.00  sec  2.00 KBytes  16.4 Kbits/sec  420.648 ms  0/2 (0%)  
[  5]  35.00-36.00  sec  2.00 KBytes  16.4 Kbits/sec  424.181 ms  0/2 (0%)  
[  5]  36.00-37.00  sec  3.00 KBytes  24.6 Kbits/sec  419.650 ms  0/3 (0%)  
[  5]  37.00-38.00  sec  2.00 KBytes  16.4 Kbits/sec  422.471 ms  0/2 (0%)  
[  5]  38.00-39.00  sec  2.00 KBytes  16.4 Kbits/sec  421.240 ms  0/2 (0%)  
[  5]  39.00-40.00  sec  2.00 KBytes  16.4 Kbits/sec  420.777 ms  0/2 (0%)  
[  5]  40.00-41.00  sec  2.00 KBytes  16.4 Kbits/sec  420.380 ms  0/2 (0%)  
[  5]  41.00-42.00  sec  3.00 KBytes  24.6 Kbits/sec  416.386 ms  0/3 (0%)  
[  5]  42.00-43.00  sec  2.00 KBytes  16.4 Kbits/sec  415.931 ms  0/2 (0%)  
[  5]  43.00-44.00  sec  2.00 KBytes  16.4 Kbits/sec  411.895 ms  0/2 (0%)  
[  5]  44.00-45.00  sec  2.00 KBytes  16.4 Kbits/sec  418.314 ms  0/2 (0%)  
[  5]  45.00-46.00  sec  3.00 KBytes  24.6 Kbits/sec  415.234 ms  0/3 (0%)  
[  5]  46.00-47.00  sec  2.00 KBytes  16.4 Kbits/sec  415.114 ms  0/2 (0%)  
[  5]  47.00-48.00  sec  2.00 KBytes  16.4 Kbits/sec  419.360 ms  0/2 (0%)  
[  5]  48.00-49.00  sec  2.00 KBytes  16.4 Kbits/sec  423.635 ms  0/2 (0%)  
[... 1 lines cut ...]
iperf 3.9
```

iperf3 client
```
$ iperf3 -u -6 -c fda0:678:6bec:f424:4864:a1e9:3f90:279f -b 250k -l 1k
Connecting to host fda0:678:6bec:f424:4864:a1e9:3f90:279f, port 5201
[  5] local fd39:cd4b:1a7:1:313b:4e77:691e:a674 port 52315 connected to fda0:678:6bec:f424:4864:a1e9:3f90:279f port 5201
[ ID] Interval           Transfer     Bitrate         Total Datagrams
[  5]   0.00-1.00   sec  31.0 KBytes   254 Kbits/sec  31  
[  5]   1.00-2.00   sec  31.0 KBytes   254 Kbits/sec  31  
[  5]   2.00-3.00   sec  30.0 KBytes   246 Kbits/sec  30  
[  5]   3.00-4.00   sec  31.0 KBytes   254 Kbits/sec  31  
[... cut ...]
[  5]   7.00-8.00   sec  31.0 KBytes   254 Kbits/sec  31  
[  5]   8.00-9.00   sec  30.0 KBytes   246 Kbits/sec  30  
[  5]   9.00-10.00  sec  31.0 KBytes   254 Kbits/sec  31  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.00  sec   306 KBytes   251 Kbits/sec  0.000 ms  0/306 (0%)  sender
[  5]   0.00-10.00  sec  0.00 Bytes  0.00 bits/sec  0.000 ms  0/0 (0%)  receiver
iperf3: error - control socket has closed unexpectedly
```

HeFeng1947, 2022-04-11:
> There seems to be some issues in my environment. But the test results are already available ubuntu 21.04 + efr32mg12 (brd4161a) raspi 4B + efr21mg21 (brd4180a) RCP commit cef0a4c on ot-efr32 OTBR [...]

This seems to be related to the version of iperf, which I'm currently testing at version 3.9. We tested version 3.10.1 on OpenWRT and it worked.

HeFeng1947, 2022-04-11:
@jwhui @silabs-JimL Thanks Jonathan and Jim, I will close this issue later.

closed 2022-04-11 by HeFeng1947