openthread/openthread #7579: [CSL] overflow in calculating semiWindow.

zwx1995tju, 2022-04-08:
There is a bug-fixed PR [ [csl] fix overflow in calculating semiWindow #6734 ](https://github.com/openthread/openthread/pull/6734) 

Due to the code changed form PR [ [sub-mac] fix elapsed time in GetCslWindowEdges() #7299 ](https://github.com/openthread/openthread/pull/7299)  The overflow now pops up again.

Refer to the discussion [ [csl] Issues/inconsistencies from otPlatRadioGetNow() and mInfo.mRxInfo.mTimestamp as uint64_t #7054 ](https://github.com/openthread/openthread/issues/7054). Can we using this formula to calculate the value of `semiWindow` to avoid the overflow with using the type `uin32_t` of all relevant variable?

```
semiWindow = (elapsed / 1000000) * (Get<Radio>().GetCslAccuracy() + mCslParentAccuracy) + (elapsed % 1000000) * (Get<Radio>().GetCslAccuracy() + mCslParentAccuracy) / 1000000;
```

George-Stefan, 2022-04-08:
^^ @marius-preda

George-Stefan, 2022-04-08:
@zwx1995tju what are the values of variables so that `semiWindow` overflows? 

Isn't easier to cast to (uint64_t) and downcast back?
Since CSL phase is on 16bit => the final value of `semiWindow` can't overflow on 32bit even after converting to us

zwx1995tju, 2022-04-10:
> what are the values of variables so that `semiWindow` overflows

Hi @George-Stefan , the overflow issue may be described in this PR [ [csl] fix overflow in calculating semiWindow #6734 ](https://github.com/openthread/openthread/pull/6734)

zwx1995tju, 2022-04-10:
@jwhui Hi, Jonathan, this Issue was not created for a question but for a bug during calculating `semiWindow`, which I have found a few months ago.  [ [csl] fix overflow in calculating semiWindow #6734 ](https://github.com/openthread/openthread/pull/6734). And this bug need to be fixed.

zwx1995tju, 2022-04-10:
Hi All, this bug is also fixed with this PR[[sub-mac] fix semiWindow overflow in GetCslWindowEdges() #7549 ](https://github.com/openthread/openthread/pull/7549). Sorry for trouble everyone.

closed 2022-04-10 by zwx1995tju