openthread/openthread #7587: A nrtwork pressure test is required to verify stability of openthread

HeFeng1947, 2022-04-11:
**Is your feature request related to a problem? Please describe.** See this issue [openthread/ot-efr32#280](https://github.com/openthread/ot-efr32/issues/280), I think we need a viable network pressure test to determine some of the parameters in the OpenThread stack(like OPENTHREAD_CONFIG_NUM_MESSAGE_BUFFERS) and verify stability of openthread and obtain actual network performance indicators.

MichalPoterek, 2022-05-02:
Thats very good idea. We using more than 50 devices in Thread network. After migration to Thread 1.2  we see some stability issues - mosty related to TX time out

root@beaglebone:/home/debian/ot-br-posix# git show
commit cafd9a7a8ae606147b97244fbf1afa31ca11f0b4 (HEAD -> main, origin/main, origin/HEAD)
Author: dependabot[bot] <49699333+dependabot[bot]@users.noreply.github.com>
Date:   Thu Mar 31 18:10:45 2022 -0700

    submodule: bump third_party/openthread/repo from `f9f3163` to `5eb973c` (#1323)

    Bumps [third_party/openthread/repo](https://github.com/openthread/openthread) from `f9f3163` to `5eb973c`.
    - [Commits](https://github.com/openthread/openthread/compare/f9f316314b8be2dbe9adb195518331bf3d35e553...5eb973ce00d3eec5898d6cc63fd4a20f9e914111)

    ---
    updated-dependencies:
    - dependency-name: third_party/openthread/repo
      dependency-type: direct:production
    ...

    Signed-off-by: dependabot[bot] <support@github.com>

    Co-authored-by: dependabot[bot] <49699333+dependabot[bot]@users.noreply.github.com>

diff --git a/third_party/openthread/repo b/third_party/openthread/repo
index f9f316314b..5eb973ce00 160000
--- a/third_party/openthread/repo
+++ b/third_party/openthread/repo
@@ -1 +1 @@
-Subproject commit f9f316314b8be2dbe9adb195518331bf3d35e553
+Subproject commit 5eb973ce00d3eec5898d6cc63fd4a20f9e914111

[image]
[image]
[image]

jwhui, 2023-06-27:
Closing stale issue.

closed 2023-06-27 by jwhui