openthread/openthread #7590: RCP issues with RadioSpinelNoResponse

canisLupus1313, 2022-04-12:
Recently is our internal CI we have been observing increased number of issues where spinel interface times out. However the last reset reason on Nordic SoC shows Power On (without any potential assertions).  After some investigation there are few identified issues.

1. Initial communication with RCP fails
1. Frame is stuck in driver due to wifi coex, CSMA back off, too many retries with big inter packet gap.
1. Frame with incorrect CRC

No. 2 is disputable but I'm bringing it because there are easy ways to reproduce/simulate problems with it.

All of above could be handled in HDLC protocol but I can see that the one implemented in Openthread is somehow lite version.

So first of all few questions from me and I hope after a proposal of change.

1. Are there any frame delimiters implemented in HDLC (I couldn't find any)?
1. Is HDLC polling implemented?
2. Is there any frame re transmissions mechanism implemented on HDLC level ?

canisLupus1313, 2022-04-13:
@jwhui 
> The HDLC delimiter is the Flag Sequence value 0x7e - see [RFC 1662 Section 3.1](https://datatracker.ietf.org/doc/html/rfc1662#section-3.1).

I'm always mistaking 0x7f with 0x7e.

What do You think about altering then to implement Polling and acknowledgement in OT HDLC to alter of [this](https://github.com/openthread/openthread/blob/main/src/lib/spinel/radio_spinel_impl.hpp#L1970) timer.

The most straight forward would be to implement S-frame type in this HDLC (but this is overkill).
So let's simplify the thing saying that there will be a periodic check if the RCP is alive during radio operation and if it's ongoing and the radio is enabled. This way We would know that if the radio is processing the frame. In case of unexpected reset the Radio would state that it's not enabled so we could fire up recovery procedure.

abtink, 2022-04-13:
@canisLupus1313 the idea is that the `RadioSpinel` is somewhat independent of the underlying interface between host and RCP. We have an abstraction of this in template `InterfaceType` type and then two implementations `HdlcInterface` and `SpiInterface`. I am not sure if we have a clear definition of set of APIs we expect from an `InterfaceType`(at least I could not find it). But I think the `HdlcInterface` and `SpiInterface` can act as examples.

We leave frame retransmission/reliably model to the interface implementation, e.g. `SpiInterface` has its own mechanism. 
Regarding UART and HDLC, it is possible to add a new interface implementation which and updated/enhanced HDLC  but it requires some new designs. In general, I am myself not in favor this. As @jwhui mentioned, the expectation is that UART interface to be reliable. 

Regarding the [timer](https://github.com/openthread/openthread/blob/main/src/lib/spinel/radio_spinel_impl.hpp#L1970) I think it is for tx timeout (basically if radio does not provide a tx done update after a tx request). I think would be good to have this independent of how the interface behaves (the interface dropping the spinel frame is a possible situation for this to happen, but I guess there may be other possibilities, e.g. radio driver itself misbehaving, etc.).

canisLupus1313, 2022-04-13:
@abtink ok so I will be straight forward with You. The biggest problem for me is this [timer](https://github.com/openthread/openthread/blob/main/src/lib/spinel/radio_spinel_impl.hpp#L1970). Similar thing is present in wpantund [here](https://github.com/openthread/wpantund/blob/master/src/wpantund/NCPConstants.h#L23). But when we have both openthread stack and radio on same chip this seems to work a bit better. Increasing the value to eg.  12 s improved results in performance tests. 
As mentioned on the beginning the I've checked the reset reason on the chip and there were no crash just some delay in SW during heavy traffic.

> expectation is that UART interface to be reliable.

I understand it and tend to agree that for PCBs like we are providing it should be reliable.

To sum up extending interface with some check if RCP is alive would be slightly better approach than assuming it's dead.
If You don't like the idea I'm open to proposition.

abtink, 2022-04-13:
@canisLupus1313 I think increasing the tx timeout timer in radio_spinel may be a good approach (basically waiting for longer in case it takes longer for RCP/radio to send the radio frame and report back the status). My suggestion is we add a config like `OPENTHREAD_SPINEL_CONFIG_TX_WAIT_DELAY` to let it be configurable by platform/product.

------

The the two timers in `RadioSpinel` and `wpantund` are somewhat different and serve slightly different purposes:
- The `wpantund` one is a response timeout (whenever we send a spinel request frame, we expect a response frame within this timeout).
- The `RadioSpinel` one is not tied to request/response spinel commands over the interface, but to radio entering tx state. We basically expect to get "tx done" event/callback (success or failure) from radio within this time window.
   - We can still have a functioning interface (can exchange spinel frames between host and RCP) but somehow miss the "tx done" event so having this timer independent of how we may check the status of interface may be good.

canisLupus1313, 2022-04-15:
@abtink Sorry for delay. How about creating creating config option and cli command to set this timer also add timer on the RCP site to cancel the transmission if there were no response from radio driver in slightly lower value?

abtink, 2022-04-20:
Adding a config option sounds reasonable to me. But I am not sure if we want to allow it to be run-time configurable. I think it makes more sense to set it at build-time based on what the platform behavior.

On RCP side, my suggestion is for the radio platform to implement a timeout (if needed), i.e. If radio platform cannot send a frame within certain time, it can cancel the tx request and invoke `TxDone()` (with `OT_ERROR_ABORT).

abtink, 2022-04-20:
BTW, do you happen to have logging enabled on RCP side? 

Under certain log conifgs, logs from RCP are passed through spinel interface to host. They can add delay. If that is the case, I suggest disabling the logs on RCP during trest (or maybe reducing log level on RCP side)?

MichalPoterek, 2022-05-02:
[image]

root@beaglebone:/home/debian/ot-br-posix# git show
commit cafd9a7a8ae606147b97244fbf1afa31ca11f0b4 (HEAD -> main, origin/main, origin/HEAD)
Author: dependabot[bot] <49699333+dependabot[bot]@users.noreply.github.com>
Date:   Thu Mar 31 18:10:45 2022 -0700

    submodule: bump third_party/openthread/repo from `f9f3163` to `5eb973c` (#1323)

    Bumps [third_party/openthread/repo](https://github.com/openthread/openthread) from `f9f3163` to `5eb973c`.
    - [Commits](https://github.com/openthread/openthread/compare/f9f316314b8be2dbe9adb195518331bf3d35e553...5eb973ce00d3eec5898d6cc63fd4a20f9e914111)

    ---
    updated-dependencies:
    - dependency-name: third_party/openthread/repo
      dependency-type: direct:production
    ...

    Signed-off-by: dependabot[bot] <support@github.com>

    Co-authored-by: dependabot[bot] <49699333+dependabot[bot]@users.noreply.github.com>

diff --git a/third_party/openthread/repo b/third_party/openthread/repo
index f9f316314b..5eb973ce00 160000
--- a/third_party/openthread/repo
+++ b/third_party/openthread/repo
@@ -1 +1 @@
-Subproject commit f9f316314b8be2dbe9adb195518331bf3d35e553
+Subproject commit 5eb973ce00d3eec5898d6cc63fd4a20f9e914111

jwhui, 2022-09-13:
Closing stale issue.

closed 2022-09-13 by jwhui

canisLupus1313, 2022-09-26:
@jwhui @abtink Sorry for the long delay. we Have identified what was the reason for UART to get stuck on RCP site for nrf52840 dk. So there seems to nothing wrong with nRF SoC and with openthread stack. The dk itself has a problem with UART to USB converter that we are using on nRF52840 DKs (up until now all the revisions of the bards experience similar issue). Other version of DKs eg. nRF52820dk work without problem.

jwhui, 2022-09-26:
@canisLupus1313 , thanks for reporting back! For future reference, what board revisions are affected or not affected by this issue? Or is there a link we can refer to? Thanks!

canisLupus1313, 2022-09-26:
@jwhui our HW teams are working on this issue but the problem resides on nRF52840 DK currently all available revisions.