openthread/openthread #7595: Enabling CONFIG_OPENTHREAD_COAPS fails build

FlYnN91, 2022-04-13:
**Describe the bug**
When compiling openthread libraries through west, implementation of `mbedtls_base64_encode` is missing.

**To Reproduce**

1. Initialise nRF SDK - `west init -m https://github.com/nrfconnect/sdk-nrf --mr 1.9.1`
2. `cd v1.9.1\nrf\samples\openthread\coap_server`
3. Edit `prj.conf` to include `CONFIG_OPENTHREAD_COAPS=y` line 12
4. `west build -b nrf52840dk_nrf52840 -d build_nrf52840dk_nrf52840`

**Expected behavior**
Example project (coap_server) to compile with `CONFIG_OPENTHREAD_COAPS=y` or without.

**Console/log output**
When `CONFIG_OPENTHREAD_COAPS=y` is set in `prj.conf` the build fails.

```
C:\nordic_sdk\v1.9.1\nrf\samples\openthread\coap_server>west build -b nrf52840dk_nrf52840 -d build_nrf52840dk_nrf52840
[0/1] Re-running CMake...
Including boilerplate (Zephyr base (cached)): C:/nordic_sdk/v1.9.1/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server
-- Using NCS Toolchain 1.9.1 for building. (C:/nordic_sdk/v1.9.1/toolchain/cmake)
-- Zephyr version: 2.7.99 (C:/nordic_sdk/v1.9.1/zephyr), build: v2.7.99-ncs1-1
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: C:/nordic_sdk/v1.9.1/zephyr/.cache
-- Found dtc: C:/nordic_sdk/v1.9.1/toolchain/opt/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/nordic_sdk/v1.9.1/toolchain/opt)
-- Found BOARD.dts: C:/nordic_sdk/v1.9.1/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/boards/nrf52840dk_nrf52840.overlay
-- Generated zephyr.dts: C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/build_nrf52840dk_nrf52840/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/build_nrf52840dk_nrf52840/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/build_nrf52840dk_nrf52840/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/build_nrf52840dk_nrf52840/zephyr/dts.cmake
Parsing C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/Kconfig
Loaded configuration 'C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/build_nrf52840dk_nrf52840/zephyr/.config'
No change to configuration in 'C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/build_nrf52840dk_nrf52840/zephyr/.config'
No change to Kconfig header in 'C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/build_nrf52840dk_nrf52840/zephyr/include/generated/autoconf.h'
C:\nordic_sdk\v1.9.1\toolchain\opt\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
-- OpenThread Source Directory: C:/nordic_sdk/v1.9.1/modules/lib/openthread
-- Package Name: OPENTHREAD
-- Setting default package version: thread-reference-20200818-ncs1-rc3-560-g02e61a2ed
-- Package Version: thread-reference-20200818-ncs1-rc3-560-g02e61a2ed
CMake Warning at C:/nordic_sdk/v1.9.1/zephyr/CMakeLists.txt:764 (message):
  No SOURCES given to Zephyr library: subsys__net__pkt_filter

  Excluding target from build.

CMake Warning at C:/nordic_sdk/v1.9.1/zephyr/CMakeLists.txt:1687 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: C:/nordic_sdk/v1.9.1/nrf/samples/openthread/coap_server/build_nrf52840dk_nrf52840
[57/821] cmd.exe /C "cd /D C:\nordic_sdk\v1.9.1\nrf\samples\openthread\coap_server\build_nrf52840dk_nrf52840\modules\...";MBEDTLS_USER_CONFIG_FILE="nrf-config-user.h"" -P C:/nordic_sdk/v1.9.1/modules/lib/openthread/etc/cmake/print.cmake"
OPENTHREAD_CONFIG_ASSERT_ENABLE=1
OPENTHREAD_CONFIG_BACKBONE_ROUTER_DUA_NDPROXYING_ENABLE=0
OPENTHREAD_CONFIG_BACKBONE_ROUTER_MULTICAST_ROUTING_ENABLE=0
OPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=0
OPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=0
OPENTHREAD_CONFIG_COAP_API_ENABLE=1
OPENTHREAD_CONFIG_COAP_SECURE_API_ENABLE=1
OPENTHREAD_CONFIG_MAC_CSL_AUTO_SYNC_ENABLE=0
OPENTHREAD_CONFIG_PING_SENDER_ENABLE=1
OPENTHREAD_SPINEL_CONFIG_RCP_RESTORATION_MAX_COUNT=0
PACKAGE_NAME=OPENTHREAD
OPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1
OPENTHREAD_CONFIG_NCP_HDLC_ENABLE=1
KERNEL
__ZEPHYR__=1
[... 30 lines cut ...]
C:\nordic_sdk\v1.9.1\toolchain\opt\bin\arm-none-eabi-g++.exe -DBUILD_VERSION=v2.7.99-ncs1-1 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"nrf-config.h\" -DMBEDTLS_USER_CONFIG_FILE=\"nrf-config-user.h\" -DNRF52840_XXAA -DNRF_802154_ACK_TIMEOUT_ENABLED=1 -DNRF_802154_CCA_CORR_LIMIT_DEFAULT=2 -DNRF_802154_CCA_CORR_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_ED_THRESHOLD_DEFAULT=45 -DNRF_802154_CCA_MODE_DEFAULT=NRF_RADIO_CCA_MODE_ED -DNRF_802154_CSMA_CA_ENABLED=1 -DNRF_802154_ECB_PRIORITY=-1 -DNRF_802154_ENCRYPTION_ENABLED=0 -DNRF_802154_IE_WRITER_ENABLED=0 -DNRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0 -DNRF_802154_PENDING_EXTENDED_ADDRESSES=16 -DNRF_802154_PENDING_SHORT_ADDRESSES=16 -DNRF_802154_RX_BUFFERS=16 -DNRF_802154_SECURITY_WRITER_ENABLED=0 -DNRF_802154_SWI_PRIORITY=1 -DNRF_802154_TX_STARTED_NOTIFY_ENABLED=1 -DNRF_802154_USE_RAW_API=1 -DOPENTHREAD_CONFIG_ASSERT_ENABLE=1 -DOPENTHREAD_CONFIG_BACKBONE_ROUTER_DUA_NDPROXYING_ENABLE=0 -DOPENTHREAD_CONFIG_BACKBONE_ROUTER_MULTICAST_ROUTING_ENABLE=0 -DOPENTHREAD_CONFIG_COAP_API_ENABLE=1 -DOPENTHREAD_CONFIG_COAP_SECURE_API_ENABLE=1 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=0 -DOPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=0 -DOPENTHREAD_CONFIG_LOG_LEVEL=3 -DOPENTHREAD_CONFIG_MAC_CSL_AUTO_SYNC_ENABLE=0 -DOPENTHREAD_CONFIG_NCP_HDLC_ENABLE=1 -DOPENTHREAD_CONFIG_PING_SENDER_ENABLE=1 -DOPENTHREAD_CONFIG_THREAD_VERSION=OT_THREAD_VERSION_1_1 -DOPENTHREAD_FTD=1 -DOPENTHREAD_PROJECT_CORE_CONFIG_FILE=\"openthread-core-zephyr-config.h\" -DOPENTHREAD_SPINEL_CONFIG_RCP_RESTORATION_MAX_COUNT=0 -DPACKAGE_NAME=\"OPENTHREAD\" -DUSE_PARTITION_MANAGER=0 -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -Imodules/openthread/build/etc/cmake -IC:/nordic_sdk/v1.9.1/modules/lib/openthread/etc/cmake -IC:/nordic_sdk/v1.9.1/modules/lib/openthread/include -IC:/nordic_sdk/v1.9.1/modules/lib/openthread/src/core -Imodules/nrfxlib/nrfxlib/nrf_security/src/include/generated -IC:/nordic_sdk/v1.9.1/nrfxlib/nrf_security/include -IC:/nordic_sdk/v1.9.1/nrfxlib/nrf_security/include/mbedtls -IC:/nordic_sdk/v1.9.1/nrfxlib/nrf_security/include/psa -IC:/nordic_sdk/v1.9.1/mbedtls/include -IC:/nordic_sdk/v1.9.1/mbedtls/include/mbedtls -IC:/nordic_sdk/v1.9.1/mbedtls/include/psa -IC:/nordic_sdk/v1.9.1/mbedtls/library -IC:/nordic_sdk/v1.9.1/nrfxlib/nrf_security/../crypto/nrf_oberon/include -IC:/nordic_sdk/v1.9.1/nrfxlib/nrf_security/../crypto/nrf_oberon/include/mbedtls -IC:/nordic_sdk/v1.9.1/nrfxlib/nrf_security/../crypto/nrf_oberon/include/psa -IC:/nordic_sdk/v1.9.1/nrfxlib/crypto/nrf_cc310_platform/include -IC:/nordic_sdk/v1.9.1/nrfxlib/crypto/nrf_oberon/include/mbedtls -IC:/nordic_sdk/v1.9.1/nrfxlib/crypto/nrf_oberon/include -IC:/nordic_sdk/v1.9.1/modules/lib/openthread/src -IC:/nordic_sdk/v1.9.1/modules/lib/openthread/examples/platforms/zephyr -IC:/nordic_sdk/v1.9.1/nrf/drivers/mpsl/clock_control -IC:/nordic_sdk/v1.9.1/zephyr/include -Izephyr/include/generated -IC:/nordic_sdk/v1.9.1/zephyr/soc/arm/nordic_nrf/nrf52 -IC:/nordic_sdk/v1.9.1/zephyr/lib/libc/newlib/include -IC:/nordic_sdk/v1.9.1/zephyr/lib/util/fnmatch/. -IC:/nordic_sdk/v1.9.1/zephyr/soc/arm/nordic_nrf/common/. -IC:/nordic_sdk/v1.9.1/zephyr/subsys/net/lib/sockets/. -IC:/nordic_sdk/v1.9.1/zephyr/subsys/net/lib/openthread/platform/. -IC:/nordic_sdk/v1.9.1/zephyr/subsys/settings/include -IC:/nordic_sdk/v1.9.1/nrf/include -IC:/nordic_sdk/v1.9.1/nrf/lib/multithreading_lock/. -IC:/nordic_sdk/v1.9.1/zephyr/drivers/flash -IC:/nordic_sdk/v1.9.1/modules/hal/cmsis/CMSIS/Core/Include -IC:/nordic_sdk/v1.9.1/modules/hal/nordic/nrfx -IC:/nordic_sdk/v1.9.1/modules/hal/nordic/nrfx/drivers/include -IC:/nordic_sdk/v1.9.1/modules/hal/nordic/nrfx/mdk -IC:/nordic_sdk/v1.9.1/zephyr/modules/hal_nordic/nrfx/. -IC:/nordic_sdk/v1.9.1/modules/debug/segger/SEGGER -IC:/nordic_sdk/v1.9.1/modules/debug/segger/Config -IC:/nordic_sdk/v1.9.1/zephyr/modules/segger/. -IC:/nordic_sdk/v1.9.1/nrfxlib/mpsl/include -IC:/nordic_sdk/v1.9.1/nrfxlib/mpsl/include/protocol -IC:/nordic_sdk/v1.9.1/nrfxlib/nrf_802154/driver/include -IC:/nordic_sdk/v1. [... cut ...]
      |     ^~~~~~~~~~~~
[154/821] Building CXX object modules/openthread/build/src/core/CMakeFiles/openthread-ftd.dir/meshcop/meshcop_leader.cpp.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\nordic_sdk\v1.9.1\toolchain\opt\bin\cmake.EXE' --build 'C:\nordic_sdk\v1.9.1\nrf\samples\openthread\coap_server\build_nrf52840dk_nrf52840'
```

**Additional context**
nrf5 Connect SDK 4.1 uses a pre-build version of `libopenthread-ftd.a` where this issue does not exist.

canisLupus1313, 2022-04-14:
@FlYnN91 Try adding 
CONFIG_OPENTHREAD_COAPS=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK_ENABLED=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_ECDSA_ENABLED=y
To Your conf.

FlYnN91, 2022-04-25:
@canisLupus1313, thanks for the suggestion. This works and solved my issue. 
Somewhat disappointed that I was not able to figure this out myself based on looking at the repo.

closed 2022-04-25 by FlYnN91