openthread/openthread #7644: SRP server fails to update existing service from SRP client which got new IP addresses

AlanLCollins, 2022-04-27:
**Describe the bug** 
SRP server fails to update existing service from SRP client which got new IP addresses

**To Reproduce** 
github/openthread (commit '686c0de5fa79d5afe5b3128fabdb61e05610607d') Dec-14
github/ot-br-posix (commit '72fa16e595af28d2f2a3fbd19c3da9f5c3d397de') Dec-15
Use avahi for DNS

- Leader border router creates a network with credentials <ABC>
- Thread end node joins the <ABC> network, gets IP <XYZ> and registers service <NNN_MMM>
- Leader border router does factory reset (e.g. ot factory reset) and goes offline for ~3min, for the Thread ED to realize the leader/parent is gone. 
- Turn on Leader BR, and create the same <ABC> network. 
- Thread node will re-join and get new IP addr <OPQ>. Then it will attempt to update serice <NNN_MMM>.
- BR will respond service got updated, but the it will continue having the old IP address <XYZ> instead of <OPQ>.

**Expected behavior** 
I would expect IP address to get updated.
Even that this is a rare corner case that may not be applicable because a real factory reset should also wipe out the SRP proxy / avahi service table. However, this example exposes a potential bug in the SRP server failing to update the important fields of a service.

**Console/log output** 

**Additional context** 
I found this, by using Matter (CHIP) application framework. (version TE8_rc3) 
Amazon echo device as Matter controller (chip-tool alike).
nRF-DK board as matter light-bulb

wgtdkp, 2022-04-28:
@AlanLCollins could you share the BR and end device logs?

abtink, 2022-04-28:
@AlanLCollins , I can suggest couple of things to investigate further:

- First may be to check whether the matter code-base on device is behaving correctly and is updating the host address upon device reattaching and getting a new set of addresses.
- The OT SRP client expects the user of the SRP Client API (e.g. matter code base) to set the host name and also select and set the address (or addresses) it wants to be registered (along with other info about the service(s) to register).
- OT stack's SRP client will re-register after a new address is set (`otSrpClientSetHostAddresses()` is called).
  - These behaviors are covered by a set of tests in OT and  I did quickly check of this using CLI and it worked as expected.
- My guess/theory  is that perhaps the matter code base is somehow not set/update the new address on SRP client and SRP client is therefore registering the previous host name and address that was set on it (when it discovers the SRP server again).
- You can check the OT APIs related to this here:
https://github.com/openthread/openthread/blob/9a2d84a4b78413e20f269df96a855e573e04a616/include/openthread/srp_client.h#L403-L427

----

The other thing I can suggest is to check the service/address info on the BR SRP server after the reboot and device re-registering:
- If you have access to OT CLI on BR you can use the cli commands: https://github.com/openthread/openthread/blob/main/src/cli/README_SRP_SERVER.md
- The goal of this to see if it is correct on SRP server, and maybe the old service entry is somehow being cached somewhere (and therefore old address is still seen).

AlanLCollins, 2022-04-29:
@abtink , thank you for the guidance. I confirmed that SRP server updates correctly the IP address (`srp server service` command). However, shouldn't the SRP server notify the SRP proxy to update the mDNS daemon?. in my case, I am using Avahi.

jwhui, 2022-05-04:
@AlanLCollins , when you say you are factory resetting OpenThread, are you simply invoking the `factoryreset` OT CLI command? If so, I wonder if this is related to https://github.com/openthread/ot-br-posix/pull/1339. Is it possible for you to include this fix and see if it helps address your issue?

jwhui, 2022-10-20:
Closing stale issue.

closed 2022-10-20 by jwhui

AlanLCollins, 2023-02-06:
My apologies. I failed to respond to this ticket. Back then, I tested https://github.com/openthread/ot-br-posix/pull/1339 , and newest OT versions from Q3/Q4 2022. That fixed the issue. 
Thank you !

jwhui, 2023-02-06:
@AlanLCollins , great to hear. Thanks for reporting back!