openthread/openthread #7645: SRP server fails to send response to SRP client: "[server] send fail response: 5"

AlanLCollins, 2022-04-27:
**Describe the bug** 
SRP server fails to send response to SRP client: "[server] send fail response: 5"

**To Reproduce** 
github/openthread (commit '686c0de5fa79d5afe5b3128fabdb61e05610607d') Dec-14
github/ot-br-posix (commit '72fa16e595af28d2f2a3fbd19c3da9f5c3d397de') Dec-15
Use avahi for DNS

1. Leader border router creates a network 
2. Thread end node joins the network, and registers service with random name"NNN_MMM"
3. Thread end node does factory reset.
4. repeat from #2 for 100 interactions. 

in some test cycles, (~15 of 100).  , the SRP server fails to send a response to the SRP client:
`04-27 05:36:09.263 I/otbr-agent( 4426): [INFO]-MAC-----: Received IPv6 UDP msg, len:313, chksum:2694, from:0x8400, sec:yes, prio:normal, rss:-55.0
04-27 05:36:09.263 I/otbr-agent( 4426): [INFO]-MAC-----:     src:[fdba:be11:2aa3:0:80bd:2583:d2f5:a593]:49154
04-27 05:36:09.263 I/otbr-agent( 4426): [INFO]-MAC-----:     dst:[fdba:be11:2aa3:0:f051:f8f:393:442a]:53535
04-27 05:36:09.263 I/otbr-agent( 4426): [INFO]-SRP-----: [server] Received DNS update from fdba:be11:2aa3:0:80bd:2583:d2f5:a593
04-27 05:36:09.263 I/otbr-agent( 4426): [INFO]-SRP-----: [server] send fail response: 5`

I noticed that in normal success operation, the accessory device sends SRP DNS update within 1 second after it got CHILD ID response. , but during the failure operation, the SRP DNS update arrives ~20 seconds later.

**Expected behavior** 
The SRP server should be able to register the service.

**Console/log output** 

**Additional context** 
I found this, by using Matter (CHIP) application framework. (version TE8_rc3)
Amazon echo device as Matter controller (chip-tool alike).
nRF-DK board as matter light-bulb

jwhui, 2022-05-04:
@AlanLCollins , thanks for raising this issue. I believe this is related to https://github.com/openthread/ot-br-posix/issues/1272 and a fix was provided in https://github.com/openthread/ot-br-posix/pull/1273. Can you see if this fix addresses your issue?

AlanLCollins, 2022-05-17:
Hello @jwhui , Thank you for the guidance! 
I updated my repo to TOT for both openthread and ot-br-posix. The issue is no longer happening. 
I will close this ticket.

closed 2022-05-17 by AlanLCollins

jwhui, 2022-05-17:
> Hello @jwhui , Thank you for the guidance! I updated my repo to TOT for both openthread and ot-br-posix. The issue is no longer happening. I will close this ticket.

@AlanLCollins , awesome to hear. Thanks for reporting back!