openthread/openthread #7748: Thread 1.1 - GRL test 5.1.11 inconclusive - Nordic and Silabs DUT send Child ID Req in IEEE 802.15.4 version 2015

EricSTMicro, 2022-05-25:
**Describe the bug**
* When testing GRL test 5.1.11, which is a Thread 1.1 test, but with a DUT that runs a Openthread 1.2 stack, the test ends inconclusive because the DUT sends a Child ID Request in 802.15.4 version 2015. Of course, the GUs can't answer that as they use a former version of the standard.

I end up with the same result running this test on our chip running OT1.2 stack (for which one I want to pass the GRL certification).

**To Reproduce**
* This can be reproduced with GRL v54.2 and v55.1.

* Use official topology: 
	 '5_1_11':{
		'Leader':[DevCapb.V1_1, 'Kirale'],
		'REED_1':[DevCapb.V1_1, 'OTS'],
		'Router_2':[DevCapb.V1_1, 'OTFF'],

* Firmwares GUs 1.1:
    * OTFF: ot-20200818-54b31928.bin
    * OTS: ot-cli-ftd_4166A_1.2.3.0_a935fc51f.s37
    * Kirale: KiNOS-GOLDEN-KTWM102-1.2.7516.44190.dfu

* Firmwares DUT 1.2: (for GRL v55.1)
    * OTNCS: ot-cli-ftd-1.2-20220322-faed5e32c.zip
    * OTS: ot-cli-ftd-4166A-grl-21q4-dev-b1481.s37
    
* Run test ROUTER 5.1.11

**Expected behavior**
* The DUT should send a Child ID Request in 802.15.4 version 2003 or 2006.

**Attached file**
* Wireshark captures:
[5.1.11 test wireshark.zip](https://github.com/openthread/openthread/files/8769269/5.1.11.test.wireshark.zip)

EricSTMicro, 2022-05-30:
Hello Jonathan,

I can't debug on Nordic/Silabs dongles, but on our board, we end up in this case:

[image]

The boolean _aIePresent_ is set to false, and we don't have the define _OPENTHREAD_CONFIG_MAC_CSL_TRANSMITTER_ENABLE_ so the other cases are not triggered.

In function _IsEnhAckProbingActive()_, I couldn't get the values of _mEnhAckProbingMetrics_, but it is like it set to a value != 0.

edmont, 2022-06-01:
Hi @jwhui, yes, I looked into our logs and we can also observe this issue. It doesn't reproduce always.

EricSTMicro, 2022-06-01:
Hi @edmont, 
indeed the issue is not always reproduced, but I would say most of the time it is.

I would add that Kirale dongle doesn't seem affected by this issue.

closed 2022-08-23 by jwhui