openthread/openthread #7769: [settings] explicitly forbid to use both Set and Add on the given key

LuDuda, 2022-05-27:
Currently Settings API defines two functions which create/update/add a record.

- `otPlatSettingsSet()`
- `otPlatSettingsAdd()`

In the current OpenThread implementation, above APIs are always used exclusively for given `keyId`. In fact, `otPlatSettingsAdd()` is only used for storing a new entry in Child Table.

However, the API description for `Set` function says: `If there was more than one value previously associated with aKey, then they are all deleted and replaced with this single entry.`

For certain platforms and KVS implementations, this requirement translates to perform both `Delete` and `Add/Update` operations, which may cause unnecessary flash operations - compare to just performing `Add/Update` operation (if we would be sure that otPlatSettingsAdd was not called before on the given `keyId`).

I wonder, if we could change the API descirption to have a contract that on the specific `keyId` either `otPlatSettingsSet()` or `otPlatSettingsAdd()` operation can be used, but never both of them.

simonlingoogle, 2022-05-27:
Evaluating `Count > 1` takes one flash read, and it is always `false` for setting keys that do not use `otPlatSettingAdd()` so there should be no `Delete`.

Is it really a concern?

LuDuda, 2022-05-27:
> Evaluating `Count > 1` takes one flash read, and it is always `false` for setting keys that do not use `otPlatSettingAdd()` so there should be no `Delete`.
> 
> Is it really a concern?

Thanks for the feedback @simonlingoogle. This is possible, however evaluating `Count_Entries(keyId) > 1` may take quite a lot of time on some platforms, especially if there is quite a lot of entries already stored.

As you correctly pointed out, this evaluation will return always `false` for current OT implementation - but API documentation theoretically does not forbid OT core to change this behavior. That's why I would like to consider changing the contract in the API documentation for the future. I don't think that removing `otPlatSettingAdd` is necessary.

abtink, 2022-05-28:
> I wonder, if we could change the API descirption to have a contract that on the specific keyId either `otPlatSettingsSet()` or `otPlatSettingsAdd()` operation can be used, but never both of them.

@LuDuda I think what are you proposing sounds good and reasonable to me. 
OT stack either uses a settings key as a single-value property or a list-based property.

We should require such a contract for any key id (not tied to any specific key - in general I think platform should not care or be aware of what each key means or represents). 

Maybe something like this:

_OT stack guarantees that to use `otPlatSettingsSet()` for a `aKey` that was either previously set using`otPlatSettingsSet()` (i.e., contains a single value) or is empty and/or fully deleted (contains no value). Platform layer can rely and use this fact for optimizing its implementation ._

And then something similr for `otPlatSettingsAdd()`. 

@jwhui any concern?

-----

> Being unfamiliar with how Settings work, I wonder why otPlatSettingsAdd is necessary as an interface.
> 
> Having to implement both Set and Add may complicate the platform implementation.
> I think we can implement Add functionality by only using Set .

@simonlingoogle, it make sense for platform to implement `Add` since they can do this more efficiently (can use different approaches based on the capability of the underlying settings mechanism). It is an unordered list and items can be removed in any order.

abtink, 2022-05-28:
One thing to consider is that we can only guarantee this behavior (not use `Set` and `Add` for same key) from OT stack.

If the `otPlatSettings` APIs happen to be used by app layer code (above OT stack) or maybe internally in platform implementation, we cannot know how they may be used. Though I found it unlikely that a key is used as both single-value and list-based model.

Technically all `otPlat` APIs are only intended for use by OT stack but I know they have been cases where the OT platform APIs were used by app or platform layer code. This is something that I highly discourage.

simonlingoogle, 2022-05-29:
> One thing to consider is that we can only guarantee this behavior (not use `Set` and `Add` for same key) from OT stack.

Agree. We can add such restriction to the API contract. 

> If the `otPlatSettings` APIs happen to be used by app layer code (above OT stack) or maybe internally in platform implementation, we cannot know how they may be used. Though I found it unlikely that a key is used as both single-value and list-based model.

App developers should read the API doxygen and use API correctly. Platform code should already honor its own implementation restrictions. 

Maybe it's fine to define such restriction in API contracts after all.

LuDuda, 2022-05-30:
Thank you! I will create a PR with proposed changes.

PR #7774 (2022-06-01): [platform] update documentation of otPlatSettings API
Fixed: #7769
closed 2022-06-01 by jwhui