openthread/openthread #7803: Advertise the DNS/SRP service unicast with address/port in the Service TLV (not Server TLV)

EskoDijk, 2022-06-09:
Feedback from testing was that the BR advertises its DNS/SRP service unicast in the 'legacy' format i.e. with the server IPv6 address and port in the Server TLV.  This should be in the Service TLV as now required by the test cases.

The comments for the function PublishDnsSrpServiceUnicast() in network_data_publisher.hpp seem to suggest already that the data is included in the right place (Service TLV) but I couldn't find where in the code this Service TLV is created so couldn't verify.

If we believe this issue is already addressed, it can be closed. A pointer to the code where it actually happens would be nice :)

jwhui, 2022-06-09:
The SRP server does indeed currently publish itself using a Server TLV.

There is code to support advertising either as a Service TLV or a Server TLV:

Service TLV:
https://github.com/openthread/openthread/blob/b2132a903162be76eb4ea8a107e82c4e0737525f/src/core/thread/network_data_publisher.cpp#L627-L630

Server TLV:
https://github.com/openthread/openthread/blob/b2132a903162be76eb4ea8a107e82c4e0737525f/src/core/thread/network_data_publisher.cpp#L632-L635

But it is currently setup to only advertise as Server TLV:
https://github.com/openthread/openthread/blob/b2132a903162be76eb4ea8a107e82c4e0737525f/src/core/net/srp_server.cpp#L150

This is to support compatibility with existing HomeKit Accessories that expect Server TLV.

We should add a build configuration to switch between the Service and Server TLVs.

jwhui, 2022-06-09:
> Feedback from testing was that the BR advertises its DNS/SRP service unicast in the 'legacy' format i.e. with the server IPv6 address and port in the Server TLV. This should be in the Service TLV as now required by the test cases.

Is it appropriate to require publishing in Service TLV? The specification supports both Service TLV and Server TLVs.

EskoDijk, 2022-06-10:
> Is it appropriate to require publishing in Service TLV? The specification supports both Service TLV and Server TLVs.

The spec does state that inclusion in Service TLV is 'preferred', but it's not mandatory. Thinking about this: the only reason why inclusion in Service TLV is preferred is for a case where potentially multiple BRs advertise the same unicast address/port. Then it's more efficient. However if the Unicast service is used to announce a BR's own (ML-EID) address then it wouldn't matter how the BR would include it in terms of efficiency. So I'm now leaning towards relaxation the tests to allow both.

> We should add a build configuration to switch between the Service and Server TLVs.
Given the above - where to include the address/port might even be dependent on the particular server address included. E.g., if ML-EID of BR itself then include in Server TLV, if a manually configured (external) SRP server then include in Service TLV. And that wouldn't work if it's build-time determined. However I think the most immediate need for OT-BR is to support the ML-EID local server so we could leave such optimizations for later also.

simonlingoogle, 2022-06-10:
Agree on relaxation. 

To be honest, I have always been unable to appreciate why the Service TLV is preferred for adding the unicast address and port.

Conceptually, `Service Data` should identify the service, which is `unicast SRP service`, and `Server Data` should locate the server with IP6 address and port.

abtink, 2022-06-12:
> Feedback from testing was that the BR advertises its DNS/SRP service unicast in the 'legacy' format i.e. with the server IPv6 address and port in the Server TLV. This should be in the Service TLV as now required by the test cases.

We have a clear difference in how SRP clients would treat and  prefer between these options ([SPEC-1056](https://threadgroup.atlassian.net/browse/SPEC-1056))
https://github.com/openthread/openthread/blob/b2132a903162be76eb4ea8a107e82c4e0737525f/src/core/net/srp_client.cpp#L1845-L1850

- First one (address/port in "Service TLV") is intended when we have an infra provided common SRP server.
   - OT provides `otNetDataPublishDnsSrpServiceUnicast()` to publish entry in net data by BRs.
   - In such a case, BRs themselves most likely would not be acting as SRP server (so OT SRP server would not be enabled).
   - I expect there'd be a mechanism for BR to discover the common SRP server address/port and use the API to publish it in netdat (for clients to use).
- Second one (anycast) is for when we have SRP replication enabled.
- Third one (address/port in "Server subTLV") is intended when BRs are providing SRP server functionality directly.
  - OT SRP server can be configured to use this model (or anycast model when SRPL is enabled).
  - The address used is always the ML-EID of BR itself.
     - Initially we used this model and considered it a "legacy" way to be compatible with HomeKit accessories
     - But it now serves an additional purpose to distinguish and prefer correctly on client side between this, SRPL anycast model and then infra provided common SRP server.
     
--------

> To be honest, I have always been unable to appreciate why the Service TLV is preferred for adding the unicast address and port.

@simonlingoogle , the idea is that in the infra provided common SRP server case, we want address/port to be part of the common "Service TLV" so if multiple BRs publish this, we avoid repeating the same info in every "Server subTLV" by each BR (basically optimizing network data use).

simonlingoogle, 2022-06-13:
> @simonlingoogle , the idea is that in the infra provided common SRP server case, we want address/port to be part of the common "Service TLV" so if multiple BRs publish this, we avoid repeating the same info in every "Server subTLV" by each BR (basically optimizing network data use).

Having one BR advertise the infra SRP server should be enough? End devices can still use all BRs for routing the SRP updates as long as one BR advertises the server IP and port.

jwhui, 2022-08-23:
Closing stale issue.

closed 2022-08-23 by jwhui